olikraus/u8g2 #966: Using pages results in a slow drawing

Nirach-N09, 2019-08-06:
https://github.com/Nirach-N09/Dredd-2012-Lawgiver/blob/master/Lawgiver.ino

In the code there the only way I could get the u8g2 to display to the screen correctly was to use u8g2.firstPage(); - But that seems to have the adverse effect of making the loading bar very slow compared to the initial rendition.

I'm convinced this is my implementation, not the library itself, but I can't tell if this behaviour is by design, or if I'm using pages incorrectly.

First using a 128x32 (Although it is the same on the 256x64 if I don't use a page (but then the rest doesn't display)) https://i.imgur.com/oNa2RcQ.mp4
And then using a 256x64 and a page (necessary to make it display the entire thing): https://i.imgur.com/8MAa9zp.mp4

Is this expected behaviour? If so, is there a better way to do it?

olikraus, 2019-08-06:
1. Do not mix clearBuffer and sendBuffer with the first next loop. Just remove clearBuffer and sendBuffer.
2. Do not call digitalRead within first next loop

Nirach-N09, 2019-08-06:
I'm afraid I'm a bit of a newbie, so while I've understood and done the first point, I'm struggling with point 2.

When you say don't call digitalRead, I assume you mean remove the 

`  buttonState = digitalRead(buttonPin);`
and
`        if (buttonState != lastButtonState)
        {        
          if (buttonState == LOW && idcheck == false)
          {
          FMJCounter--;
          }
        }
        lastButtonState = buttonState;`

sections of the code.

Out of curiosity I did try that, and it didn't change much - Although following point 1 did improve the timing by about half a second.

olikraus, 2019-08-06:
The problem with point two is: the number of iterations depends on the display. So you will not know how often digitalRead gets called. In fact it may create unexpected artefacts when the value changes during the first next loop.

olikraus, 2019-08-06:
https://github.com/olikraus/u8g2/blob/6def8048c0ef2c1ca19ed58d470f71becbc3c7fb/doc/faq.txt#L184

Nirach-N09, 2019-08-07:
Hm. I think I've done what you've suggested and it hasn't changed much. That said, this is obviously not an issue with the library but my implementation of things, so I'll close it.

Thank you for your input!

closed 2019-08-07 by Nirach-N09