olikraus/u8g2 #968: two 128x32 display as one 128x64

FxIII, 2019-08-08:
I would like to use two stacked 128x32  display as one 128x64. I would like to write a new class that wraps the thing out and expose the same API. 
How complex it should be and which is the best approach I can follow?
In general there is some work in progress to make this kind of displays "tile-able"?

olikraus, 2019-08-08:
It would be possible to implement a virtual u8x8 devices which redirects the tiles to either the upper or the lower real display. In fact it could be more flexible in the sense of supporting multiple displays.

It probably wouldn't be that complicated, but still the question is, is it worth to do this?

Implementation idea:
 - global list of u8x8 pointers, along with a x/y tile position where the u8x8 is placed on the virtual screen
 - new fn to add a real display to that global list: "u8x8_AddToVirtualDisplay(u8x8_t *u8x8, uint8_t x, uint8_t y). This function should work for both C++ and C interface. It will place the real display at the specified position.
 - add one or more virtual display callback functions. The problem will be that the virtual display requires a fixed size. In your case it will be 16x8 tiles.

FxIII, 2019-08-10:
what about u8g2?

olikraus, 2019-08-10:
U8g2 will be based on the above u8x8 virtual display.

olikraus, 2019-10-03:
Maybe it shouldn't be a list, but a chained list of objects, connected with a next pointer...

olikraus, 2020-01-19:
difficult to implement... postponed

CWGwaltney, 2023-12-26:
was this ever implemented,,, now we need 2 x 128 x 64

olikraus, 2023-12-26:
that had been an idea, but I never implemented this... There is an issue here, where something similar was implemented here: https://github.com/olikraus/u8g2/issues/1351

At the end of that issue, there is some instruction on how to solve this 2x128x64 topic.