olikraus/u8g2 #969: Error including custom font .c from bdfconv

Maverick87Shaka, 2019-08-09:
Hi,
Trying to include custom font in my projects make my compilation fails with this error:
`Arduino:1.8.9 (Windows 10), Scheda:"Arduino/Genuino Uno"

Opzioni di compilazione cambiate, ricompilo tutto
sketch\ds-digital.c:7:7: error: unknown type name 'uint8_t'

 const uint8_t u8g2_font_dsdigital_tf[1713] U8G2_FONT_SECTION("u8g2_font_dsdigital_tf") = 

       ^

sketch\helvb18_tf.c:7:7: error: unknown type name 'uint8_t'

 const uint8_t u8g2_font_helvB18_tf[2253] U8G2_FONT_SECTION("u8g2_font_helvB18_tf") = 

       ^

sketch\ds-digital.c:7:44: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'U8G2_FONT_SECTION'

 const uint8_t u8g2_font_dsdigital_tf[1713] U8G2_FONT_SECTION("u8g2_font_dsdigital_tf") = 

                                            ^

sketch\helvb18_tf.c:7:42: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'U8G2_FONT_SECTION'

 const uint8_t u8g2_font_helvB18_tf[2253] U8G2_FONT_SECTION("u8g2_font_helvB18_tf") = 

                                          ^

exit status 1
`

fails also if Ii'm going to copy/paste the content directly on my projects.

Used command line like example:

`bdfconv.exe -v -f 1 -m "32-127" ../bdf/helvB18.bdf -o helvb18_tf.c  -n u8g2_font_helvB18_tf -d ../bdf/helvB18.bdf`

olikraus, 2019-08-09:
You probably need to include u8g2.h

Maverick87Shaka, 2019-08-09:
Thanks for fast feedback, seems not working also including u8g2.h manually:
```
#include "ds-digital.c"
#include "C:\Users\Pado\Documents\Arduino\libraries\U8g2\src\clib\u8g2.h"
```

the result is:
```
Arduino:1.8.9 (Windows 10), Scheda:"Arduino/Genuino Uno"

sketch\ds-digital.c:7:7: error: unknown type name 'uint8_t'

 const uint8_t u8g2_font_dsdigital_tf[1713] U8G2_FONT_SECTION("u8g2_font_dsdigital_tf") = 

       ^

sketch\helvb18_tf.c:7:7: error: unknown type name 'uint8_t'

 const uint8_t u8g2_font_helvB18_tf[2253] U8G2_FONT_SECTION("u8g2_font_helvB18_tf") = 

       ^

sketch\ds-digital.c:7:44: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'U8G2_FONT_SECTION'

 const uint8_t u8g2_font_dsdigital_tf[1713] U8G2_FONT_SECTION("u8g2_font_dsdigital_tf") = 

                                            ^

sketch\helvb18_tf.c:7:42: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'U8G2_FONT_SECTION'

 const uint8_t u8g2_font_helvB18_tf[2253] U8G2_FONT_SECTION("u8g2_font_helvB18_tf") = 

                                          ^

exit status 1
```

olikraus, 2019-08-09:
U8g2 must be included first

olikraus, 2019-08-09:
BTW you could also include U8g2lib.h

Maverick87Shaka, 2019-08-09:
Uhm, seems is not that the problem:
```
#include "C:\Users\Pado\Documents\Arduino\libraries\U8g2\src\U8g2lib.h"
#include "C:\Users\Pado\Documents\Arduino\libraries\U8g2\src\clib\u8g2.h"
#include "ds-digital.c"
```
Results:
```
Arduino:1.8.9 (Windows 10), Scheda:"Arduino/Genuino Uno"

Opzioni di compilazione cambiate, ricompilo tutto
sketch\ds-digital.c:7:7: error: unknown type name 'uint8_t'

 const uint8_t u8g2_font_dsdigital_tf[1713] U8G2_FONT_SECTION("u8g2_font_dsdigital_tf") = 

       ^

sketch\ds-digital.c:7:44: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'U8G2_FONT_SECTION'

 const uint8_t u8g2_font_dsdigital_tf[1713] U8G2_FONT_SECTION("u8g2_font_dsdigital_tf") = 

                                            ^

exit status 1
```

I also try to add manually the file content into src/clib/u8g2_fonts.c  but same problem ( even not using the font yet with function setFont()

In the zip the resulting font after conversion

[bdfconv.zip](https://github.com/olikraus/u8g2/files/3488125/bdfconv.zip)

olikraus, 2019-08-09:
Hmmm... Isn't that an Arduino environment?
Maybe you need to include stdint.h

Maverick87Shaka, 2019-08-09:
Yes, It's Arduino IDE, I've trying including stdint.h, even should be already done with #include <Arduino.h> that is on the top of my .ino project.

There is some working .c example to include to test if it's an error on the content? maybe some escape characters or something like that.

v1993, 2019-08-09:
I think that your `.c` file is different compilation unit from `.ino` one, so it does not share includes with it. Try adding include there as well I think.

Maverick87Shaka, 2019-08-09:
Stupid me! you can close this issues, it's now working!
I need to put the line:
`#include "C:\Users\Pado\Documents\Arduino\libraries\U8g2\src\clib\u8g2.h"`
directly in the .c file of the font, and not in my .ino project :-(

Thanks for really fast support ;)

v1993, 2019-08-09:
Glad it worked! You also don't need to put full path (it's very bad for sharing), using

```C++
#include <u8g2.h>
```
instead.

v1993, 2019-08-12:
I guess this can be closed?

olikraus, 2019-08-12:
Ok...

closed 2019-08-12 by olikraus