olikraus/u8g2 #970: Porting to Atmel 1614 for SSD1306 using I2C

sognatore99, 2019-08-10:
I have been trying to port this to work on a Atmel 1614 and SSD 1306.  It has taken considerable time to try to figure out the I2C using Atmel Studio.  I believe that I have it working correctly but I do not get anything on the display.  I have read your porting procedure (https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#hardware-i2c-communication).  Below is my routine I use, which is exactly like yours.  I have also attached a screen shot of the I2C data capture.  So I know data is going across but not sure if the timing is correct.  Btw nothing in the "u8x8_avr_gpio_and_delay" is being called except the first case statement "U8X8_MSG_GPIO_AND_DELAY_INIT".  Any help would be appreciated as I have spent the last several weeks trying to get this to work.  Thanks.

uint8_t u8x8_byte_arduino_hw_i2c(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{	
		
	  static uint8_t buffer[32];		/* u8g2/u8x8 will never send more than 32 bytes between START_TRANSFER and END_TRANSFER */
	  static uint8_t buf_idx;
	  uint8_t *data;
	  
	switch(msg)
	{
		case U8X8_MSG_BYTE_SEND:
		
			data = (uint8_t *)arg_ptr;
			while( arg_int > 0 )
			{
				buffer[buf_idx++] = *data;
				data++;
				arg_int--;
			}
			
		break;
		case U8X8_MSG_BYTE_INIT:
		
			address = u8x8_GetI2CAddress(u8x8);
		
		break;
		case U8X8_MSG_BYTE_SET_DC:
		break;
		case U8X8_MSG_BYTE_START_TRANSFER:
			buf_idx = 0;
	
		break;
		
		case U8X8_MSG_BYTE_END_TRANSFER:
				
			I2C_Write_Data(address, buffer, buf_idx);
		
		break;
		default:
		return 0;
	}
	return 1;
}

[image]

olikraus, 2019-08-10:
The callback looks ok. How does the setup/rest of your code look like?

sognatore99, 2019-08-11:
I have  listed the main.c below.  I am developing in Atmel studio so it generates a lot of other files.  I also apologize in advance for the poor layout, I am not a c/c++ programmer and I was basically throwing everything I could think of to get the I2C to work :)

#include <atmel_start.h>
#include "clock_config.h"
#include <avr/io.h>
#include <utils/atomic.h>

#include <u8g2.h>	//https://github.com/olikraus/u8g2
					//Porting to a new micro controller  https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#hardware-i2c-communication

#include <util/delay.h>

#define P_CPU_NS (1000000000 / F_CPU)
#define SSD1306  0x78
#define I2C_WRITE   0

u8g2_t u8g2;
uint8_t u8x8_avr_delay(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr);
uint8_t u8x8_avr_gpio_and_delay(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr);
uint8_t u8x8_byte_arduino_hw_i2c(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr);

uint8_t address;
i2c_error_t error;

volatile uint8_t startTransferCounter = 0;

/** Structure passed into read_handler to describe the actions to be performed by the handler */
typedef struct {
	uint8_t *data;
	uint8_t  size;
} transfer_descriptor_t;

uint8_t u8x8_avr_delay(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
	uint8_t cycles;

	switch(msg)
	{
		case U8X8_MSG_DELAY_NANO:     // delay arg_int * 1 nano
		// At 20Mhz, each cycle is 50ns, the call itself is slower.
		break;
		case U8X8_MSG_DELAY_100NANO:    // delay arg_int * 100 nano seconds
		
		
		// Approximate best case values...
		#define CALL_CYCLES 26UL
		#define CALC_CYCLES 4UL
		#define RETURN_CYCLES 4UL
		#define CYCLES_PER_LOOP 4UL

		cycles = (100UL * arg_int) / (P_CPU_NS * CYCLES_PER_LOOP);

		if(cycles > CALL_CYCLES + RETURN_CYCLES + CALC_CYCLES)
		break;

		__asm__ __volatile__ (
		"1: sbiw %0,1" "\n\t" // 2 cycles
		"brne 1b" : "=w" (cycles) : "0" (cycles) // 2 cycles
		);
		break;
		case U8X8_MSG_DELAY_10MICRO:    // delay arg_int * 10 micro seconds
		for(int i=0 ; i < arg_int ; i++)
		_delay_us(10);
		break;
		case U8X8_MSG_DELAY_MILLI:      // delay arg_int * 1 milli second
		for(int i=0 ; i < arg_int ; i++)
		_delay_ms(1);
		break;
		default:
		return 0;
[... 203 lines cut ...]
		return 0;
	}
	return 1;
}

int main(void)
{
	/* Initializes MCU, drivers and middleware */
	atmel_start_init();
	
	ENABLE_INTERRUPTS();
	
	u8g2_Setup_ssd1306_i2c_128x64_noname_f(&u8g2, U8G2_R0, u8x8_byte_arduino_hw_i2c, u8x8_avr_gpio_and_delay);
	u8g2_SetI2CAddress(&u8g2, SSD1306);
	u8g2_InitDisplay(&u8g2);
	u8g2_SetPowerSave(&u8g2, 0);

	u8g2_ClearBuffer(&u8g2);
	u8g2_SetFont(&u8g2, u8g2_font_fur35_tn);
	u8g2_SetFontRefHeightText(&u8g2);
	u8g2_SetFontPosTop(&u8g2);
	u8g2_DrawStr(&u8g2, 0, 14, "AVR");
	u8g2_SendBuffer(&u8g2);

}

olikraus, 2019-08-11:
From my perspective everything looks good. If you remove everything after ClearBuffer, do you see some snow on the screen?

Did you check wiring?

Which voltage do you use?

sognatore99, 2019-08-11:
Using 5v and did check the wiring, and no snow on the screen just blank.  What I do is switch projects and run a oled lcd code from (https://github.com/PaulKlinger/tinypc) to make sure everything is working.  This project uses a different I2C (simple_i2c_fgras78.c) but I was not able to get that to work with your code either.  Your graphic library is more robust and that is why I am putting a lot of effort into it.  What is confusing is that I ran the same test on a Nano and captured the same signals.  If you look they look very similar.  Now I do not know about the timing maybe that is wrong?  I have tested the baud at 100khz, 400khz and 800 khz with 20Mhz clock and even prescaled it to 5Mhz.  Wow my mind is blown.
[image]

sognatore99, 2019-08-11:
One thing I did notice is that it does not erase the screen if something is on it already but makes it brighter

sognatore99, 2019-08-12:
Ok I had to take a step back and look to see what could be wrong.  One of the things I tested was to remove the PicKit4 programmer from the circuit.  So now I see snow on the screen.  When I comment out the remaining code thou it clears the screen but no text.  Any suggestion from this point?  Thanks for you help.

sognatore99, 2019-08-12:
ok I got it all working.  I had to use the following which I found in another in another issue (https://github.com/olikraus/u8g2/issues/325).  Thanks for your help!  Note to any users of the Pickit4 when I switched to release it started to work, I guess debugging was the issue!

	u8g2_SetI2CAddress(&u8g2, SSD1306);
	u8g2_InitDisplay(&u8g2);
	u8g2_SetPowerSave(&u8g2, 0);

	u8g2_SetFont(&u8g2, u8g2_font_6x12_tf);
	u8g2_ClearBuffer(&u8g2);
	u8g2_DrawStr(&u8g2, 0,12, "STM32L031");
	u8g2_DrawStr(&u8g2, 20,24, "MHz");
	u8g2_SendBuffer(&u8g2);

closed 2019-08-12 by sognatore99

olikraus, 2019-08-12:
ah... ok...