olikraus/u8g2 #973: esp8266 webserver with u8x8 issue

Paul760, 2019-08-14:
I can have an ESP8266 running a simple webserver and I can have an ESP8266 using an oled display with u8x8lib.h

But I cannot combine them. When adding the line

U8X8_SSD1306_128X64_NONAME_HW_I2C OLED_1(/* reset=*/ U8X8_PIN_NONE); 

The webserver does not respond anymore. While the oled displays the right text and is working. Is there a known issue with the using the u8x8lib and a webserver on a esp8266??

v1993, 2019-08-14:
I think that you somehow block server thread when you're not supposed to? Drawing is also pretty slow and probably should be moved into another thread if possible, I think.

dilucide, 2019-08-14:
Just to clarify: The error occurs immediately when adding the aforementioned declaration line or are you calling any additional methods of u8g2?

olikraus, 2019-10-03:
No reply from @Paul760 ... closing

closed 2019-10-03 by olikraus