olikraus/u8g2 #974: Platform IO Library Issue

pthompson8594, 2019-08-15:
I installed u8g2 library from the platform io library installer and when using that version my display is shifted to right by about 20 pixels and seems to roll over ghosted. [Display = SSD1322_NHD_256X64_F_4W_HW_SPI] However if I overwrite the platform IO library manually with same version of u8g2 from this github everything works fine. Seems as if there might be an issue with library that platform IO downloads.

Platform IO Downloaded Library version = 2.26.14
 [image]
Github Downloaded Library version=2.26.14
[image]

olikraus, 2019-08-15:
Not sure whether Platform IO people will read the issue here...
At least from my perspective (u8g2 author) i do not know how to support here.

olikraus, 2019-08-15:
Maybe
https://community.platformio.org/c/libraries
would  be a better place to ask.

pthompson8594, 2019-08-15:
Yeah, that makes sense. Though I would bring it to your attention anyway. Although I guess your version works so not really your problem. Thanks

closed 2019-08-15 by pthompson8594

olikraus, 2019-08-15:
👍