olikraus/u8g2 #975: ST7920 with I2C supported?

simonemauri, 2019-08-20:
Hallo. I bought a ST7920 LCD 128x64 monochrome with soldered a I2C expander board...

Wich constructor must I use?

Thanks a and compliments for the very good job with this library!

[image]

olikraus, 2019-08-20:
Unfortunately, this is not supported.
I think in most cases, this is an intelligent expander chip, which serves itself as a graphics processor. This means, that all the commands for the ST7920 are not understood by the I2C interface, so it is much more complicated than just adding I2C as an additional interface.

You must use the lib supplied with your expander board.

simonemauri, 2019-08-21:
Ok. I understand...
I already used your library for the first LCDs with easy success! :-)

well, in case this is the documentation suggested from the dealer:

https://robotdyn.com/i2c-graphic-128x64-lcd-adapter.html
http://www.gammon.com.au/forum/?id=10940
https://www.microchip.com/wwwproducts/en/MCP23017

Thanks for all.

Andy2No, 2019-09-08:
It's not an intelligent device; it's a straight I2C (or on some versions SPI) to parallel interface.

The only thing that's LCD specific is a preset pot (trim pot) to adjust the contrast voltage.

That sort of interface is supported for the character based LCDs, (by including LiquidCrystal_I2C.h), by the LiquidCrystal library.

The version I've ended up trying, successfully, with character based LCDs, was:

https://bitbucket.org/fmalpartida/new-liquidcrystal/wiki/Home

See also:
 https://github.com/johnrickman/LiquidCrystal_I2C
 https://gitlab.com/tandembyte/liquidcrystal_i2c

olikraus, 2019-09-08:
> The version I've ended up trying, successfully, with character based LCDs, was:

Well, the initial request was for a graphical display not for a character LCD. So the links will not help.

Andy2No, 2019-09-08:
I thought you might find it useful because there is code in that library to use that i2c interface.

As far as I'm aware, the LCD is used in parallel interface mode, by setting up pins on the i2c interface pictured in the first post.  The ST7920 also has a parallel interface, so it ought to be possible, if your library supports one.

The difference is, instead of using I/O pins on the Arduino, directly connected to the parallel interface of the LCD, the i2c to parallel interface makes those connections.

Andy2No, 2019-09-08:
An i2c interface like that can be found on ebay as, for example, "IIC I2C TWI SPI Interface Board Module PCF8574T for Arduino 1602 LCD 2004 L I5P5"

The PCF8574T is an 8 bit I/O expander, so the LCD is apparently used in 4 bit parallel interface mode, with the other bits used for the control signals.

olikraus, 2019-09-08:
In general I am not a fan of these expanders. They will make the communication very slow. This might not be a problem on character LCDs but will have a significant impact on graphics LCDs. The other question is: Why shell I add a I2C expander to a parallel graphics LCD if there are graphics LCD with the same price which include I2C interface.

Andy2No, 2019-09-08:
I agree they make more sense with a character based LCD - and only if there's a good reason to save on the number of I/O pins on the Arduino.  It's convenient to have a few less wires between the two, especially when they're DuPont cables, that tend to pull out of sockets and breadboards too easily, but that's probably the only real benefit.

I2C seems a bit inconvenient to use in general, because of the low speed.

olikraus, 2019-09-08:
> I2C seems a bit inconvenient to use in general, because of the low speed.

This comes on top. For a 128x64 display we need to transfer 1024 Bytes. In the expander we need to transfer the byte and change the write trigger twice. Not sure about the MCP23017, but this will be 3 bytes + 1 address?
So let me assume 4 bytes to transfer 1 byte for the display, so speed will be reduced by a factor of 4.

BTW the ST7920 includes a SPI interface, so why not using SPI if the user wants to save some pins?

All in all, I am not sure why a port expander could help here..

Andy2No, 2019-09-08:
Agreed.  It couldn't help, and using the built in SPI would be better than fitting it with an external i2c interface.

I suppose the people making it thought that since character based ones with one of those expanders fitted sell well, a graphic LCD with one fitted should sell too...  It doesn't make a lot of sense though, no.

closed 2019-10-03 by olikraus

dpwe, 2022-03-10:
BTW, I bought one of these.  The ST7920 LCD board works fine in SPI mode as long as you tie PSB to ground (you can throw away the backpack board, although you'll have to provide your own pot to drive Vo).  I used this initializer: `U8G2_ST7920_128X64_2_HW_SPI u8g2(U8G2_R0, /* CS=*/ 10, /* reset=*/ 8);` 

For an Arduino Uno, I connected R/W = SPI MOSI to D11, and E = SPI CLK to D13.  

CS (connected to Arduino D10) is labeled RS on the module, and RST goes to Arduino D8, per the initialized.