olikraus/u8g2 #976: Unable to use SW I2C with other slaves

iMina, 2019-08-22:
I am trying to use the constructor U8G2_SSD1309_128X64_NONAME2_1_SW_I2C. However, calling Wire.begin(SDA, SCL) anywhere in the code renders the LCD unresponsive.
I know the library has its own Wire.begin(), but it doesn't seem to work with other slaves if I rely on the library initializing the Wire.
Here is a snippet:
```
#include <U8g2lib.h>
#include <Wire.h>
#include <MPU9250_asukiaaa.h>
MPU9250_asukiaaa mpu;
U8G2_SSD1309_128X64_NONAME2_1_SW_I2C u8g2(U8G2_R0, /* clock=*/27, /* data=*/25, /* reset=*/2);
void setup() {
   u8g2.begin();
   u8g2.firstPage();
   do {
      u8g2.setFont(u8g2_font_ncenB10_tr);
      u8g2.drawStr(10, 24, "Setup");
   } while (u8g2.nextPage());
   Wire.begin(25, 27);
   mpu.setWire(&Wire);
   mpu.beginAccel();
}
void loop() {
   u8g2.firstPage();
   do {
      u8g2.setFont(u8g2_font_ncenB10_tr);
      u8g2.drawStr(10, 24, "Loop");
   } while (u8g2.nextPage());
   mpu.accelUpdate();
   Serial.println(mpu.accelX());
   delay(1000);
}
```
In the case above, the screen displays the first page (Setup) but fails to change after `Wire.begin()`, and removing `Wire.begin()` fixes the LCD but disables the other slaves.
I have tried it with multiple sensors and multiple libraries, always the same problem.

olikraus, 2019-08-22:
Nice questions, probably this is still #1 in the questions here ;-)

https://github.com/olikraus/u8g2/blob/master/doc/faq.txt#L9

olikraus, 2019-08-22:
Let me say it like this:
...SW_I2C: Will not use Wire.begin()
...HW_I2C: Will use Wire. begin()

iMina, 2019-08-22:
Oh thank you ! How silly of me to not notice that !
The PCB is already designed so I think HW_I2C is the way to go. I just can't seem to get it to work.
What is the correct order that I should follow?
```
#include <Wire.h>
#include <U8g2lib.h>
#include <MPU9250_asukiaaa.h>
MPU9250_asukiaaa mpu;
U8G2_SSD1309_128X64_NONAME0_1_HW_I2C u8g2(U8G2_R0, /* reset=*/2, /* clock=*/27, /* data=*/25);
void setup() {
   Wire.begin(25, 27);
   u8g2.begin();
   u8g2.firstPage();
   do {
      u8g2.setFont(u8g2_font_ncenB10_tr);
      u8g2.drawStr(10, 24, "Setup");
   } while (u8g2.nextPage());
   mpu.setWire(&Wire);
   mpu.beginAccel();
}
void loop() {
   u8g2.firstPage();
   do {
      u8g2.setFont(u8g2_font_ncenB10_tr);
      u8g2.drawStr(10, 24, "Loop");
   } while (u8g2.nextPage());
   mpu.accelUpdate();
   Serial.println(mpu.accelX());
   delay(1000);
}
```
Doesn't seem to work for either the LCD or the sensor :/

olikraus, 2019-08-22:
Whether
```
U8G2_SSD1309_128X64_NONAME0_1h_HW_I2C u8g2(U8G2_R0, /* reset=*/2, /* clock=*/27, /* data=*/25);
```
will work, depends a lot on your PCB and your uC. The last two options will only work for ESP systems, because for all other systems the I2C pins are fixed.

Does your display work with SW I2C?

iMina, 2019-08-22:
I am in deed using a ESP32.
And yes, SW I2C works.

olikraus, 2019-08-22:
Then, I would expect, that
```
U8G2_SSD1309_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0, /* reset=*/2, /* clock=*/27, /* data=*/25);
```
also should work.
Let me check the code... The Wire.begin() command is here:
https://github.com/olikraus/u8g2/blob/master/cppsrc/U8x8lib.cpp#L1150

In fact U8g2 depends on a special extension of the Arduino HAL where Wire.begin has two more arguments, allowing to specify the pins for hardware I2C.

This will simply call this code:
https://github.com/espressif/arduino-esp32/blob/master/libraries/Wire/src/Wire.h#L70

So from my perspective everything fits together. 

Maybe there is an HW issue on your PCB. Did you include the I2C pull up resistors? U8g2 SW I2C does not require these resistors, but they are mandatory for hardware I2C to work.

iMina, 2019-08-23:
Yes I found the code for the Wire.begin() command, even tried to force the pins in all begin() function calls. But nothing worked.
I don't think it's a HW issue since everything works if tested separately.
And yes I included the pull up resistors.
Right now the only thing that seems to work with the LCD is the SW I2C, which can't be used with other slaves.

olikraus, 2019-08-23:
Maybe bus clock is too high.

iMina, 2019-08-24:
No, I have been playing with the bus clock ranging from 50k to 400k.
I also removed all other slaves and I am trying to get the LCD to work alone on HW I2C, but nothing seems to work.
Any ideas ?

olikraus, 2019-08-25:
So as a conclusion, you say that 
`U8G2_SSD1309_128X64_NONAME2_1_SW_I2C u8g2(U8G2_R0, /* clock=*/27, /* data=*/25, /* reset=*/2);`
will work, but any code with `Wire.begin(25, 27)`.
Did you test any other slave on this  Bus with `Wire.begin(25, 27)`? Or is it only the display not working?

iMina, 2019-08-25:
Yes, other slaves work on the same bus perfectly. Only the LCD not working.

olikraus, 2019-08-25:
Maybe your display needs a different HW SPI mode. On your local copy, can you change to some other mode than 0?
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1309.c#L146

Valid modes are from 0..3
Will any mode except 0 work with HW SPI?

iMina, 2019-08-25:
I can't use SPI. The bus has multiple I2C slaves already.

olikraus, 2019-08-25:
Oh, excause, my mistake. Then the only idea left is a malconfugured display. Do you have the datasheet of the display?

olikraus, 2019-08-25:
Are D1 and D2 pins connected? With an ohm meter and power disconnected you should measure 0 ohm between these pins (see datasheet of your display for this)

iMina, 2019-09-06:
Been a long time !
I have been experimenting a lot with the LCD and I decided to work with HW I2C because I am limited by the already implemented hardware.
Right now I have another question.
I added some print statements to the Wire.cpp file and noticed the following.
When I do not set the I2C address manually, the library communicates with address 60 (`0x3C`) by default. When I noticed that the display is not working, I made a loop that tries to `beginTransmission()` and `endTransmission()` with all possible I2C address (0 to 127) and got 0 error from 3 devices which are naturally the LCD itself and the two I2C sensors that I'm using. The tricky part is that it only detects addresses `0x0D, 0x48 & 0x68`,
I tried to use `u8g2.setI2CAddress(0x0D);`or `u8g2.setI2CAddress(13);` or `0x48` or `0x68` but they all do not work. And also, each time I use  `u8g2.setI2CAddress();` I notice that the library is using a totally different address to talk to the LCD. For example, setting the address to `0x0D` makes the library use address 6.
What am I missing ?

iMina, 2019-09-06:
Also, I found an old in which you said if I want address 0x3d, I should set it to 0x3d*2.
Here : https://github.com/olikraus/u8g2/issues/77#issuecomment-254758626
But in the current implementation, I see you don't use the *2 part, here : https://github.com/olikraus/u8g2/blob/c2a06c08dca7a70d5834480f1dcfb7ba3664057e/cppsrc/U8g2lib.h#L73
If the *2 is really needed, this might explain why it uses address 6 when I set it to 13.
Any ideas ??

olikraus, 2019-09-06:
Yes, you need to multiply the address by 2.

closed 2019-10-03 by olikraus