olikraus/u8g2 #979: U8g2 Menu System

olikraus, 2019-08-25:
Check https://github.com/neu-rah/ArduinoMenu
Maybe add examples
U8x8 output???

neu-rah, 2019-08-25:
there is an example for each, U8x8, U8G2, U8GLib and UCGLib

thank for your awesome library and for the time you dedicate to it and to the community

olikraus, 2019-08-25:
Thanks for your work on ArduinoMenu. I am really impressed. Looking forward to do some tests.
I think it shares a lot of ideas with my old m2tk lib.

olikraus, 2019-08-25:
[image]

Nice, works out of the box, but I guess I need to learn more...

neu-rah, 2019-08-27:
nice!
after my journey on menu UI i have some experience, learning with the errors also. I'm quite sure that AM code is not organised as yours, but we will see...

you can also reach me on https://gitter.im/ArduinoMenu/Lobby or by email, you're welcome.

olikraus, 2019-08-27:
:)

AlexFundorin, 2019-09-02:
Wow, guys. This can't be real. olikraus wasn't aware that ArduinoMenu existed, even though it's already in version 4 and AM5 is planned?
Wish you two collaborated and made AM5 with all those bells and whistles, like smooth animations, transitions, cursor shapes, items and titles aligning, different fonts per menu item and everything else.

dremeier, 2019-09-02:
btw. there is also another nice solution for Menu withe the U8g2 and U8x8 lib, called [TcMenu](https://www.thecoderscorner.com/products/arduino-libraries/tc-menu/tcmenu-plugins/u8g2-renderer-plugin/).

olikraus, 2019-09-02:
Is TcMenu available as an Arduino Lib?

AlexFundorin, 2019-09-02:
https://github.com/davetcc/tcMenuLib/
https://github.com/davetcc/tcMenu
https://www.thecoderscorner.com/products/arduino-libraries/tc-menu/

closed 2020-07-05 by olikraus