olikraus/u8g2 #980: unknown display - can anyone help please

Majestix1970, 2019-08-29:
Hi

I have the following display. 

[image]

[image]

Does anyone have an idea what contontroler it is using and how to connect/use it (pinning, u8g2 driver settig, etc.)

Best regards
Majestix

v1993, 2019-08-29:
I'll take a wild guess that it's probably LCD.

Majestix1970, 2019-08-29:
Thanks v1993...
so far your guess is right. it is a lcd.
Any idea which controller and about the connection/pinning?

v1993, 2019-08-29:
No idea about that, sorry. LCDs I own have 16 pins, not 17.

olikraus, 2019-08-29:
Difficult... I have no idea

v1993, 2019-08-29:
@Majestix1970 Sorry if I was too broad: I meant that it's probably alphanumeric LCD and not common-purpose LCD/OLED.

closed 2019-10-03 by olikraus

Majestix1970, 2024-10-04:
yes I do have this display... but not working yet

Majestix1970, 2024-10-04:
ok thanks
I will do
judt to be sure:

+5V to Pin 1 and Pin 14
GND to Pin 15
correct?

kolcocnc, 2024-10-04:
Yes

Majestix1970, 2024-10-04:
thanks will do

kolcocnc, 2024-10-05:
this is for display test  function show all pixels

Majestix1970, 2024-10-05:
Hello kolcocnc,
looks like you know what you are doing!!!
can you please give me more details about this LCD e.g. complete pinout and how to programm it?
Thank you very much!

[image]

kolcocnc, 2024-10-06:
Hello
this lcd is for linde forklift and is specially designed for them
I want to buy it from you if you don't use it
do you use skype

Majestix1970, 2024-10-06:
hm.. it lays around for many, many yesrs now.
I do not really need it. I found it in an old HW store and thought it was a normal LCD. It has some light scratches but seams to work thanks to your hint.

Where are you located?

kolcocnc, 2024-10-06:
i am from macedonia

Majestix1970, 2024-10-07:
Hi,
Ok... Munich to Macedonia 
No idea what shipping might cost.
What do you have in mind for the display?

kolcocnc, 2024-10-07:
I have LCD with  broken glass and I don't know if he has the same characters as yours
lcd is new or used

Majestix1970, 2024-10-07:
my display was used, as far as I know. As mentioned before, I baught it many, many years ago from an scrap electronic shop in Munich. I thought it is a generic LCD.

If you make a fair offer incl. shipping cost I am willing to sell it. But I can not give any garanty. If you have more details regarding how to communicate with the display, which controler etc. I might be able to send characters to it so you can see if it is what you need.

kolcocnc, 2024-10-07:
Pin 1 = Kontrast 0V -  +5V über 10K Poti
Pin 2 = D7
Pin 3 = D6
Pin 4 = D5
Pin 5 = D4
Pin 6 = D3
Pin 7 = D2
Pin 8 = D1
Pin 9 = D0
Pin 10 = E
Pin 11 = RW
Pin 12 = RS
Pin 13 = +5V DC
Pin 14 = +5V DC
Pin 15 = 0V Masse
Pin 16 = [LED] Anode
Pin 17 = [LED] Kathode
do you have arduino pcb
i send you hex file for test

Majestix1970, 2024-10-08:
thanks
I have all sorts of Arduinos and ESPs
could you send me the source code. Tvis would make things easier

kolcocnc, 2024-10-09:
Hi
I programing with BasvomAVR

Majestix1970, 2024-10-09:
C  code is C code... just send it please and I see if I can convert it to Arduino....

kolcocnc, 2024-10-11:
$regfile = "m8def.dat"
$crystal = 1000000

Config Lcd = 20 * 4

Config Lcdpin=Pin,Db4=Portb.3, Db5=Portb.2, Db6=Portb.1, Db7=Portb.0,E=Portb.4,Rs=Portb.5

Cursor Off
Cls

Lcd "*** HD44780  LCD ***"
Lowerline
Lcd "Line number 2"
Thirdline
Lcd "AaBbCcDdEeFfGgHfIiJj"
Fourthline
Lcd "1234567890"

End

Majestix1970, 2024-10-11:
ok gues that should tell me enought to programm it... just need to find the time

kolcocnc, 2024-10-12:
https://docs.arduino.cc/learn/electronics/lcd-displays/

Majestix1970, 2024-10-12:
did a short test... not workin
LiquidCrystal Lib from Adafruit did connect, but all I see is the fuul Test Screen with all Symbols/characters on. 
Also, I do not understand your "code"
The display has only 2 text lines with 11 characters, not 20x4 => see my screnn shot I made earlier....

I also tried to use 4 bit and 8 bit connection. Also no success to display "*Line1**OK*" and "*Line2**OK*" to fill the first 2 lines...

Are you sure we talk about the same display

Any idea?

Code I use:
#include "Wire.h"
#include "Adafruit_LiquidCrystal.h"

#define LCD_RS 12
#define LCD_EN 10
#define LCD_RW 11
#define LCD_D0  2
#define LCD_D1  3
#define LCD_D2  4
#define LCD_D3  5
#define LCD_D4  6
#define LCD_D5  7
#define LCD_D6  8
#define LCD_D7  9
Adafruit_LiquidCrystal lcd(LCD_RS, LCD_EN, LCD_RW, LCD_D0, LCD_D1, LCD_D2, LCD_D3, LCD_D4, LCD_D5, LCD_D6, LCD_D7);
//Adafruit_LiquidCrystal lcd(LCD_RS, LCD_EN, LCD_RW, LCD_D4, LCD_D5, LCD_D6, LCD_D7);
void setup() {
  Serial.begin(115200);
  Serial.println("LCD TEST");
  // set up the LCD's number of rows and columns: 
  if(lcd.begin(20, 4))
  Serial.println("LCD OK");
  else
  Serial.println("LCD not OK");

  // Print a message to the LCD.
  lcd.setCursor(0, 1);
  lcd.print("*Line1**OK*");
  lcd.setCursor(0, 2);
  lcd.print("*Line2**OK*");
}

void loop() {
  //just make LED on the arduino Mini blink....
  digitalWrite(13,HIGH);
  delay(250);
  digitalWrite(13,LOW);
  delay(750);
}

kolcocnc, 2024-10-12:
Hi
Pin1 is for display contrast 
Conect 10kohm potenciometer
[image]

Majestix1970, 2024-10-12:
I did already... with or without it, its not working...
Do you have a datasheet? If yes, please send.
I know it is a HD44780, but I assume to turn on/off the other symbols and amsl display numbers on the clock secmdnts it must be addressed specially. Or?

kolcocnc, 2024-10-14:
pin13 and pin14 ...+5v?

Majestix1970, 2024-10-15:
correct, both pins on 5V.

Do you have a datasheet?

kolcocnc, 2024-10-15:
i think it's this one

lcd-module.de
https://www.lcd-module.de › ...PDF
KS0073 34COM / 60SEG DRIVER & CONTROLLER FOR DOT MATRIX ...

kolcocnc, 2024-10-22:
Hi 
News????

Majestix1970, 2024-10-22:
unfortunately no
too much topics at work... have no time at the moment

kolcocnc, 2024-10-23:
Send me i test display