olikraus/u8g2 #981: [Font request] Problem with convert new font for Persian

idreamsi, 2019-09-01:
Hi @olikraus 
I want to create a new font for the library, I used the following command, **bdf** file created successfully. 
`otf2bdf.exe -r 72 -p 16 btitr.ttf -o btitr16.bdf`

But after using the below command the created file is empty!
`bdfconv -f 1 -m '32-128,$600-$6ff,$FB50-$FBB1,$FE70-$FEFF' -n btitr16 -o btitr16.c btitr16.bdf`

Unfortunately I don't have access to Linux for test.
OS: Windows 7 32bit
Fonts: [font.zip](https://github.com/olikraus/u8g2/files/3563499/font.zip)
Ref: [Here](https://github.com/olikraus/u8g2/issues/532#issuecomment-370211199)

Is it a problem with the font or the commands I used?
Thanks

olikraus, 2019-09-01:
If I remember correctly, there was an issue with the quotes on windows. You must use double quotes on windows:
https://github.com/olikraus/u8g2/blob/master/doc/faq.txt#L229

Also the $ will not work on all systems, maybe better use decimal numbers only, at least
`-m "32-65535"` should output everything. Maybe also add "-v" for debug output.

idreamsi, 2019-09-02:
The problem was solved.
Thank you very much.
only one question:
Do you need special tricks to convert fonts?
After converting the font I feel some letters are bent. Also some characters are not displayed well.
But the [four fonts](https://github.com/olikraus/u8g2/issues/532#issuecomment-381334507) you convert are more beautiful.

My font data after convert, Used syntax:
`bdfconv -f 1 -m "32-65535" -n btitr20 -o btitr20.c btitr20.bdf`
```
const uint8_t btitr20[6706] U8G2_FONT_SECTION("btitr20") = 
  " \0\4\4\5\5\4\6\6\34#\372\365\17\0\17\374\0\0\0\0\2` \6\0 H\2!\20\344"
  ")h\302\210\27\223\10\11\23l\304\12\0%\36+\246\327\206\240\21d\204\220\221\245\204\42\5\16\24)"
  "P\244\64\322\20\21Cb\220\10\0(\24f*\207\222\230!c\206\214\31B\277\231h\314\240A\2)"
  "\27f*\207B(\211\306\220\231\206\10\235\214\31Bd\314\220\61\202\0**\256i\10\327@\21C\304"
  "\220\20\61hD\10R\42B\10\42\61d\4\212\22\201\2\211\31\21b\20\11\63#\246\33\22p\30\0"
  "+\16\255\345\347Z\340\70{\260,p<\3,\17$)g\302\10\26B\204\4\21\22\6\0-\10M"
  "d\351\302\203\5.\11\204(h\302\210\25\0/\26(\246\247\232T\302\244\22&\225\60Q\243\204\211\22"
  "&\225\60\0\60\17\347\244\230N\250\61%N\224\31\25\6\0\61\15\4\352\207F\30\21#X\314D~"
  "\62\24)\346\267^\220P\42\304\254\300\304\14\261\331\311\273P\0\63\30\13&\330\206\230(\206\210x\220"
  "BE\32\22\202\6\22\234C\221\222\1\64\32+\346\307\332\30\21F\212\210(V\210I\232C\246F\16"
  "\234P\244\244\301\0\65\37*\346\307R\300q\304H\31\62tD\10\21\61CBU\22\206\210\220\7&"
  "\222\214\30\2\0\66\31\12\346\307\322(\63\202\204\4\24f\304\5\242B\244\210\215\223P`H\0\67\34"
  "\13&\310F\270 \242D\14\32Af\4\221\22\224 \62e\214\34\205\42e\5\0\70\36\13\246\307V"
  "\320\220\3\247#W\254\224!\22cHPB\202\314\241\21\203\206\4\23\1\0\71\27(\346\247\312(\62"
  "FL\10\31!d\304M\214\11\33\65\63q\1:\13d)h\302\210\325#V\0=\13\313\344\330\302"
  "\203\363\340\36\34[\13d*gB\232\370\377\6\1]\13d*g\2\232\370\377&\1{\25Dr\247"
  "N\10!a\242\21\23$\232@\221\10\11\23#\1}\21Dj\247B\240h\342Q\230 a\342I\64"
  "\0\253\24\7e\210N\220x\42B\210\10\61A\2\5\11\24$\0\267\7b\244I\202\1\273\25\6e"
  "\210B\220 \42\204D\42B\26ABH\21$\10\0\327\20\307d\230B\250\20a\242\10\25E\230("
  "\0\367\24\253\245\327\222@r\4\305\203xp\36\254@r\4E\1\0\0\0\10\350,\10*\377\377\2"
  "\331\10b\244I\202\1\6\14\20$)hN\20\21B\202\214`!\2\0\6\33\22\344)hN\20\21"
  "B\202\10\31\261r\304\12\0\6\37\27\346)\210\6\211$T\4\31\64*\10\221\221c\210\20\31\2\0"
  "\6!\21\7%\210\312\230\22\242\242(RbP\60\0\6\42\24\11f\230\306\220\25A\306\211\242g\63"
  "\23'](\0\6#\22$&X\206\210\60\206\36< !D\222 \0\6$\36h\346\226\216\250p"
  "\204F\205\7\61\212L\231 CD\14\261\246\324\240\61EL\30\1\6%\26\204\346VJ\220\7(\346"
  "B\210\230`\42$\11!\42\14\0\6&!\15\346\366\212\320\241A\2\15\31T\42\224\71\61!BI"
  "s\42P\252t\347^,Qe\10\0\6'\17\344%XN\20\21/\346\23IB\0\6(\31\316%"
  "\367nh!\1I\4\64i\362E\223U\347\1\22%+\14\0\6)\30\7&\230\212 QA\2\11"
  "\12.\250\210\211$e^\24!\2\0\6*\30ne\370\222\210\240\205\303\10\11)$\240I\223\17\234"
  "\254:\4\0\6+\32\256e\370Z\340\261b\213\206\10#$\244\220\200&M>p\262\352\20\0\6,"
  "\32\314\345\326J\22-\304\10\234\215\30\42\224\214!\23\250 \212\60j\216\0\6-\25\314\345\326J\22"
  "-\304\10\234!\15\13\242\10\243\346\10\0\6.\31L&\327ZP\202\343\301$\321B\214\300\31\322\260"
  " \212\60j\216\0\6/\24h%\230R\260Qd\14\221\10D\352\201\10\63D\0\6\60\31\350e\230"
  "N\260Q\343\1\4\33E\250\324\210@\244\36\210\60C\4\0\6\61\24h%\227V\260Q\243\210\211\13"
  "&j\14\21\23F\0\6\62\31\10&\227R\260Q\343\1\6\33Ej\230\270`\242\306\20\61a\4\0"
  "\6\63!\262%G\77\220\360 D\6\12\22R\210\220\20\301T\4[\250\362\204\70q\13\25\246\65\11"
  "\0\6\64*r&G{x\20\343\1\204\10^<<\224\341\201\204\14\24$\244\20!!\202\251\10\266"
  "P\345\11q\342\26*Lk\22\0\6\65%\324\345V\77\210\361\0L\13\31\31D\314H\21aH\4"
  "c\21lE\300$!\15\211\33\310Rij\263\0\6\66.t&W\77\200\360`\306\3\31\17&<"
  "\244\343\1\230\26\62\62\210\230\221\42\302\220\10\306\42\330\212\200IB\32\22\67\220\245\322\324f\1\6\67"
  "\34\356%\370V`\261Di;\244\220\20C\42\16\221\31\64\350A\220E\351J\1\6\70#\356%\370"
  "V`A\301\210\214\42B\210\214(\262C\12\11\61$\342\20\231A\203\36\4Y\224\256\24\0\6\71\32"
  ",\346\326\332\300b\207$\32\231\6\231\241Aa\6\16\244\260\272\65(\0\6: \254&\327\236P\221"
  "$E\16,vh\214 \241\243\302 \63\204\206\34\205\4\253[\203\2\0\6@\11\215 \310\302\17\16"
  "\6A\34\356e\370j\340\261\343\301\212%\31dD\260\60\42\202\241C\371\300\11#\64\0\6B\42+"
  "&\327^\210p\345B\204\7\64\216\234\10QCB\15\11\21\306D\230S\7\37\34IT\6\0\6C"
  "\32\321e(\373X\223X\32-\25\216T@\62\42\307\14\34\343\21\273T\0\6D\35\213\346\306fH"
  "\201\344\310\15\34\71p\226\42\302\211\10\67p \61\27IP\225\1\6E\26\311\345\246\322\250\62\42\210"
  "\14\21\361@\305\210P\342\344\31\0\6F\32\314%\327V\320\221C\303\10\35\21\214D\270\202$\311\271"
  "P\242\310\14\0\6G\20'%\230\312\230\42&\20=\60B\4\0\6H\30\250\345\226\316(\62e\202"
  "\14\21\61\304\232R\203\306\24\61a\4\0\6I\32\315\345\366\346\310r\346\304\204\10%\315\211@\251\322"
  "\235{\261D\225!\0\6J\32\315\345\366\346\310r\346\304\204\10%\315\211@\251\322\235{\261D\225!"
  "\0\6K\13\244 \14N\10\23e\0\6L\14\204 \14\212\220\21!\204\0\6M\13\244`\5N\10"
  "#d\0\6N\11D`\14\6\31\0\6O\13\204 \14\212\210!D\0\6P\13d\340\5\212\10\21"
  "a\0\6Q\12\204\340\13\312\10$\0\6R\12c$\14\202\210\22\0\6`\20\347\244\230N\250\61%"
  "N\224\31\25\6\0\6a\16\4\352\207F\30\21#X\314D~\6b\25)\346\267^\220P\42\304\254"
  "\300\304\14\261\331\311\273P\0\6c\31\13&\330\206\230(\206\210x\220BE\32\22\202\6\22\234C\221"
[... cut ...]
RLE Compress: Font code generation, selected glyphs=288, total glyphs=288
RLE Compress: ASCII gylphs=32, Unicode glyphs=256
RLE Compress: Glyphs per unicode lookup table entry=100
RLE Compress: Unicode lookup table len=2, written entries=2
RLE Compress: Unicode lookup table first entry: delta=8, encoding=59436
RLE Compress: Unicode lookup table last entry: delta=2090, encoding=65535
RLE Compress: 'A' pos = 0, 'a' pos = 0
RLE Compress: Font size 6706
bf_WriteU8G2CByFilename: Write file 'btitr20.c'
```

AnHardt, 2019-09-02:
Usually these problems don't come from `bdfconv` , converting one bitmap format to an other. Usually they come from `otf2bdf`, automatic converting and **scaling** the TTF-(vector)-font to a fixed size bitmap. The smaller the bitmap the more problems. 
The usual solution is editing the .bdf file in a font editor, like `FontForge`, to your taste. But from a 
certain smallness on you can't find a satisfying solution. For example 'Chinese' is impossible in 8x8.

olikraus, 2019-09-02:
There are several commandline options to otf2bdf which may work. 
https://github.com/olikraus/u8g2/blob/master/tools/font/otf2bdf/otf2bdf.c#L1233
Glyph hinting or auto hinting may help in otf2bdf (I think -a is an extention to my otf2bdf version). 

Like @AnHardt said, small point sizes are always difficult, but also higher point sizes might have issues. For example 16 may work, 17 produces bad results, 18 again works.
I suggest to install the freetype tool suite (https://github.com/genesi/freetype/tree/master/ft2demos) and use "ftview" tool to see under which conditions the conversion with otf2bdf produces a good result.

idreamsi, 2019-09-02:
Thank you both dear friends, very useful tips.
@olikraus The results with `-n` are not interesting and requires more tests and also `-a` not available.

closed 2019-10-03 by olikraus