olikraus/u8g2 #983: Alignment for text

idreamsi, 2019-09-02:
Dear @olikraus
I need to align words with a function. But I didn't find any specific function for this. I tested many ways to align words (Especially **UTF-8** words). The main problem is how to find the width of a word for example in font size 16 or etc. [As we know, changing the font changes the number of pixels used.]
Is there a way to auto alignment words?

Thanks for this great library.

olikraus, 2019-09-02:
There is getUTF8Width() function here: https://github.com/olikraus/u8g2/wiki/u8g2reference#getutf8width
It calculates the width of the provided string. If you pass the word itself, it will return the pixel width. Of course you need to set the font before calling this function.

Note: getUTF8Width() is not 100% correct, it might be wrong by 1 or 2 pixel.

idreamsi, 2019-09-03:
Thanks @olikraus 
I had seen this function on the wiki and I did a lot of testing but **getUTF8Width()** does not work right. by changing the font and size (UTF-8 font), the numbers are not accurate. For example, I use the **128x64** display and the **getUTF8Width()** output is 127, however, the text occupies only part of the screen width and part of the display is blank !

For example for the right-alignment I use the following code but it misses a few letters and gets much more error with changing font and size.
```
#define LCDWidth                          u8g2.getDisplayWidth()
#define ALINE_RIGHT(t)                  LCDWidth - u8g2.getUTF8Width(t)
```

idreamsi, 2019-09-03:
Problem solved with some changes.
Thank you for your attention. The following code can be used to alignment the UTF-8 letters (especially right-to-left languages like Arabic and Persian).
```
#define LCDWidth                        u8g2.getDisplayWidth()
#define ALINE_CENTER(t)                 ((LCDWidth - (u8g2.getUTF8Width(t))) / 2)
#define ALINE_RIGHT(t)                  (LCDWidth -  u8g2.getUTF8Width(t))
#define ALINE_LEFT                      0
```

closed 2019-09-03 by idreamsi

olikraus, 2019-09-03:
:)

NikilRagav, 2020-07-13:
Petition to add this functionality directly to the library

olikraus, 2020-07-13:
It should not take much effort to implement this by yourself.

NikilRagav, 2020-07-13:
sure - I can make a PR for it

ShivamJoker, 2021-09-01:
We could have an built api for align text centre then ?

olikraus, 2021-09-01:
There will be also a new drawing function which will allow centered text. 
Code is here: 
https://github.com/olikraus/u8g2/blob/master/csrc/u8g2_button.c

Not yet released or documented...