olikraus/u8g2 #985: Variable width vs Monospace

nunojpg, 2019-09-03:
Hi!

From the documentation I can't figure out why _tx_ is variable width, and _mx_ is fixed.

> u8g2_xxx_tx | Transparent gylphs with variable width
> u8g2_xxx_mx | Monospace/fixed width glyphs
> u8g2_xxx_hx | Glyphs with variable width and common height
> u8g2_xxx_8x | Monospace/fixed width glyphs in a 8x8 box

For example I wrote "abcdefg" both with u8g2_font_crox1c_mf and u8g2_font_crox1c_tf and the output is exactly the same.

Does _tx_ means it supports fonts with variable width, and that all variable width fonts also support transparency?

Thanks

AnHardt, 2019-09-03:
Documentation seems not to be consistent.
https://github.com/olikraus/u8g2/wiki/u8g2reference#setfont
vs
https://github.com/olikraus/u8g2/wiki/fntlistall#u8g2-font-names

Likely depends on the source font. If that was variable width, u8g2_xxx_tx gets  variable width, if not, not.
With some tricks you can make a (bad locking) fixed width font from a variable one source. Simply make all glyphs as wide as the widest one, maybe center. The other direction is much more difficult.

olikraus, 2019-09-03:
Indeed, probably I should update the docs.
Like @AnHardt  said: "m" will enforce the font to be monospaced (which might look ugly). On the other side, I had several fonts which had been designed to be monospaced, but actually did deviate a little bit. Especially the space often was smaller. 

"m" fonts will have a second property: Each glyph will have the same height. This means, each glyph in the "m" font will fit into the same bounding box. This makes the "m" font suitable for "none-transparent" font mode.

In fact there might be fonts in t and m variants, but the m variant will always require more space in flash mem  because of the background bounding box rectangle.

nunojpg, 2019-09-09:
Thanks for the info.

closed 2019-09-09 by nunojpg