olikraus/u8g2 #986: Unsupported Display/Controller (ST7528i)

NaxNir, 2019-09-04:
Hi olikraus, we are trying to get this display to work with an AVR chip and therefore need to get this working with the C version of your library. We have already tried the existing drivers in your library. Any suggestions how we could add this controller? We are still new to this and are learning a lot but need some help here. We need to use this display as it was the only one with the exact dimensions we require. We have linked the display datasheet and the display controller datasheet as well. Thank you for your time.

Display: [newhavendisplay.com/specs/NHD-C160100DiZ-FSW-FBW.pdf](http://www.newhavendisplay.com/specs/NHD-C160100DiZ-FSW-FBW.pdf)
Driver: [newhavendisplay.com/app_notes/ST7528.pdf](http://www.newhavendisplay.com/app_notes/ST7528.pdf)

olikraus, 2019-09-04:
Is there any example code for this display?

NaxNir, 2019-09-04:
> Is there any example code for this display?
It seems there is at the bottom of the display datasheet.

olikraus, 2019-09-07:
hmmm... I will be complicated and even impossible... The main problem is, that I do not have a display in my lab to do the testing and evaluation. I can create some code for it, but I need a lot of help from your side.

olikraus, 2019-09-07:
One more thing to consider: U8g2 will only support black/white mode. Grayscale is not supported by u8g2.

olikraus, 2019-09-07:
SPI: probably "u8x8_cad_001"
I2C: Probably "u8x8_cad_ssd13xx_i2c", maybe also the fast version will work

olikraus, 2019-09-07:
In the NHD datasheet, there is the following sub-sequence:
```
I2C_out(0x38);//external mode
I2C_out(0x75);//
I2C_out(0x97);//3frc, 45 pwm
```
The controller is switched to ext mode 1, but then a ext mode 0 command is sent. This does not make sense...

olikraus, 2019-09-07:
New constructor: `U8G2_ST7528_NHD_C160100_1_HW_I2C  u8g2(U8G2_R0);`

I have created beta release 2.27.0. Can you do some tests with this constructor? Will your display work?

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

olikraus, 2019-10-03:
any feedback?

olikraus, 2019-11-23:
:-(

Crinst, 2020-05-29:
Hello Olikraus, is there any progress on adding support for ST7528i? I have tried your library (latest rev. 2.28.6) with STM Nucleo-F401RE and NHD-C160100DiZ-FSW-FBW and it didn't work correctly. It "controlled" only left half of the screen and it showed wrong "things" on the screen (see attached [https://drive.google.com/file/d/1fsMRjgmUyCNt7ZJ4ayw-5-TDWN6dAogh/view?usp=sharing](url) ). So I tried another library ([https://github.com/muxall/ST7528i](url)) which worked flawlessly. I also attached main.c class so you can see my definitions.
So my question is whether you plan on finishing support for ST7528i. If you do then I could help you out with testing. 

Thank you in advance.

BTW I can also supply some reports from logic analyzer if you need so. 

`
/* USER CODE END Header */

/* Includes ------------------------------------------------------------------*/
#include "main.h"

/* Private includes ----------------------------------------------------------*/
/* USER CODE BEGIN Includes */
#include "st7528.h"
#include "ctype.h"
#include "font5x7.h"
#include "font7x10.h"
#include "string.h"
#include "math.h"

#include "u8g2.h"
/* USER CODE END Includes */

/* Private typedef -----------------------------------------------------------*/
/* USER CODE BEGIN PTD */

/* USER CODE END PTD */

/* Private define ------------------------------------------------------------*/
/* USER CODE BEGIN PD */
/* USER CODE END PD */

/* Private macro -------------------------------------------------------------*/
/* USER CODE BEGIN PM */

/* USER CODE END PM */

/* Private variables ---------------------------------------------------------*/
I2C_HandleTypeDef hi2c1;

UART_HandleTypeDef huart2;

/* USER CODE BEGIN PV */
static u8g2_t u8g2;
u8x8_t u8x8;

// ST7528 LCD
uint8_t i2cDeviceAddressOLED = (0x3F << 1);		// I2C OLED address eq. 0x78 without shift

/* USER CODE END PV */

/* Private function prototypes -----------------------------------------------*/
void SystemClock_Config(void);
static void MX_GPIO_Init(void);
static void MX_I2C1_Init(void);
static void MX_USART2_UART_Init(void);
/* USER CODE BEGIN PFP */

/* USER CODE END PFP */

/* Private user code ---------------------------------------------------------*/
/* USER CODE BEGIN 0 */
/* to send the data to the uart */
void send_uart(char *string) {
	uint16_t len = strlen(string);

	HAL_UART_Transmit(&huart2, (uint8_t*) string, len, 10); // transmit in blocking mode
}
[... 156 lines cut ...]
  /* USER CODE END 2 */

  /* Infinite loop */
  /* USER CODE BEGIN WHILE */
  while (1)
  {
	  send_uart("while start");
	  /*
		LCD_Rect(3, 17, 156, 32, 15); //line1
		LCD_Rect(3, 33, 156, 48, 0); //lne2 oblast chzbneho vzkresleni
		LCD_Rect(3, 49, 156, 64, 0);  //line3
		LCD_Rect(3, 65, 156, 80, 0); //line4
		LCD_PutStr(5, 4, "GPS:", &Font5x7);
		ST7528_Flush();*/
		send_uart("while end");
		HAL_Delay(500);*/

    /* USER CODE END WHILE */

    /* USER CODE BEGIN 3 */
  }
  /* USER CODE END 3 */
}

`

olikraus, 2020-06-05:
Thanks for your offer. I always like to add new displays to u8g2. However, at the moment, it is difficult to find time to work on u8g2. Especially the ST7528 would require a lot of testing and analysis and currently I do not have the time for this.

BTW: Which u8g2 code did you use the create the picture? What will be displayed with U8x8 hello world example?

Crinst, 2020-06-05:
Thank you for answer. I have used code from previous comment to create attached picture. For u8g2 Hello world example I have just uncommented line "u8g2_DrawStr(&u8g2, 10, 10, "Hello World!");" and commented "u8g2_DrawLine(&u8g2, 5, 5, 155, 6);". I also tried u8x8 and it also didn't work. You can see results here: [https://drive.google.com/drive/folders/1Ryw-X1T_PB5nIruRfAdsrVg4b-G5tRNu?usp=sharing](url). 
One thing I have noticed, when I tried to "correct" definitions for ST7528i that it was sending data incorrectly. Meaning It send I2C adress, then 0x00 and command. And after that LCD expected argument, but library send I2C adress, 0x00 and then argument, which might have broken communication. I have tried to adjust that It might worked, but there were still some errors with correct pixel adressation or something. I have uploaded changed files from u8g2 library (to GDrivearchive ) so you could check it out. 
BTW All test prints (pictures) were done with original library without modifications.

olikraus, 2020-06-05:
Thanks for the insight into i2c communication. It would definitely help.

olikraus, 2020-06-05:
Found this display with the same controller:
https://www.ebay.com/itm/2-LCD-Module-160x64-Graphic-Display-Parallel-SPI-Serial-I2C-w-Tutorial-Connector/291054712550
Maybe I will order this for further testing.

olikraus, 2020-06-14:
I have ordered the above mentioned display from ebay.

olikraus, 2020-10-28:
Added support for the ERC16064 with St7528 controller.
NHD C160100 not tested, but should also work.

closed 2020-10-28 by olikraus