olikraus/u8g2 #987: Poor SW Spi performance on NANO EVERY (atmega4809)

Simone73, 2019-09-05:
Hi, when using SW SPI constructors on a Arduino NANO EVERY (atmega4809) i've noticed very poor performance (flickering an slow screen redraws) compared with the same layout on a Arduino NANO V3 (atmega328p)

olikraus, 2019-09-05:
SW SPI is always a little bit slow. What is your exact constructor name? Can you use HW SPI?

Simone73, 2019-09-05:
The problem here is not to use SW OR HW SPI but the fact that SW on 4809 is significantly slower than on a 328 when the contrary is expected.

Simone73, 2019-09-05:
Btw that is my constructor:

`	#define CL					4
	#define DA					3
	#define CS					10
	#define DC					8
	#define RES					9

U8X8_ST7565_JLX12864_4W_SW_SPI display(CL, DA, CS, DC, RES);`

olikraus, 2019-09-05:
Nevertheless HW SPI would be faster

Simone73, 2019-09-05:
Obviously!

BTW, if you can fix SW SPI for 4809 i would appreciate it

olikraus, 2019-09-06:
The code is here: https://github.com/olikraus/u8g2/blob/master/cppsrc/U8x8lib.cpp#L358

What exactly shell I do to optimize SW SPI?

olikraus, 2019-09-06:
More specific, I could enable the SW SPI optimization for the Atmega4809 if I would know the __AVR_ARCH__ option and wehter this is compatible with the other AVR Architectures.

Simone73, 2019-09-06:
ATmega 4809 ARCH is xmega3

xmega3 == 103

The following works for me!

`#elif __AVR_ARCH__ == 4 || __AVR_ARCH__ == 5 || __AVR_ARCH__ == 51 || __AVR_ARCH__ == 6 || __AVR_ARCH__ == 103`

olikraus, 2019-09-06:
Ah, ok :)

Simone73, 2019-09-06:
i think you can include ARCH 100 to 103 at least,

also there are some other fixes needed:

```
#if (__AVR_ARCH__ < 100)
typedef enum {
	LOW = 0,
	HIGH = 1,
	CHANGE = 2,
	FALLING = 3,
	RISING = 4,
} PinStatus; 
#endif

```

and

```
	  if ( u8x8_GetPinIndex(u8x8, msg) < U8X8_PIN_OUTPUT_CNT )
	  {
	    digitalWrite(i, (PinStatus)arg_int);
	  }
	  else

```

to be compatible with arduino 1.8.9 new megaAVR core

Simone73, 2019-09-06:
do you think to update the library with those fixes ?

olikraus, 2019-09-06:
Good question. I think I need to test 1.8.9.

Simone73, 2019-09-06:
> Good question. I think I need to test 1.8.9.

The proble is the new megaAVR core, if you compile for 328 on 1.8.9 there are no problems but if you select NANO EVERY or arduino uno WiFi V2 (4809) you need those fixes

olikraus, 2019-09-06:
Sounds strange....

Simone73, 2019-09-06:
> Sounds strange....

Please verify what i say ;)

olikraus, 2019-09-06:
Hmm.. looks like a more general problem: https://github.com/arduino/ArduinoCore-API/issues/25

olikraus, 2019-09-06:
Correct check probably uses
```
#if (ARDUINO_API_VERSION >= 10000)
...
#endif
```

Simone73, 2019-09-07:
> Hmm.. looks like a more general problem: [arduino/ArduinoCore-API#25](https://github.com/arduino/ArduinoCore-API/issues/25)

Yep

Simone73, 2019-09-07:
```
#if (ARDUINO_API_VERSION >= 10000)
typedef enum {
	LOW = 0,
	HIGH = 1,
	CHANGE = 2,
	FALLING = 3,
	RISING = 4,
} PinStatus; 
#endif
```

do NOT work for me,

stick to :

```
#if (__AVR_ARCH__ < 100)
typedef enum {
	LOW = 0,
	HIGH = 1,
	CHANGE = 2,
	FALLING = 3,
	RISING = 4,
} PinStatus; 
#endif

```

olikraus, 2019-09-07:
Oh, that was just a hint for myself. You should stay with your solution, but for u8g2 I need to find a more generic solution which will work with all kind of processors.

Simone73, 2019-09-07:
> Oh, that was just a hint for myself. You should stay with your solution, but for u8g2 I need to find a more generic solution which will work with all kind of processors.

Sure. Hope to see a fix and a new stable release soon

Simone73, 2019-09-07:
Well, after a bit of research i found that ARDUINO_ARCH_MEGAAVR exists so:

```
#if not defined (ARDUINO_ARCH_MEGAAVR)
typedef enum {
	LOW = 0,
	HIGH = 1,
	CHANGE = 2,
	FALLING = 3,
	RISING = 4,
} PinStatus; 
#endif
```

can work better ;)

olikraus, 2019-09-07:
Maybe, but the pinstatus is not related to xmega.

egilkv, 2019-10-13:
Are you using Serial at the same time? There is a bug that seriously degrades performance when using serial on these devices. Not sure if the fix is officially released yet.

olikraus, 2019-11-21:
I did some tests today. For some reason, I do not have ArduinoCore-API installed. So checking for 103 (as mentioned above) is sufficient to improve SW SPI speed.

olikraus, 2019-11-21:
I have downloaded arduino linux64 1.8.10. 
How can I reproduce the bug with ArduinoCore API?

closed 2020-01-19 by olikraus