olikraus/u8g2 #988: LCD to PC

Bonnie12, 2019-09-06:
Hi guys, I have a machine which has an LCD panel with T6963c display controller. My objective is to show the display content from the LCD to PC (speed doesn't matter much). The restriction is I can only access to the data pins (20pins) on the machine. Therefore, I made a prototype for my testing purposes. I use an Arduino Mega to program the LCD and Arduino Due to extract the data from data pins. Everything works fine until I extracted the data. The data is in binary and I am not able to decipher it. 

Code for LCD display :
`#include <Arduino.h>
#include <U8g2lib.h>
#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

U8G2_T6963_240X128_F_8080 u8g2(U8G2_R0, 8, 9, 10, 11, 4, 5, 6, 7, /*enable/wr=*/ 17, /*cs/ce=*/ 14, /*dc=*/ 15, /*reset=*/ 16); // Connect RD with +5V, FS0 and FS1 with GND
void setup(void) {
  u8g2.begin();
}

void loop(void) {
  u8g2.clearBuffer();					// clear the internal memory
  u8g2.setFont(u8g2_font_ncenB08_tr);	// choose a suitable font
  u8g2.drawStr(0,10,"H");	// write something to the internal memory
  u8g2.sendBuffer();					// transfer internal memory to the display
}`

Code for extracting the data from LCD:
`
void setup() {
  // put your setup code here, to run once:
pinMode(26,INPUT); //LCDpin10
pinMode(28,INPUT); //LCDpin11
pinMode(30,INPUT); //LCDpin12
pinMode(32,INPUT); //LCDpin13
pinMode(34,INPUT); //LCDpin14
pinMode(36,INPUT); //LCDpin15
pinMode(38,INPUT); //LCDpin16
pinMode(40,INPUT); //LCDpin17

pinMode(4,INPUT); //LCDpin5
pinMode(3,INPUT); //LCDpin7
pinMode(2,INPUT); //LCDpin8
pinMode(5,INPUT); //LCDpin9

Serial.println("D7:D6:D5:D4:D3:D2:D1:D0  WR:RD:CD:CE");
Serial.begin(9600);
}

void loop() {
  // put your main code here, to run repeatedly:
digitalWrite(2,LOW);
__asm("nop");
digitalWrite(3,LOW);
digitalWrite(4,LOW);
__asm("nop");
__asm("nop");
__asm("nop");
Serial.print(digitalRead(40)); //D7
Serial.print(digitalRead(38)); //D6
Serial.print(digitalRead(36)); //D5
Serial.print(digitalRead(34)); //D4
Serial.print(digitalRead(32)); //D3
Serial.print(digitalRead(30)); //D2
Serial.print(digitalRead(28)); //D1
Serial.println(digitalRead(26)); //D0
digitalWrite(2,HIGH);
digitalWrite(3,HIGH);
digitalWrite(4,HIGH);
}`

Data extracted from serial monitor:
`10110010 //Auto Reset
[... 1385 lines cut ...]
00100100
00100100
00100100
00100100
00000000
00100000
00000110
00000000
00000000
00000100
00000000
00000000
00101000
00000000
00000000
00110000
00000001
00000000
10000000
10000000
00010000
00000000
00000000
10010000
`

olikraus, 2019-09-06:
I can see, you have started decode the data already. The meaning of the data is described in the T6963 manual. Personally I would observe the transition of the enable signal to sample the data lines. 

The data which is sent to the display by u8g2 is listed here: https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_t6963.c

Also note that u8g2 will put your display in graphics Mode. This means you will not see "H" as an ASCII code, but instead a bitmap pixel graphics which will look like a H depending on the selected font. Also note that the bitmap graphics has the dimensions 8x240 pixel. This means, because you have drawn the H at (0,10), you will see only the upper part of the H (lower to pixel rows will be missing) in that bitmap. 

The bitmap graphics is transfered here: https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_t6963.c#L97

tobiea27, 2019-09-07:
Just to clarify. 
You have an Arduino Board connected to a T6963c display. Now you want to display the shown content on a PC, for debugging, documentation or whatever reason. Is that right?

When this is what you want to achieve, I made something similar in a project with an OLED-display. I printed the internal buffer of the u8g2 library over the serial port. No need for an external device. 
To interpret this data, I copied the printed buffer into Mathematica and generated the picture.

I don't know if this will work for you, because a serial interface is needed. Maybe SW-serial over Pin 20?

This solution is not perfect, but worked for me.
When this could help you, I have to search for the code and will post it here.

Bonnie12, 2019-09-09:
> I can see, you have started to decode the data already. The meaning of the data is described in the T6963 manual. Personally I would observe the transition of the enable signal to sample the data lines.
> 
> The data which is sent to the display by u8g2 is listed here: https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_t6963.c
> 
> Also note that u8g2 will put your display in graphics mode. This means you will not see "H" as an ASCII code, but instead a bitmap pixel graphics which will look like a H depending on the selected font. Also note that the bitmap graphics has the dimensions 240x8 pixel. This means, because you have drawn the H at (0,10), you will see only the upper part of the H (lower to pixel rows will be missing) in that bitmap.
> 
> Other parts of the bitmap graphics will be sent later (until you get the full 240x128 bitmap)
> 
> The bitmap graphics is transfered here: https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_t6963.c#L97

Dear Oliver,
Thank you for your explanation.
I will try this and update my progress here 

Regards,
Bonnie

Bonnie12, 2019-09-09:
> Just to clarify.
> You have an Arduino Board connected to a T6963c display. Now you want to display the shown content on a PC, for debugging, documentation or whatever reason. Is that right?
>
> When [...]

Dear tobiea27,

I would be grateful if you post the code here for my reference.

Regards,
Bonnie

olikraus, 2019-09-09:
> I would be grateful if you post the code here for my reference.

At least a pointer to the internal memory is returned here:

https://github.com/olikraus/u8g2/wiki/u8g2reference#getbufferptr

tobiea27, 2019-09-10:
Okay, this is my quick and dirty solution to generate a image from the internal buffer:

The Arduino prints the internal buffer in a HEX-format. Also the number of Tiles in the X direction are needed.

**Arduino code:**
```C++
#include <U8g2lib.h>
#include <Wire.h>

U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE, /* clock=*/ 21, /* data=*/ 20);

void setup() {
  Serial.begin(115200);
}

void loop() {

  uint8_t * bufferPointer = u8g2.getBufferPtr();

  u8g2.clearBuffer();

  //Draw start

  u8g2.drawFrame(5,5,118,54);

  u8g2.setFont(u8g2_font_ncenB14_tr);
  u8g2.drawStr(20,38,"Test");

  u8g2.setFont(u8g2_font_unifont_t_symbols);
  u8g2.drawGlyph(80, 38, 0x2603);

  //Draw End

  Serial.print("\nBuffer width: ");
  Serial.print(u8g2.getBufferTileWidth());
  Serial.print(" Tiles \nBuffer height: ");
  Serial.print(u8g2.getBufferTileHeight());
  Serial.println(" Tiles");
  
  printBuffer(bufferPointer, 8 * u8g2.getBufferTileWidth() * u8g2.getBufferTileHeight());//Print the Buffer to serial console in on line as HEX-String

  u8g2.sendBuffer();

}

void printBuffer(uint8_t * buffer, int size)
{
  for(int i = 0; i < size; i++)
  {
    printHex(buffer[i]);
  }
  Serial.println();
}

void printHex(uint8_t number)
{
  char binChar[3];
  sprintf(binChar, "%02X",number);
  Serial.print(binChar);
}
```

**The Serial Console will show the following:**
```
Buffer width: 16 Tiles 
Buffer height: 8 Tiles
0000000000E02020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020E000000000000000000000FF0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000FF00000000000000000000FF0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000FF00000000000000000000FF00000000000000000000000000001F030101FFFFFF0101031F000080C0C060202060C0C0800080C0E020202040E0000020F0F8FE2020000000000000000000000000000000000000000011802400F50C578787570CF4018820040000000000000000000000000000000000000000000000000000FF00000000000000000000FF0000000000000000000000000000000000203F3F3F2000000000000F1F1F3222222223131B003913232327273E1E0C00001F3F3F202018000000000000000000000000000000000000000012103854838282AA828354381010010000000000000000000000000000000000000000000000000000FF00000000000000000000FF0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000FF00000000000000000000FF0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000FF00000000000000000000070404040404040404040404040404040404040404040404040404040404040404040404040404040404040404040404040404040404040404040404040404040404040404040404040404040404040404040404040404040404040404040404040404040404040404040404040404040404040404070000000000
```

Now the Mathematica script creates the image from the raw data. Because the script doesn't read the Serial data, you have to manually copy the number of X-Tiles and the buffer.
[... 42 lines cut ...]
tileBoolArray = 
  Function[x, Reverse@IntegerDigits[x, 2, 8]] /@ # & /@ tileIntArray;
tileBoolArrayTrans1 = Transpose[#] & /@ tileBoolArray;

tileBoolArrayTrans2 = Partition[tileBoolArrayTrans1, tilesX];

tileBoolArrayTrans3 = Transpose[#] & /@ tileBoolArrayTrans2;

bitList = Flatten@tileBoolArrayTrans3;

dataTable = Partition[bitList, 128];

Image[dataTable, ImageSize -> {128, 64}*4]
```

**The script will output this image:**

[image]

This solution isn't very elegant, but it worked for me. I think this could be easily transferred into a Python script in the future.

I hope this will help you to build a working solution for your project.

Best regards,
tobiea27

closed 2019-10-03 by olikraus

Vineeth-B007, 2021-12-27:
> Okay, this is my quick and dirty solution to generate a image from the internal buffer:
>
> The Arduino prints the internal buffer in a HEX-format. Also the number of Tiles in the X direction are [...]

Sir,
I have used the u8g2lib for one of my projects to print UNICODE in paper using thermal printer,I have the buffer with me but there needs some bit manipulation for the accessed array to print it in paper ,but when I refer to your mathematica script it could be helpful to format my array accordingly to print in paper.
but I'm not understanding the script is there any source code available in c/python. Or else is there any function in u8g2lib to align the buffer so that I can print it in paper .
Also I have used adafruit GFX library to print TTF in paper, adafruit GFX lib returns the buffer in paper printable format but u8g2lib needs bit manipulation .
any help would be grateful.
Thankyou