olikraus/u8g2 #989: I am confused with "u8x8_d_ssd1306_128x64_noname_init_seq"

puppies, 2019-09-07:
I am porting ssd1315 for u8g2, and there is something wrong with the display while the seller said it is compatible with ssd1306.

When i look around the source code, i find this.
`U8X8_CA(0x020, 0x000),		/* page addressing mode */`

I find '0x00' is for 'Horizontal Addressing Mode', and '0x02' is for 'Page Addressing Mode' in the datasheet. But here it is 0x00 for 'Page Addressing Mode'. 

Do i miss something about 'Addressing Mode' for ssd1306?

olikraus, 2019-09-07:
There is no difference for u8g2 regarding the page or horizontal addressing mode. They differ only in the way how a page end wrap around is handled. However the wrap around will never happen with u8g2, so it does not matter to use 0 or 2 for the addressing mode.

puppies, 2019-09-11:
> There is no difference for u8g2 regarding the page or horizontal addressing mode. They differ only in the way how a page end wrap around is handled. However the wrap around will never happen with u8g2, so it does not matter to use 0 or 2 for the addressing mode.

I see. Thanks.

closed 2019-09-11 by puppies