olikraus/u8g2 #991: STM32F103 Works on SW_SPI but not in HW_SPI

BSG-SalvoL, 2019-09-09:
I have an STM32F103 board with the STM32duino bootloader. Using U8G2_ST7920_128X64_F_SW_SPI constructor the display works fine, but becouse my project requires more FPS i tried to use U8G2_ST7920_128X64_F_HW_SPI. With this constructor the board freeze with the PC13 led blinking fast and nothing works.
Tried U8G2_ST7920_128X64_F_2ND_HW_SPI too and all other components works fine but not the LCD where nothing is displayed.
The display is connected on SPI2 interface on 
MOSI -> PB15
SCK -> PB13
SS -> PB12
I tried to connect to SPI1 too with no results.
Am i missing something or there is a problem selecting the right SPI?

olikraus, 2019-09-09:
HW_SPI will use SPI.h of your Board Environment. 
At least for Uno, calling the SPI functions works without problem. So my suggestion is to search for any incompatibilies of the SPI Arduino functions for your environment. 
I just rely on the fact that the same procedures behave in the same way across all boards.

BSG-SalvoL, 2019-09-09:
Thank you for your response, i've already using the other SPI port (this board has two) with more MCP23S17 and them work fine. 
Due to your response i'm trying to switch the ports for the components. 
It can be a good idea to pass a SPIClass instance to the constructor.

olikraus, 2019-09-09:
I just use the standard SPI object, which is defined by the Arduino Environment: https://github.com/olikraus/u8g2/blob/master/cppsrc/U8x8lib.cpp#L750

BSG-SalvoL, 2019-09-09:
Ok maybe i found what's the problem. 
For first for me SPI_INTERFACES_COUNT was not defined, so U8X8_HAVE_2ND_HW_SPI will not be defined by your library; i've actually defined manually in my code. 
Now SPI1, that is used in the library is of type spi_dev* for me which declaration is in file spi_f1.c while SPI according to arduino library is of type SPIClass.

olikraus, 2019-09-09:
Hmmm

BSG-SalvoL, 2019-09-09:
I've done another test. 
Defined a pointer in SPI.h and SPI.cpp
`extern SPIClass STMSPI1; //SPI.h
SPIClass STMSPI1(2); //SPI.cpp`

And modified in the function SPI1 with STMSPI1 and still not working.

olikraus, 2019-09-09:
Will u8g2 Work with SPI instead of SPI1?

BSG-SalvoL, 2019-09-09:
With the U8G2_ST7920_128X64_F_HW_SPI constructor the board freeze.. i've tried replacing STMSPI1 with SPI and still nothing appear, but the board is running.

BSG-SalvoL, 2019-09-09:
Ok, testing i've experienced that my definition of SPI_INTERFACES_COUNT not interfere with the library, then i define it in the library (U8x8lib.h). Now with U8G2_ST7920_128X64_F_2ND_HW_SPI the board is freezing.

BSG-SalvoL, 2019-09-09:
Defining SPI_INTERFACES_COUNT 2 in the SPI.h the board is not freezing, but i can see that the function:
`extern "C" uint8_t u8x8_byte_arduino_hw_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)`
is never called.
I'm really confused.

BSG-SalvoL, 2019-09-09:
The freeze come when SPI.Transfer function is called, but i've multiple MCP23S17 in the other SPI that are working all good. There is something wrong.

BSG-SalvoL, 2019-09-11:
I've tested using SPI.Transfer manually setting CS pin LOW and HIGH after sending random numbers.
The board doesn't freeze and i can see some dots on the screen. No way working with the u8g2 library right now.

olikraus, 2019-09-11:
Hm not sure how help here.

BSG-SalvoL, 2019-09-11:
Can you help me to write a specific function for my case? i need to understand what happens before this function Is called (function chain).. 
i see that this function handle specific messages type that i think i have to rewrite..

olikraus, 2019-09-11:
https://github.com/olikraus/u8g2/blob/master/cppsrc/U8x8lib.cpp#L773

U8x8lib.cpp is installed on your PC. You can updated the code and add some debug output in above mentioned location. Do you need to know any special about the code?

closed 2019-11-23 by olikraus