olikraus/u8g2 #992: How to drive this display, anyone can help?

ngmacha, 2019-09-10:


ngmacha, 2019-09-10:
Thank you very much, @olikraus , for the great library, I have quite a few project using it! Apologies in advance, however, if this is NOT the place to post this kind of questions - please kindly advise where, then!

This is the graphic LCD from the control panel of the salvaged HP Deskjet 3050 printer I'm trying to reuse.
Front:
[image]

and back:
[image]

I tried to search for CH376, no match... Seems there are 6 lines from the connector on the back coming to the LCD pins, maybe SPI (MOSI, MISO, SCLK, SS and power)? I've already found the power lines (7 is GND and 10 is Vcc, counting from the V key), but still hesitating to power it. The DS32 and DS33 seen on two sides of the LCD connector are backlights...

Anyone could advise how to findout the pinouts and how to use it with this great library? Any help is very much appreciated. Thanks a lot in advance!
Mac Ha

olikraus, 2019-09-10:
I have no idea...

olikraus, 2019-10-03:
no further comments... closing...

closed 2019-10-03 by olikraus