olikraus/u8g2 #996: Is there a command to clear part display?

markmccosker, 2019-09-15:
Hi Guys,

Anyone up for a challenge?

I have set up a Duinotech 164x64 Display

[](url) (https://www.jaycar.com.au/arduino-compatible-128x64-dot-matrix-lcd-display-module/p/XC4617

using the u8g2 library to display a menu  from another program (DCSBIOS Library).

The issue is that sometime the menu (void onUfcOptionDisplay1Change) item will be selected and a ":" (void onUfcOptionCueing1Change) will be placed before the menu item.

For example;

  abcdefg (not selected)
: abcdefg (selected).

Each of the voids run fine without the other but placing them together (as per below) I get the rest of the screen blanking out when there is a selection.

I understand that this is from the ClearBuffer and nextPage commands.

Yet, I hope someone can think of a workaround to keep the menu item on the screen while the selector changes.

``

#define DCSBIOS_IRQ_SERIAL

#include "DcsBios.h"

#include "U8glib.h"
#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R3, /* clock=*/ 18, /* data=*/ 16, /* CS=*/ 17, /* reset=*/ 13); // Feather HUZZAH ESP8266, E=clock=14, RW=data=13, RS=CS

void onUfcOptionCueing1Change(char* newValue) {
   u8g2.setCursor(0,0);
   u8g2.setFont(u8g2_font_samim_16_t_all);//u8g2.setFont(u8g_font_5x7);
   u8g2.clearBuffer();
   u8g2.drawStr(15, 25, newValue);
}
  DcsBios::StringBuffer<1> ufcOptionCueing1Buffer(0x5420, onUfcOptionCueing1Change);

void onUfcOptionDisplay1Change(char*newValue) {

    u8g2.setCursor(0,0);
    u8g2.setFont(u8g2_font_crox3cb_tr);//u8g2.setFont(u8g_font_5x7);
    u8g2.clearBuffer();
    u8g2.drawStr(5, 15, newValue);

}
  DcsBios::StringBuffer<4> ufcOptionDisplay1Buffer(0x542a, onUfcOptionDisplay1Change);

void setup() {
    u8g2.begin();
    u8g2.clear();
   
  DcsBios::setup();
}

void loop() {
   
 
    onUfcOptionCueing1Change;
    onUfcOptionDisplay1Change;
 
    u8g2.nextPage();          // transfer internal memory to the display
   
 
  DcsBios::loop();
``

Thanks in advance for any help/comments.

olikraus, 2019-09-15:
Not sure whether I understood the question, but if this helps: You can erase a rectangle with the DrawBox command, just use draw color 0.

markmccosker, 2019-09-15:
Hi olikraus,

Thanks I will try that tomorrow. After I set the colour and draw the box I guess I have to reset the colour. Is that correct?

olikraus, 2019-09-15:
Yes, back to 1.

markmccosker, 2019-09-16:
Hi Olikraus,

Thanks so much for your great idea. 
It worked perfectly.

I will close this now.

Regards
Mark

closed 2019-09-16 by markmccosker

olikraus, 2019-09-16:
:)