olikraus/u8g2 #997: Making tekst automatically go to next line using U8x8 library

SFChen-stack, 2019-09-17:
dear Olikraus,

For a project I'm trying to make connect an Arduino Mkr-Zero, with and Usb host mini V2 and a 5.5" 256x64 Green Graphic OLED Display Module-SPI. I also used your u8x8 library  with it, since this device can be used with those libraries. I was experimenting what is possible to programm. I was able to make the screen type the letters I want thanks to the library and I also tried to make extra functions like backspace etc. There was one function however where I struggled a bit to implement it. It is the function where if you type a word and you reach the end of the horizontal screen--> word will show totally on the next line 
[image]`

I thought of a way to implement the function:
- Make an array for string of bytes
- After spacebar is pressed save the distance between current cursor and end X distance of current line
- When serialdata is available again and it is a letterkey read it and add it to the array/buffer.
-if the string in an array is larger than differencelength of current position cursor and end of the screen
place the word in the next line
-otherwise place the word in the same line.

**This is how i wrote the code!
main code!**
```
#include <Arduino.h>
#include <SPI.h>
// V6 u8g8
#include <U8x8lib.h>

U8X8_SSD1322_NHD_256X64_4W_SW_SPI u8x8(/* clock=*/ 12, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9);
const int ArraySize = 15;
char incomingByte = 0; // for incoming serial data
long timer = 0;
char xdis = 0; //for location on display
char ydis = 0; //for location on display
char stringBuffer[ArraySize];
 
void setup(void) {

  Serial1.begin(9600); // opens serial port, sets data rate to 9600 bps
  pinMode(LED_BUILTIN, OUTPUT); //sets the led on the arduino as an output

  u8x8.begin(); //starts the library
  u8x8.setPowerSave(0);
  u8x8.setFlipMode(1); //flips the screen for right oriëntation
  u8x8.setFont(u8x8_font_pxplusibmcgathin_r); //this one was the best personally
  
  u8x8.drawString(0,0,"DuoCOM versie 2.0");      
  delay(2000);
  u8x8.clear(); //clears the screen
}

void loop(void) {
  timer++; //add 1 to the timer
  if (xdis > 31) { //the end of the screen  
     char new2_ydis = ydis;
        new2_ydis = ydis+1;
        u8x8.clearLine(new2_ydis);  // clear next line
    xdis = 0; //start on the left of screen
    ydis ++; //goes one row lower
  }
  else
  {  // stay in the same position
    xdis = xdis;
    ydis = ydis;
  }

  if (ydis > u8x8.getRows()) { //bottom of the screen
    ydis = 0; //back to the top of the screen
    u8x8.clearLine(0);
  }
  else{
  // stay in the same position
    xdis = xdis;
    ydis = ydis;
  }
  
  if (timer > 100000) { //keeps the timer on 1000000
    timer = 1000000;
  }  
  if (Serial1.available() > 0) { //serial input comes in
        digitalWrite(LED_BUILTIN, HIGH); //turn the LED on (HIGH is the voltage level)
    incomingByte = Serial1.read(); //reads the serial data and stores it into a buffer
    delay(50); //an delay for the LED
        digitalWrite(LED_BUILTIN, LOW); //turn the LED on (HIGH is the voltage level)
    if (timer > 100000) { //program can only go further with timer of this value to prevent double characters      
      if (incomingByte >= 33 ) { //print normal keys
        print_keys();
      }
      else if (incomingByte == 32) { //adds a space
        print_keys();
      }
      else if (incomingByte == 8) { //backspace
       backspace();        
      }
      else if (incomingByte == 27) { //clear screen function
        u8x8.clear();
        xdis = 0;
        ydis = 0;
      }
      else if (incomingByte == 13) { //enter function
        char new_ydis = ydis; //char for variable for enter
        new_ydis = ydis+1;
        u8x8.clearLine(new_ydis); // when enter clear the next line
        xdis = 0;
        ydis ++;
      }
      else if (incomingByte == 23) { //up arrow
        xdis = xdis;
        ydis = ydis -1;
        u8x8.setCursor(xdis, ydis);
      }
      else if (incomingByte == 19) { //down arrow
        xdis = xdis;
        ydis = ydis +1;
        u8x8.setCursor(xdis, ydis);
      }
       else if (incomingByte == 1) { //left arrow
        xdis = xdis -1;
        ydis = ydis ;
        u8x8.setCursor(xdis, ydis);
      }
       else if (incomingByte == 4) { //right arrow
        xdis = xdis +1;
        ydis = ydis ;
        u8x8.setCursor(xdis, ydis);
      }
      else {
         print_keys();
      }
      timer = 0; //reset the timer when actions are done
    }
  }
}
```
**function which i tried to implement**
```
void print_keys(void) {
char byteofString = 0; // char used for the array
char counter = 0; // used as a counter everytime data is available
char xEndDis = 31; //end distance of horizontal led screen
char xdisDiff = 0; // xdis difference
xdisDiff = xEndDis - xdis; // max value of screen minus current position of xdis cursor
if (Serial1.available() > 0 ) { //serial input comes in
  counter++;      
    byteofString = Serial1.read(); //reads the serial data and stores it into a buffer
    if (byteofString >= 33) { // when a normal key is pressed
      
      if (xdis > 31 && xdisDiff > counter) //when the size of the string is greater than the max length print string in new line
      {
        char new2_ydis = ydis;
          new2_ydis = ydis+1;
          u8x8.clearLine(new2_ydis); //clear the next line 
          stringBuffer[counter] = byteofString; // buffer with the string of bytes
          u8x8.write(stringBuffer, sizeof(stringBuffer)); //write the string
          xdis ++; //cursor goes to the next spot
      }
      else //stay in the same line 
      {
          stringBuffer[counter] = byteofString; // buffer with the string of bytes
          u8x8.write(stringBuffer, sizeof(stringBuffer)); //write the string
          xdis ++; //cursor goes to the next spot
      }
    }
   else
    {
      xdis ++; // just put cursor on next position
    }
```
The result is still that i can't get anything on the screen(I changed u8x8 to Serial1 in my original code). I also changed my code again of this function and I only get -1 and 0 now. Maybe it is possible to do this simpeler. So my question is  mainly do you maybe have an idea how I can implement my intended tekstflow(see attachment) or maybe some tips how I can do it?
thanks in advance.

kind regards

chen

olikraus, 2019-09-18:
I think this is not a real U8g2 issue, more a general programming question.
The answer to your question also depends on the exact specification of the problem:
You want line breaks, but do you also need upward scrolling, if the last line is filled?
If so, the best thing would be the introduction of a character buffer of the same size as the U8x8 display (I mean, if you work with a 8x8 font, then on a 128x64 display you can put 16x8 chars, so your buffer needs to be 16x8).
The you do all the line wrapping and scrolling on this buffer only. If you have modified something you would copy everything to u8x8.

This would be my suggested solution.

SFChen-stack, 2019-09-19:
Dear Olikraus,

For now I don't want to scroll it upward, maybe I will try that out in the future. The way I made it now is, that when yhe last line is filled, it will automatically put the cursor on the first line after clearing the first line. So I can type again from there.
But ok you suggest making a multidimensional array, I never thought about that, I always thought that making an array in one dimension was good enough since it just prints some data from a keyboard buttons. I will try your suggestion and see what I can do with it and see what the result is.
I will let you know the result.

SFChen-stack, 2019-09-23:
Dear Olikraus,

I am still busy with your suggestion about making a 2D array to make a line break function, but I was wondering, is it possible to achieve the same thing when I try to write  something in a 1D array( which means it becomes a string of bytes). And that I will use u8x8.drawString or u8x8.print to get something like this. 
with print:
u8x8.setCursor(0, 1);
u8x8.print(array[]); 

or with drawString
u8x8.drawString(0, 1, array[]);

Or do you think that won't work?

olikraus, 2019-09-23:
Sure, a array of string will also work (I mean, there is no difference between 2D array of chars and a 1D array of strings ;-)

SFChen-stack, 2019-09-24:
Heey olikraus,

I wanted to thank you again for the idea of using a 2d array. the automatic linebreak is done now!!

kind regards,
Leon

closed 2019-10-03 by olikraus