olikraus/u8g2 #998: ESP32 hardware SPI and ST7920

Saur0o0n, 2019-09-19:
Hi,
I'm kind of out of other ideas how to solve this. I use 12864B LCD with ST7920 controller and ESP32-Wrover. Software SPI display is just fine, with hardware I have screen glitches. First I thought that's because ESP32 is 3,3v and LCD requires 5v logic - but I don't thinks so any more, since software SPI is ok and hardware SPI does not look like just some random errors with transfer, the same patterns stays over even after full screen refresh and are kind of multiplied. Also first screen initialization also looks ok - next  u8g2.clearBuffer()/u8g2.sendBuffer() creates glitches.

[image]
Red frame shows glitches that are persistent between screens.

LCD is conneted to one of three ESP SPIs - this one is called VSPI (MOSI-23, MISO-19, CLK-18, CS-5)
```
#define LCD_RESET 4   // RST on LCD
#define LCD_CS 5      // RS on LCD
#define LCD_CLOCK 18  // E on LCD
#define LCD_DATA 23   // R/W on LCD

//U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R2, /* clock=*/ LCD_CLOCK, /* data=*/ LCD_DATA, /* CS=*/ LCD_CS, /* reset=*/ LCD_RESET);
U8G2_ST7920_128X64_F_HW_SPI u8g2(U8G2_R2, /* CS=*/ LCD_CS, /* reset=*/ LCD_RESET);
```
Any ideas/hints?

I can live with software SPI, but it buggers me :) (other HW SPI is used for temperature readout).

olikraus, 2019-09-20:
The SPI timing is very critical with the ST7920, I still not sure whether the timing parameters are correct.
You could try to adjust the timing parameters on your local copy of the code:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_st7920.c#L185

Candidates might be pre and post_chip_enable_wait_ns and also sck_clock_hz

Saur0o0n, 2019-09-20:
It seems like lowering the clock by 5% fixes the issue for me. I'll have to test it a bit more - but it looks ok. Changing post_chip_enable_wait_ns doesn't seems to change anything in my case.
Do you want me to test something else? And big thx!

olikraus, 2019-09-20:
I am happy if it works for you. 
There is also a command to change the clock frequency, so for this you do not need to change the timing structure.

Saur0o0n, 2019-09-20:
Yes, u8g2.setBusClock(950000); also works just fine. Closing and thanks again!

closed 2019-09-20 by Saur0o0n