olikraus/u8g2 #1000: UI button presses not recognized

muelli01, 2019-09-20:
Hello there,

I have a strange problem with u8g2 using the menu functions.
Button presses besides the Home/Cancel button are not recognized.

Short sample code:

```
#include <U8g2lib.h>
U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R0, 13, 12, 10);
void setup() {
        u8g2.begin(/*Select=*/ 3, /*Right/Next=*/ 11, /*Left/Prev=*/ 6, /*Up=*/ U8X8_PIN_NONE, /*Down=*/ U8X8_PIN_NONE, /*Home/Cancel=*/ 8);
        Serial.begin(9600);
}

void loop() {
    uint8_t r;
    u8g2.setFont(u8g2_font_6x10_tf);
    u8g2.setFontRefHeightAll();
    r=u8g2.userInterfaceMessage("Title1", "Title2", "Title3", " Ok \n Cancel ");
    Serial.println(r);
    Serial.println(u8g2.getMenuEvent());
}
```

Besides the Home/Cancel button, which correctly outputs 0 to the serial monitor, no buttons are recognized. If I do a digitalRead on these buttons, they are correctly identified as active low.
Any advice?

Thanks
Stefan

muelli01, 2019-09-21:
The problem seems to have been caused by a faulty touch sensor. I replaced it and now the error has disappeared.

closed 2019-09-21 by muelli01

v1993, 2019-09-21:
Issue #1000 was solved by replacing faulty device. This is perfection as is!