olikraus/u8g2 #1001: Adafruit Feather M0 doesn't work with Sh1107

yahya2018, 2019-09-20:
I'm struggling to get the Adafruit Feather M0 Bluefruit LE work with Sh1107.  I tried different U8g2 versions (using Arduino 1.8.10  ) and all listed settings of the Sh1107 with SPI, but nothing worked 

Please Help me

below is the code that i used

#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

 U8G2_SH1107_64X128_1_4W_SW_SPI u8g2(U8G2_R0, 24, 23, 1, 0 , 22);
// U8G2_SH1107_64X128_2_4W_SW_SPI u8g2(rotation, clock, data, cs, dc [, reset])

//U8G2_SH1107_64X128_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
//U8G2_SH1107_128X128_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
//U8G2_SH1107_128X128_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ 8);
//U8G2_SH1107_PIMORONI_128X128_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ 8);
//U8G2_SH1107_SEEED_128X128_1_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);
//U8G2_SH1107_SEEED_128X128_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
//U8G2_SH1107_SEEED_96X96_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);

//U8G2_SSD1306_128X32_UNIVISION_1_SW_I2C u8g2(U8G2_R0, /* clock=*/ 21, /* data=*/ 20, /* reset=*/ U8X8_PIN_NONE);   // Adafruit Feather M0 Basic Proto + FeatherWing OLED
//U8G2_SSD1306_128X32_UNIVISION_1_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);   // Adafruit Feather ESP8266/32u4 Boards + FeatherWing OLED
//U8G2_SSD1306_128X32_UNIVISION_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);   // Adafruit ESP8266/32u4/ARM Boards + FeatherWing OLED
//U8G2_SSD1306_128X32_UNIVISION_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE, /* clock=*/ SCL, /* data=*/ SDA);   // pin remapping with ESP8266 HW I2C
//U8G2_SSD1306_128X32_WINSTAR_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE, /* clock=*/ SCL, /* data=*/ SDA);   // pin remapping with ESP8266 HW I2C
//U8G2_SSD1306_64X48_ER_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);   // EastRising 0.66" OLED breakout board, Uno: A4=SDA, A5=SCL, 5V powered
//U8G2_SSD1306_48X64_WINSTAR_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);   
//U8G2_SSD1306_64X32_NONAME_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
//U8G2_SSD1306_64X32_1F_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
//U8G2_SSD1306_96X16_ER_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);   // EastRising 0.69" OLED 

void setup(void) {

 
  u8g2.begin();  
}

void loop(void) {
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB10_tr);
    u8g2.drawStr(0,24,"Hello World!");
  } while ( u8g2.nextPage() );
  //delay(1000);
}

olikraus, 2019-09-21:
Is the wiring correct? Maybe sent a picture of your setup so that we can verify the wiring.

yahya2018, 2019-09-21:
OLED---> (PIN)
-----------------------------------
SCK ----> SCK (24)
SDA ----> MOSI (23)
RES----> MISO (22)
DC----> RX (0)
CS ----> TX (1)

yahya2018, 2019-09-21:
[image]
[image]
[image]

olikraus, 2019-09-21:
Here are my ideas:
- Using the TX/RX pins is a little bit critical with Arduino. 
- Using the MISO pin of your board as reset output is also critical, but it should work as long as you use SW SPI
- The pin header of your board is not soldered. This is probably the biggest issue here.
- The solderless breadboard is another great source of issues. You never know whether the connection is reliable. 

I personally started to solder everything. Or at least use female pin headers with dupont wires like here:

[image]

yahya2018, 2019-09-21:
Thank you.
i will do what you recommended and let you know if anything changes

yahya2018, 2019-09-25:
Thank you very much
It did work 
[image]

closed 2019-09-25 by yahya2018