olikraus/u8g2 #1002: reserved identifier violation

elfring, 2019-09-23:
I would like to point out that identifiers like “[`_U8G2_H`](https://github.com/olikraus/u8g2/blob/eba312fb66f63009a2c247af210c62b2dedd1a31/csrc/u8g2.h#L58 "Update candidate")” and “[`_U8X8LIB_HH`](https://github.com/olikraus/u8g2/blob/eba312fb66f63009a2c247af210c62b2dedd1a31/cppsrc/U8x8lib.h#L39 "Another update candidate")” [do eventually not fit](https://www.securecoding.cert.org/confluence/display/cplusplus/DCL51-CPP.+Do+not+declare+or+define+a+reserved+identifier#DCL51-CPP.Donotdeclareordefineareservedidentifier-NoncompliantCodeExample%28HeaderGuard%29 "Do not declare an identifier which is reserved for the compiler implementation.") to the expected naming convention of the C++ language standard.
Would you like to adjust your selection for unique names?

olikraus, 2019-10-03:
Implemented, thanks!

closed 2019-10-03 by olikraus

elfring, 2019-10-03:
Thanks for [your adjustment of affected include guards](https://github.com/olikraus/u8g2/commit/d6c129917d6f0355f18df2334b4c794c55f19719 "Renaming of a few include guards").