olikraus/u8g2 #1004: TTGO Lora ESP32 - u8g2 string functions only work with full frame buffer

skippa, 2019-09-24:
This is kinda odd - I'm using the TTGO Lora ESP32 module and I can use U8X8 easily with HW constructor `U8X8_SSD1306_128X64_NONAME_HW_I2C`, however when I use u8g2 constructor `U8G2_SSD1306_128X64_NONAME_1_SW_I2C` (or the HW constructor) I can only use drawing functions like `drawLine`, `drawTriangle` etc but not `drawStr()` or `print()` (with UTF8Print enabled of course).  I've done all the usual stuff like `setFont`, `setCursor` (for when using `print()`) but it doesn't seem to matter.

However, if I use the full frame buffer constructor `U8G2_SSD1306_128X64_NONAME_F_SW_I2C` the string functions seem to work fine...  I've used this library with ESP32 devkit many times without this drama.

Any ideas why this is?

olikraus, 2019-09-24:
Strange... Some code would be useful.

olikraus, 2019-10-03:
no further comment... closing...

closed 2019-10-03 by olikraus