olikraus/u8g2 #1005: label:question Is this posible to inverse part of buffer memory?

Saur0o0n, 2019-09-25:
Hi,
I want to do a progress bar (it's not exactly bar - but so we could understand easily let's assume that). It's hard to inverse only part of the letter for example (or line). Is this/would be possible to write proper information to buffer without sending it and then xor it somehow partially - let say rectangle x,y,w,h ?

olikraus, 2019-09-25:
There are probably many options to do this. 
If you just require XOR, then use draw color 2: https://github.com/olikraus/u8g2/wiki/u8g2reference#setdrawcolor

If you want to draw a progress bar while bypassing the sendBuffer/first-next loop, you could user the direct buffer interface or fallback to the u8x8 functions (yes you can have a u8g2 constructor and still use u8x8 c calls).

However note: you can not invert the a pixel on the display itself.

Saur0o0n, 2019-09-25:
I know I can't invert pixel on display - but until I send "u8g2.sendBuffer();" - everything is in memory right?
My idea was that I can draw anything in that memory, and later specify box that will invert pixels in this area (assuming it's b/w display - this sounds relatively easy in theory).

I'm using setdrawcolor() a lot - but this has to be used before drawing anything. And like I said - I would like to have half a letter changed (or 1/3) - something like in this example:
[image]

btw. There was (unless it looks like that) functionality to do the screenshot from uglib - is there any way to dump the screen on disk/memory in u2g2?

olikraus, 2019-09-25:
> I know I can't invert pixel on display - but until I send "u8g2.sendBuffer();" - everything is in memory right?

Yes

If you use sendBuffer() then I assume you have a full buffer constructor `_F_`.
In this case it will be very easy. set the draw color to two and just extend the bar with one or more vertical lines. The trick is not to clear the buffer, but just resend the content
Pseudo code:

clearBuffer
draw your picture
sendBuffer
while in progress
  draw with color 2 & sendBuffer

> btw. There was (unless it looks like that) functionality to do the screenshot from uglib - is there any > way to dump the screen on disk/memory in u2g2?

What exactly do you need.
There is bitmap device, which can be attached to the existing device:
https://github.com/olikraus/u8g2/blob/master/sys/bitmap/common/u8x8_d_bitmap.c#L283

The other option is, to dump the memory manually on request. 
This is the address of the internal buffer: https://github.com/olikraus/u8g2/wiki/u8g2reference#getbufferptr

Saur0o0n, 2019-09-25:
setdrawcolor(2) works great - actually when I re-read the description - all is there, but somehow I did not thought about it this way.
This is what I wanted to do - temperature graph and progress on it.
[image]

About bitmaps - I want to do screenshots to documentation. But I guess this is it - when you pointed bitmap device:
https://github.com/olikraus/u8g2/blob/master/sys/bitmap/hello_world/main.c

Saur0o0n, 2019-10-01:
I must revive this topic. Is this possible to use u8x8_SaveBitmapTGA on Arduino IDE? This is just part of the clib 
`
src/clib/u8x8.h:void u8x8_SaveBitmapTGA(u8x8_t *u8x8, const char *filename);
`
and in example it actually just writes in memory all data without displaying it.

I was looking for something that will allow me on actual hardware create screenshot. For example set the pin/button on Atmel/esp32, when shorted it will invoke the function that will dump memory as a file. So this would require to know address and size, and how to later display it on PC.

There is u8g2.getBufferPtr() and u8g2.getBufferSize() - any hint what to do with those data later on PC?

olikraus, 2019-10-02:
There are two fundamental approaches to do a screenshot
 - Place a device in parallel or in a chain with the existing constructor which writes to a bitmap. This will also work with u8x8 and is the method used by u8x8_SaveBitmapTGA
- Use getBufferPtr(), extract the page and reconstruct a picture from it. This will only work for u8g2 and you need to alter your program, but all in all it is probably more simpler.

Memory structure is here:
https://github.com/olikraus/u8g2/wiki/u8g2reference#memory-structure-for-controller-with-u8x8-support

Upper left pixel is tile 0, byte 0, bit 0.

Saur0o0n, 2019-10-02:
So I've tried to go with getBufferPtr() and save it as Netpbm (is as easy as it can be). But I'm missing something...
```
#include <U8g2lib.h>
#include <SPI.h>

U8G2_ST7920_128X64_F_HW_SPI u8g2(U8G2_R2, 5, 4);

void setup() {

  Serial.begin(115200);

  u8g2.begin();
  u8g2.setBusClock(900000);
  
  u8g2.clearBuffer();
  u8g2.drawVLine(0,0,64);
  u8g2.sendBuffer();   // just to see what I'm doing
  
  u8g2.clearBuffer();
  u8g2.drawVLine(0,0,64);
  
  uint8_t *addr=u8g2.getBufferPtr();
  for(int a=0; a<9; a++) Serial.printf("%x ",addr[a]);
}
```
This is shortest possible code to illustrate my approach. Assuming I've drawn a vline from 0,0 (uper,left corner). Bytes in memory should be FF 0 0 0 0 0... 0 right? All bits in first byte are 1 - rest is 0 - this is what I understand from memory structure description.
But the output is
`
0 0 0 0 0 0 0 0 0
`
If to print more it shows a pattern 1 (dec) every 17th byte.
```
8g2.getBufferTileHeight(8) * u8g2.getBufferTileWidth(16)
P1
128 64
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 1 0
```

olikraus, 2019-10-02:
As I mentioned earlier, the memory structure is a little bit more complicated. I mean, it does not match the PBM structure.

BTW: Should you use P4?

I also started to work on this problem...

Saur0o0n, 2019-10-02:
It's ESP32 - P4/P5 are fine. I've also tested this on Atmega (thinking it's something with esp) but results are the same.
From the memory description I've understood that each bit is a single pixel (on/off) and the only difference is they are grouped in 8 pixels/bits representing screen row, not in lines (like PBM).
So each flowing byte has pixels like
byte: (x,y)
0: (0,0) (0,1) (0,2) (0,3) (0,4) (0,5) (0,6) (0,7)
1: (1,0) (1,1) (1,2) (1,3) (1,4) (1,5) (1,6) (1,7)
... until the end of the screen width
127: (127,0) (127,1).... (127,7)
128: (0,8) (0,9) (0,10) (0,11) (0,12) (0,13) (0,14) (0,15)

And they are also grouped in tiles of width 8 or 16 - depending on constructor.

olikraus, 2019-10-02:
Hmm.. P4 might not be good for serial out. I will generate a P1 pic: I wrote a function

`void u8g2_WritePBM(u8g2_t *u8g2, void (*out)(const char *s));`

You have to provide a callback function "out" which has to write the data to the file. It may look like this:
```
void out(const char *s)
{
  printf("%s", s);
}
```

olikraus, 2019-10-02:
This is an SDL example. By pressing "c" it will output the P1 image: https://github.com/olikraus/u8g2/blob/master/sys/sdl/text_full_buffer/main.c#L102

Not sure whether this helps, but it will look like this an and should be a proper P1 image:
```
P1
128
64
1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 1 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 1 1 1 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 1 1 1 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 1 1 1 1 1 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 1 1 1 1 1 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
[... cut ...]
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 1 
```

Saur0o0n, 2019-10-02:
Ohh so with P4 you where referring to PBM type not Pin4/Pin5 on controller. So for serial output definitely P1 :).
I'm going to test your changes - but since my C-Fu is much lower then Yours - it will get me some time :)

olikraus, 2019-10-02:
Added XBM also (which requires less bandwidth when sending via serial)

```
void u8g2_WriteXBM(u8g2_t *u8g2, void (*out)(const char *s));
```

olikraus, 2019-10-02:
> I'm going to test your changes 

you just need to get the latest version of code from the repo. Only the files in u8g2/csrc need to be updated.

I will create an arduino example also... maybe later today in the evening. I think I will add C++ Arduino calls also...

Saur0o0n, 2019-10-02:
Well, this is a bit awkward - it doesn't work for me. First I've updated also those files, and it did not compile.
```
U8g2lib.h
U8x8lib.cpp
U8x8lib.h
```
But when updating just clib - it works (I'm still talking about Arduino+ESP32). Anyway this code:
```
#include <U8g2lib.h>
#include <SPI.h>
U8G2_ST7920_128X64_F_HW_SPI u8g2(U8G2_R2, 5, 4);

void out(const char *s){  Serial.printf("%s", s);}

void setup() {
  Serial.begin(115200);
  u8g2.begin();
  u8g2.setBusClock(900000);
  
  u8g2.clearBuffer();
  u8g2.drawVLine(0,0,64);
  u8g2.sendBuffer();   // just to see what I'm doing
  
  u8g2.clearBuffer();
  u8g2.drawVLine(0,0,64);
  u8g2_WritePBM(u8g2.getU8g2(),out);
}
```
Produces this image (scaled up 400%) - so kind of what I've already had.
[image]
Your example is ok:
[image]

olikraus, 2019-10-02:
oh.... its an ST7920... puhhh... 
now I understand the difficulties.
Ok, it is like this: Each display might have a different memory architecture. U8g2 basically follows this architecture and puts all data in the uC RAM in the same memory format like the target display.
Good news is, that allmost all displays can be configured in such a way, that they use the same native memory format except: ST7920 (and some other rarely used displays).

I mean, if you would use `U8G2_SSD1306_NONAME_128X64_F_HW_SPI` the it should generate the correct picture...

But... I will look into this.

olikraus, 2019-10-02:
BTW, the code can be simplified:
```
  u8g2.clearBuffer();
  u8g2.drawVLine(0,0,64);
  u8g2_WritePBM(u8g2.getU8g2(),out);
  u8g2.sendBuffer();   // just to see what I'm doing  
```

Saur0o0n, 2019-10-02:
Knowing that this is correct image but wrong approach I can now find out memory layout for ST7920 to do the screenshots :)
I know this sample could be simplified - but since I was working (writing own procedure) on memory buffer directly I've preferred to have proper display first before modify it (and probably ruin) later.

olikraus, 2019-10-02:
I have create two more functions for ST7920:

```
void u8g2_WritePBM2(u8g2_t *u8g2, void (*out)(const char *s));
void u8g2_WriteXBM2(u8g2_t *u8g2, void (*out)(const char *s));
```

I can not test this at the moment. Are you able to test these new functions?

Saur0o0n, 2019-10-02:
u8g2_WritePBM2() produces empty screen - only zeros. Btw. you can write to PBM file without spaces.
```
  u8g2.clearBuffer();
  u8g2.drawVLine(0,0,64);
  u8g2.drawVLine(10,0,64);  // add n ew line to see if we do not miss something in beginning of the memory
  u8g2_WritePBM2(u8g2.getU8g2(),out);
  u8g2.sendBuffer();
```
u8g2_WriteXBM2() produces 
[image]
There are two lines now - so it can be the same as before.

Saur0o0n, 2019-10-02:
This seems to be working for me - it's horizontally swapped, but it's easier fix it in Gimp :)
```
void do_screenshot(){
  uint8_t *addr=u8g2.getBufferPtr();

  Serial.printf("P1\n%d %d",u8g2.getDisplayWidth(), u8g2.getDisplayHeight());
  for(int a=0; a< u8g2.getBufferTileWidth()*u8g2.getDisplayHeight() ; a++){
    if(a%16==0) Serial.println();  //just to make it visually pleasant - new line
    for(int b=7; b>=0; b--){
      Serial.printf("%c",(addr[a] & (1u << b) ? '1' : '0'));
    }
  }
}
```

olikraus, 2019-10-02:
> u8g2_WritePBM2() produces empty screen -

ok, thanks. I have to look into this more closely :-p

olikraus, 2019-10-03:
Documentation: https://github.com/olikraus/u8g2/wiki/u8g2reference#writebufferpbm

olikraus, 2019-10-03:
I have created beta 2.27.2 which includes the above mentioned commands...

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

Saur0o0n, 2019-10-03:
Both XBM2 i PBM2 works, but (this is not a problem) image is flipped vertically and horizontally:
[image]
This may be because I use constructor with 180deg rotation - U8G2_R2.

Nice trick with that Print class - it works with Serial, it sometimes :) work with ESPAsyncWebServer - but this is definitely problem of Webserver with it's implementation of Print. I would still rather prefer char *writeBufferPBM2(), then use allocated memory for some tasks and free() it after all - but that's me, and I'm C :).

olikraus, 2019-10-03:
> image is flipped vertically and horizontally

yes, this is your rotation...

> edit: No u8g2_WriteXBM2/PBM2 in Arduino now :(

Why not? You can get the "u8g2_t *" c-struct pointer with u8g2.getU8g2().

Saur0o0n, 2019-10-04:
Ohh.. you have changed the name from u8g2_WritePBM2 to writeBufferPBM2... so it's there, sorry :)

olikraus, 2019-10-04:
Ah, yes, goal was to be more consistent with sendBuffer and clearBuffer functions.

Saur0o0n, 2019-10-04:
Ok. So I've tested it in many different ways - all is working fine. Thank you! And I guess I can close this topic.

closed 2019-10-04 by Saur0o0n