olikraus/u8g2 #1007: SSD1322 256x64 OLED display only 240 pixels in X

bonarm, 2019-09-27:
I used 4W_HW_SPI. Itried u8g2 full buffer and page mode, drew strings and pictures, all only first 240 pixels can be displayed.  I doubted if my OLED is damaged.
But when I tried u8x8, it works fine. All 256 pixles can be displayed.
So there may be some bugs in u8g2 mode?

olikraus, 2019-09-27:
Did you activate 16 Bit mode?

https://github.com/olikraus/u8g2/blob/master/doc/faq.txt#L159

bonarm, 2019-09-27:
> Did you activate 16 Bit mode?
> 
> https://github.com/olikraus/u8g2/blob/master/doc/faq.txt#L159

Yeah! That's the reaseon. Now it works fine. Thanks a lot!
Sorry about it, cause I used to use 128x64 OLED without any problem.

olikraus, 2019-09-27:
> cause I used to use 128x64 OLED without any problem.

It is always a trade off... If I would activate 16 bit mode by default, then you wouldn't notice the problem, but for all users of smaller  displays  the code size would be bigger....

closed 2019-09-27 by olikraus