olikraus/u8g2 #1009: Compile error: ‘yield’ was not declared in this scope

ideal54, 2019-09-28:
I downloaded this library to use with an ST7920 128x64 LCD (12864B V2.0) connected to an Arduino. I've tried the HelloWorld example supplied and I've uncommented the following constructor line and set the  pins appropriately (I have the PSB pin hardwired to ground to force SPI mode):

' U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R0, 12, 11, 13, U8X8_PIN_NONE); '

When I compile I get the following error "home/pi/sketchbook/libraries/U8g2/U8x8lib.cpp:156:14: error: ‘yield’ was not declared in this scope
        yield();"

The error appears to be generated by the constructor line but I've not been able to resolve or can I find any constructive ideas in a Google search. I'm running the Arduino IDE on a raspberry Pi 4 Model B under Ubuntu which may be contributing to the problem.

ideal54, 2019-09-28:
Ok, I've confirmed this is an issue when using the Arduino IDE on a Raspberry Pi 4 Model B running Ubuntu. I installed the  Ardiuno IDE on my Windows 10 laptop and loaded the U8G2 library and it compiles and runs just fine.

Any ideas on what might be the cause? I do all my Arduino programming on the Raspberry Pi and this is the first library clash I've come across.

olikraus, 2019-09-28:
yield is just a normal function, which should be there as part of the Aruduino IDE: https://www.arduino.cc/en/Reference/SchedulerYield

I think you can just remove the call to yield() or you add an empty yield() function to your code:
```
void yield(void) { }
```

ideal54, 2019-09-29:
Thanks for the quick response.

I had tried initially commenting the yield() call in U8x8lib.cpp file before I requested this assistance, but it released a lot more errors so I figured that wasn't the answer. After your response I found a generic recommendation to fix this error by adding 'void yield(void);' to the Arduino.h files in the user/share/arduino folders. Having done this I came up with the same errors that appeared after I commented-out the yield call in U8x8lib.cpp   -  as below. Not sure why my system doesn't behave as expected.

HelloWorld_128x64.cpp.o: In function `_GLOBAL__sub_I_u8g2':
/home/pi/sketchbook/libraries/U8g2/U8g2lib.h:6819: undefined reference to `u8g2_DrawGlyph'
HelloWorld_128x64.cpp.o: In function `_GLOBAL__sub_I_u8g2':
/usr/share/arduino/HelloWorld_128x64.ino:274: undefined reference to `u8x8_InitDisplay'
HelloWorld_128x64.cpp.o: In function `_GLOBAL__sub_I_u8g2':
/usr/share/arduino/hardware/arduino/cores/arduino/Print.h:43: undefined reference to `u8g2_ClearDisplay'
HelloWorld_128x64.cpp.o: In function `_GLOBAL__sub_I_u8g2':
/home/pi/sketchbook/libraries/U8g2/U8g2lib.h:62: undefined reference to `u8x8_SetPowerSave'
HelloWorld_128x64.cpp.o: In function `_GLOBAL__sub_I_u8g2':
/usr/share/arduino/HelloWorld_128x64.ino:274: undefined reference to `u8g2_ClearBuffer'
HelloWorld_128x64.cpp.o: In function `_GLOBAL__sub_I_u8g2':
/usr/share/arduino/hardware/arduino/cores/arduino/Print.h:43: undefined reference to `u8g2_font_ncenB08_tr'
/usr/share/arduino/hardware/arduino/cores/arduino/Print.h:43: undefined reference to `u8g2_font_ncenB08_tr'
HelloWorld_128x64.cpp.o: In function `_GLOBAL__sub_I_u8g2':
/home/pi/sketchbook/libraries/U8g2/U8g2lib.h:62: undefined reference to `u8g2_SetFont'
/home/pi/sketchbook/libraries/U8g2/U8g2lib.h:62: undefined reference to `u8g2_DrawStr'
/home/pi/sketchbook/libraries/U8g2/U8g2lib.h:324: undefined reference to `u8g2_SendBuffer'
/home/pi/sketchbook/libraries/U8g2/U8g2lib.h:62: undefined reference to `u8x8_ascii_next'
/home/pi/sketchbook/libraries/U8g2/U8g2lib.h:62: undefined reference to `u8x8_ascii_next'
/home/pi/sketchbook/libraries/U8g2/U8g2lib.h:324: undefined reference to `u8x8_utf8_init'
/home/pi/sketchbook/libraries/U8g2/U8g2lib.h:6819: undefined reference to `u8g2_cb_r0'
/home/pi/sketchbook/libraries/U8g2/U8g2lib.h:6819: undefined reference to `u8g2_cb_r0'
/home/pi/sketchbook/libraries/U8g2/U8g2lib.h:6819: undefined reference to `u8g2_Setup_st7920_s_128x64_f'
U8g2/U8x8lib.cpp.o: In function `u8x8_byte_arduino_hw_i2c':
/home/pi/sketchbook/libraries/U8g2/U8x8lib.cpp:1178: undefined reference to `u8x8_gpio_call'
/home/pi/sketchbook/libraries/U8g2/U8x8lib.cpp:1178: undefined reference to `u8x8_gpio_call'
/home/pi/sketchbook/libraries/U8g2/U8x8lib.cpp:1178: undefined reference to `u8x8_gpio_call'
collect2: error: ld returned 1 exit status

olikraus, 2019-09-29:
Looks like U8g2 is not part of the link stage.

olikraus, 2019-10-03:
no further feedback...

closed 2019-10-03 by olikraus