olikraus/u8g2 #1010: Convert ALL bdf glyphs to use in u8g2?

eried, 2019-09-30:
Hi, we are making a terminal emulator using u8g2 https://innovationgarage.github.io/TTyGO/ and we need to convert ALL the bdf glyphs of a font that has everything from 0-65533, but I just get a small 5kb file back using bdfconv :/ how do I force it?

`bdfconv.exe -f 1-m '0-65533' -o orp-medium.h orp-medium.bdf`

olikraus, 2019-09-30:
Sometimes double quotes are required for -m
Is the bdf file available? Maybe there is an issue inside.

olikraus, 2019-09-30:
Maybe you could enable verbose output with "-v".
For me it looks like this:
```
b117m:~/git/u8g2/tools/font/bdfconv$ ./bdfconv -v -f 1 -m '0-65533' ../bdf/cu12.bdf -o test.c
Parse File ../bdf/cu12.bdf: 8453 glyph(s) found
Map: map_cmd_list='0-65533'
Map: exclude=0 from=0/$0 to=65533/$fffd map=0/$0
Reduce: Start
Reduce: End
CalculateMaxBBX: x=-9, y=-10, w=40, h=30
CalculateMaxBBX: Encodings x=3893, y=3846, w=5780, h=506
bf_CalculateMinMaxDWidth: dx_min=0, dx_max=35
bf_CalculateMinMaxDWidth: x_min=-9, x_max=11
bf_CalculateMaxBitFieldSize: bbx.x=5, bbx.y=5, bbx.w=5, bbx.h=5, dwidth=7
RLE Compress: best zero bits 4, one bits 2, total bit size 1731448
RLE Compress: Font code generation, selected glyphs=8453, total glyphs=8453
RLE Compress: ASCII gylphs=191, Unicode glyphs=8262
RLE Compress: Glyphs per unicode lookup table entry=100
RLE Compress: Unicode lookup table len=82, written entries=82
RLE Compress: Unicode lookup table first entry: delta=328, encoding=356
RLE Compress: Unicode lookup table last entry: delta=2094, encoding=65535
RLE Compress: 'A' pos = 514, 'a' pos = 1190
RLE Compress: Font size 216786
bf_WriteU8G2CByFilename: Write file 'test.c'
```
The line `RLE Compress: ASCII gylphs=191, Unicode glyphs=8262`shows the number of glyphs.
And `RLE Compress: Font size 216786` is the size of the font in bytes.

eried, 2019-10-01:
Hey thanks! with double qoutes it works flawesly :D

closed 2019-10-01 by eried