olikraus/u8g2 #1011: SSD1306 I2C 128x32 - Missing bottom pixels

jjgarciajz, 2019-09-30:
Hello,
I'm testing an OLED screen (SSD1306 128x32 0.91") using I2C and a Pro Micro, with the 8x8 GraphicsTest sample code and I've noticed there are missing pixels at the bottom of the screen.

I'm using the U8X8_SSD1306_128X32_UNIVISION_HW_I2C constructor.
The problem is more clear when the demo shows the big sized character counter.

I've made a video:
https://www.youtube.com/watch?v=7KYtHGvtIfg
As you see, the screen seems to leave a big gap at the bottom.

Probably this is not even an issue and it's something related to a cheap OLED screen, but better ask :)

Cheers,
Jose

olikraus, 2019-09-30:
U8g2 FlipMode Example would be better for this test, because it draws a box with the size of the display.

Maybe the display is not 128x32 but instead 128x40 or 128x48. Maybe you can just use a 128x64 constructor and check how many pixel are visible.

jjgarciajz, 2019-09-30:
The FlipMode example and shows the border correctly. If I increase height to 40 or 48 the top or bottom border is missing. Using a 128x64 constructor makes everything look fuzzy.

So I assume the display is really 128x32 and the display has a huge bottom border. Also, I guess the big counter numbers from the 8x8 GraphicsTest sample are expected to be shown in bigger displays.

As I said, probably some messy hardware... If you want me to test anything else, just tell me :)
Thanks for the help!

olikraus, 2019-10-01:
> Also, I guess the big counter numbers from the 8x8 GraphicsTest sample are expected to be shown in bigger displays.

Yes, but with 32 pixel height (4 rows) this is clipped by the u8g2 driver itself.

jjgarciajz, 2019-10-01:
So, just to confirm. Is it normal the big char counter is cut by the bottom on a 32 pixel height display? If so, I guess there's no issue (which is good).

Thanks

olikraus, 2019-10-02:
> So, just to confirm. Is it normal the big char counter is cut by the bottom on a 32 pixel height display? If so, I guess there's no issue (which is good).

Yes, the example was developed for a 128x64 pixel screen.

closed 2019-10-03 by olikraus