olikraus/u8g2 #1012: HX1230 SPI LCD did not compile with ESP32 stretching I2C library

prisikarm, 2019-10-01:
I am using ESP32 and HX1230 SPI LCD with struct 
`U8G2_HX1230_96X68_F_3W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 14, /* data=*/ 13, /* cs=*/ 15, /* reset=*/ 26);`

At the same time, I am using CSD30 I2C CO2 sensor that is using SoftWire library (by paulvha), with clock-stretching. 

I am using SPI display, so I can't understand why I receive I2C error, when I am adding  `u8g2.begin();` to the setup().

Here are compilation errors: 

`libraries\Wire\Wire.cpp.o: In function `TwoWire::write(unsigned char const*, unsigned int)':

C:\Users\prisi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3\libraries\Wire\src/Wire.cpp:228: multiple definition of `TwoWire::write(unsigned char const*, unsigned int)'

libraries\scd30-master\SoftWire\SoftWire.cpp.o:C:\Users\prisi\Documents\Arduino\libraries\scd30-master\src\SoftWire/SoftWire.cpp:161: first defined here

libraries\Wire\Wire.cpp.o: In function `TwoWire::available()':

Wire.cpp:(.text._ZN7TwoWire9availableEv+0x0): multiple definition of `TwoWire::available()'

libraries\scd30-master\SoftWire\SoftWire.cpp.o:SoftWire.cpp:(.text._ZN7TwoWire9availableEv+0x0): first defined here

libraries\Wire\Wire.cpp.o: In function `TwoWire::read()':

Wire.cpp:(.text._ZN7TwoWire4readEv+0x0): multiple definition of `TwoWire::read()'

libraries\scd30-master\SoftWire\SoftWire.cpp.o:SoftWire.cpp:(.text._ZN7TwoWire4readEv+0x0): first defined here

libraries\Wire\Wire.cpp.o: In function `TwoWire::peek()':

Wire.cpp:(.text._ZN7TwoWire4peekEv+0x0): multiple definition of `TwoWire::peek()'

libraries\scd30-master\SoftWire\SoftWire.cpp.o:SoftWire.cpp:(.text._ZN7TwoWire4peekEv+0x0): first defined here

libraries\Wire\Wire.cpp.o: In function `TwoWire::flush()':

Wire.cpp:(.text._ZN7TwoWire5flushEv+0x0): multiple definition of `TwoWire::flush()'

libraries\scd30-master\SoftWire\SoftWire.cpp.o:SoftWire.cpp:(.text._ZN7TwoWire5flushEv+0x0): first defined here

libraries\Wire\Wire.cpp.o: In function `TwoWire::write(unsigned char)':

Wire.cpp:(.text._ZN7TwoWire5writeEh+0x0): multiple definition of `TwoWire::write(unsigned char)'

libraries\scd30-master\SoftWire\SoftWire.cpp.o:SoftWire.cpp:(.text._ZN7TwoWire5writeEh+0x0): first defined here

libraries\Wire\Wire.cpp.o: In function `TwoWire::setClock(unsigned int)':

Wire.cpp:(.text._ZN7TwoWire8setClockEj+0x0): multiple definition of `TwoWire::setClock(unsigned int)'

libraries\scd30-master\SoftWire\SoftWire.cpp.o:SoftWire.cpp:(.text._ZN7TwoWire8setClockEj+0x0): first defined here

libraries\Wire\Wire.cpp.o: In function `TwoWire::requestFrom(unsigned char, unsigned char, unsigned char)':

Wire.cpp:(.text._ZN7TwoWire11requestFromEhhh+0x0): multiple definition of `TwoWire::requestFrom(unsigned char, unsigned char, unsigned char)'

libraries\scd30-master\SoftWire\SoftWire.cpp.o:SoftWire.cpp:(.text._ZN7TwoWire11requestFromEhhh+0x0): first defined here

libraries\Wire\Wire.cpp.o: In function `TwoWire::requestFrom(unsigned char, unsigned char)':

Wire.cpp:(.text._ZN7TwoWire11requestFromEhh+0x0): multiple definition of `TwoWire::requestFrom(unsigned char, unsigned char)'

libraries\scd30-master\SoftWire\SoftWire.cpp.o:SoftWire.cpp:(.text._ZN7TwoWire11requestFromEhh+0x0): first defined here

libraries\Wire\Wire.cpp.o: In function `TwoWire::requestFrom(int, int)':

Wire.cpp:(.text._ZN7TwoWire11requestFromEii+0x0): multiple definition of `TwoWire::requestFrom(int, int)'

libraries\scd30-master\SoftWire\SoftWire.cpp.o:SoftWire.cpp:(.text._ZN7TwoWire11requestFromEii+0x0): first defined here

libraries\Wire\Wire.cpp.o: In function `TwoWire::requestFrom(int, int, int)':

Wire.cpp:(.text._ZN7TwoWire11requestFromEiii+0x0): multiple definition of `TwoWire::requestFrom(int, int, int)'

libraries\scd30-master\SoftWire\SoftWire.cpp.o:SoftWire.cpp:(.text._ZN7TwoWire11requestFromEiii+0x0): first defined here

libraries\Wire\Wire.cpp.o: In function `TwoWire::beginTransmission(int)':

Wire.cpp:(.text._ZN7TwoWire17beginTransmissionEi+0x0): multiple definition of `TwoWire::beginTransmission(int)'

libraries\scd30-master\SoftWire\SoftWire.cpp.o:SoftWire.cpp:(.text._ZN7TwoWire17beginTransmissionEi+0x0): first defined here

libraries\Wire\Wire.cpp.o: In function `TwoWire::beginTransmission(unsigned char)':

Wire.cpp:(.text._ZN7TwoWire17beginTransmissionEh+0x0): multiple definition of `TwoWire::beginTransmission(unsigned char)'

libraries\scd30-master\SoftWire\SoftWire.cpp.o:SoftWire.cpp:(.text._ZN7TwoWire17beginTransmissionEh+0x0): first defined here

libraries\Wire\Wire.cpp.o: In function `TwoWire::endTransmission()':

Wire.cpp:(.text._ZN7TwoWire15endTransmissionEv+0x0): multiple definition of `TwoWire::endTransmission()'

libraries\scd30-master\SoftWire\SoftWire.cpp.o:SoftWire.cpp:(.text._ZN7TwoWire15endTransmissionEv+0x0): first defined here

libraries\Wire\Wire.cpp.o:(.bss.Wire+0x0): multiple definition of `Wire'

libraries\scd30-master\SoftWire\SoftWire.cpp.o:(.bss.Wire+0x0): first defined here

c:/users/prisi/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: Warning: size of symbol `Wire' changed from 16 in libraries\scd30-master\SoftWire\SoftWire.cpp.o to 304 in libraries\Wire\Wire.cpp.o

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board ESP32 Dev Module.
`

prisikarm, 2019-10-01:
commented 
```
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif
```
 
and lines 1134 - 1178

in `U8x8lib.cpp ` 

resolved my issue. Thank you

olikraus, 2019-10-01:
Yes, nur I would call this an issue of the softwire lib.

olikraus, 2019-10-03:
Let me know if you need any further support.

closed 2019-10-03 by olikraus