olikraus/u8g2 #1013: only 8 lines on display

JulienDevillers, 2019-10-02:
Hello,

I'm using u8g2 with ERC240160-1 display and esp32 port.
The following code : 
`    u8g2_Setup_st7586s_erc240160_1(&u8g2, U8G2_R0, u8g2_esp32_spi_byte_cb, u8g2_esp32_gpio_and_delay_cb);  // init u8g2 structure
    ESP_LOGD(TAG, "setup\n");
    u8g2_InitDisplay(&u8g2); // send init sequence to the display, display is in sleep mode after this,
    u8g2_SetPowerSave(&u8g2, 0); // wake up display
    u8g2_ClearDisplay(&u8g2);
    u8g2_ClearBuffer(&u8g2);
    u8g2_SetDrawColor(&u8g2, 1);
    u8g2_SetFont(&u8g2, u8g2_font_ncenB14_tr);

    for (uint8_t i = 0; i < 20; i++) {
        u8g2_ClearBuffer(&u8g2);
        u8g2_DrawFrame(&u8g2, 0, 0,  239,8);
        u8g2_DrawStr(&u8g2, i, i, "Hello World!Hello W");
        u8g2_SendBuffer(&u8g2);
       vTaskDelay(300 / portTICK_PERIOD_MS);

    }
`

produces the following result : [https://www.youtube.com/watch?v=S9WoUdND6gs&feature=youtu.be](https://www.youtube.com/watch?v=S9WoUdND6gs&feature=youtu.be)

Only 8 lines are displayed.

I checked that u8g2_GetDisplayHeight and u8g2_GetDisplayWidth returns 160 and 240 has they should.

What can it be?

Thanks
Julien

v1993, 2019-10-02:
You use page buffer I guess. Not sure how to fix this in C tho.

v1993, 2019-10-02:
I don't think you need "\n" in logging btw.

olikraus, 2019-10-02:
Use `u8g2_Setup_st7586s_erc240160_f` ("f" at the end).

JulienDevillers, 2019-10-03:
Clear and sharp ! thanks.
This project is a goob job !

closed 2019-10-03 by JulienDevillers