olikraus/u8g2 #1015: Custom Font (Malayalam Language)

e-labInnovations, 2019-10-04:
Hi sir
I am interested to use the Malayalam language with my 128X32 pixel SSD1306 display. How to make a font for printing Malayalam text on my OLED?

olikraus, 2019-10-05:
You need (a) a suitable font file, perferable in .bdf format and (b) you need to figure out the unicode range for this language. Assuming, that the range is from 0xd00 to 0xd7f, it might look like this (derived from GNU unifont):
[image]

The code for this font is this:
```
/*
  Fontname: -gnu-Unifont-Medium-R-Normal-Sans-16-160-75-75-c-80-iso10646-1
  Copyright: Copyright (C) 1998-2019 Roman Czyborra, Paul Hardy,  Qianqian Fang, Andrew Miller, Johnnie Weaver, David Corbett, et al.  License GPLv2+: GNU GPL version 2 or later <http://gnu.org/licenses/gpl.html>  with the GNU Font Embedding Exception.
  Glyphs: 225/57086
  BBX Build Mode: 0
*/
const uint8_t u8g2_font_malayalam[4378] U8G2_FONT_SECTION("u8g2_font_malayalam") = 
  "\341\0\3\2\5\5\5\5\6\20\20\0\376\12\376\13\377\1\235\3\71\5\7 \6\0@\217\2!\10A"
  "Q\210\342\220\10\42\10\205H\214\42s\13#\17FE\210\252\247aH\242\226a\210z\2$\22GE"
  "\210.\34\224(\222\302y\214\244\312 f\0%\24GE\210F\223\222R\22)i\234&R\22%%"
  "M\1&\22GE\210j+e\225PL\42-\21\223L\232\2'\7\201P\214b\10(\14\203\315\207"
  "*\211\222\250\267(\13)\15\203\311\207\42\213\262\250\227(\211\0*\15\347\304\210\256Ti\333\222\246Z"
  "\6+\14\347\304\210.\256\15C\26\327\0,\11\202L\207BI\24\0-\7$H\212b\10.\7B"
  "L\210b\10/\14FE\210\266X\15\323\260\232\2\60\22FE\210J\213\222P\233\22%\332\304$\312"
  "$\0\61\14EI\210*\223\222\260O\203\0\62\17FE\210fHB\61\315\264\260\232\16\3\63\20F"
  "E\210fHB\61\215\346T\24\223!\1\64\20FE\210\62\324\222\250\222%Y\62\214i\5\65\17F"
  "E\210\342\220V\7\71M\305dH\0\66\17FE\210j\12\323tPB\307dH\0\67\14FE\210"
  "\342ZL\213i\23\0\70\20FE\210fHBc\62$\241c\62$\0\71\16FE\210fHBc"
  "\62\250\215\321\4:\11\342\314\210b\210\207\0;\12\42\315\207b\210\225D\1<\11%I\210\62\353\332"
  "\1=\11\246D\211\342N\34\6>\11%E\210\42\355\326\21\77\17FE\210fHB\61\15\253\71\230"
  "F\0@\22FE\210j\312\244DI\226H\211\224HZ\342!A\16FE\210J\213ZBq\30D"
  "\307\0B\16FE\210bPB\343\260\204\216\303\2C\16FE\210fHBkG\61\31\22\0D\16"
  "FE\210b\210\262$\364[\62D\0E\15FE\210\342\220V\7%m\35\6F\14FE\210\342\220"
  "V\7%\355\12G\16FE\210fHBki\10m\312\22H\14FE\210\42t\34\6\321c\0I"
  "\13EI\210b\220\302~\32\4J\16GE\210j\20\343\236\262(\313\66\0K\21FE\210\42\324\222"
  "\250\222\211b\222E\265$\14L\12FE\210\42\355\257\303\0M\15FE\210\42\24\247!Z<:\6"
  "N\20FE\210\42\334\66%R\42)\221\22\355\30O\14FE\210fHB\177L\206\4P\15FE"
  "\210bPB\343\260\244]\1Q\27g\305\207f\210\302$L\302$L\302$L\22%\221\244!\7\4"
  "R\20FE\210bPB\343\260D\265$KB\61S\16FE\210fHB\263\243\230\14\11\0T\12"
  "GE\210\342\220\305\375\15U\13FE\210\42\364\217\311\220\0V\21GE\210\42\265&Y\224EY%"
  "L\322\70\3W\15FE\210\42\364\342\62\15\321(\6X\17FE\210\42\24\223\250M\324\242\226P\14"
  "Y\16GE\210\42U\223,\312*i\334\15Z\13FE\210\342Z\354\65\35\6[\12\203\321\207b\210"
  "\372\247\1\134\14FE\210\42\215\253q\32W\3]\12\203\305\207b\352\237\206\0^\11f\304\214J\213"
  "\222\60_\7'\304\207\342\20`\7cH\215\42+a\16\6E\210fH\302\64\31F\233\262\4b\16"
  "fE\210\42mY\64\321qS\26\0c\15\6E\210fHB\265c\62$\0d\14fE\210\266e"
  "\321Fo\312\22e\17\6E\210fHBq\30\324b\62$\0f\14eE\210N\12K\203\24\366\4"
  "g\23fE\207\66Y\264$K\262hK\207$\24\223!\1h\14fE\210\42mY\64\321\307\0i"
  "\14eI\210*\314\21\261O\203\0j\14\245E\207\262\216\210}\224\42\11k\21fE\210\42mK\242"
  "J&&YTK\302\0l\12eI\210F\354\237\6\1m\22\7E\210bQ\242H\212\244H\212\244"
  "H\212\244\2n\13\6E\210\42Y\64\321\307\0o\14\6E\210fHB\37\223!\1p\16FE\207"
  "\42Y\64\321qS\226\64\5q\14FE\207f\321Fo\312\222\26r\13\6E\210\42Y\64Q\355\12"
  "s\15\6E\210fHB\331\61\31\22\0t\13EE\210*,\15R\330Uu\12\6E\210\42\364\233"
  "\262\4v\14\6E\210\42\64&Qo\242\4w\21\7E\210\42\225\42)\222\42)\222\42\251b\1x"
  "\17\6E\210\42\24\223(\23\265(\11\305\0y\16FE\207\42\364\230D\226\264\62$\0z\12\6E"
  "\210\342\32\366:\14{\16\244I\207J\311\302,*\326\242,\24|\10\301Q\207\342\203\0}\17\244I"
  "\207B\314\242,,\325\302,\221\0~\12gD\214F\223\42M\1\177#\20B\7#\351w\222\316I"
  "'\35\222lJ\302\64J\206$\233\222\60\235\206d\330\71\351\234tR\322\17\200$\20B\7#\351w"
  "\222\316I'm\322\62%Q\22E\207$\32\302(\211\212Q\262\350\234tN:)\351\7\0\0\0\4"
  "\377\377\15\0\13\204X\15G\211\244D\1\15\1\15\210\320\14O\213\244$\314\206\10\15\2\13\204p\12"
  "G\211\244D\1\15\3\15#\365\10\247\242T\223D\251\0\15\4&\316\305\7\343\7i\33\24\227!q"
  "\31\22\227A\332\206\377C\264(C\342\62$\306!Q\6e\220\206e\370\20\15\5\36\16A\11o\310"
  "\304J\24\15R\250dI\250\14Q\64$YRR\262\244\224M\221\2\15\6*PA\10o\310\244,"
  "L\42%\31\244LQ\262$J\22e\210\222(Q\262\244\224(YRJ\262)\222rJ\216\15\12\0"
  "\15\7\37\213E\7K\323\242$J\302P\14\207(\224\222PJ\302D\12s,\31.\71\70\134\0\15"
  "\10\42\215A\7g\311\222Z\322\242%-Z\322\62%\251\322\24)mR\230\243\71\60\350@\16\17:"
  "\0\15\11\22\7M\11g\210*\245\212\26'\203\22\17C\0\15\12\25\13I\11gK\372\242\364)i"
  "L\223-\322\1e\220\1\15\13\26\11M\11GS\242$\32\262!\7\222\64\33\264\64\33$\0\15\14"
  "\33\13I\11K\323\242$J\302(\21\223h\210\222HJJRRJ\244H\1\15\15$\316\305\7\343"
  "\7i\33\24\227!q\31\22\227A\332\206\377C\264\15\212\313\220\270\14\211\313 m\303\7\1\15\16!"
  "MA\11\37\31r$L\266(\321\242\212\26U*\303\240\344@\224\344@\224\344@\224\344\210\6\15\17"
  " MA\11\37\30r$\214\246$\323\222\232\226\324\222d\230r \311\342$\213\223,\7\42\5\15\20"
  "#PA\11\37\335\321,\231\246$Q\22-\251(\211\226tJ\222a\11\343\244\16&u\60\251\243\31"
  "\0\15\21'\316\305\7\343\7i\33\24\227!q\31\22\227A\332\206\377\203\62(\303\64D\303\220\14\312"
  "\60$\203\62l\207\17\1\15\22\25\12I\11oH\303,NrDG\6\61\311\302dS\0\15\23\27"
[... cut ...]
  "\351[\322\267\244oI\237\42%\213\222A\15y\34\16E\11K\331\201(K\263\260\26\15\203\226Da"
  "\226Da\26%i\224\245\0\15z\37mE\11\37\322\261(\207#EYJ\265$\213\42%\213\222\212"
  "T\351)\351))u\2\15{\36kI\11{\7s\60\322\224R\22%a\224\210I$&\221\230D"
  "I\226DM\12\0\15|\30iM\11W\207r(\32\224j\222&b$Fb\224dQM\1\15}"
  "\37lE\11\37\330\321\34\215.\65-\311\222(\221*\221T\211\244J$U\242$\322\24\0\15~\34"
  "kI\11{\7s\60\7\224b\224\204Q\42&\221\230Db\22%Q\223\250\0\15\177\37mE\11w"
  "\207\262\34\316\201)\316\222\70K\302\341\240DY\22IY\22IYd\231\1\0";
```

e-labInnovations, 2019-10-05:
*Thanks for your email*
How to use this? I have not used a custom font with the *U8G2 *library.
Please help me to use this code with my Arduino + 128x32 SSD1306 OLED
project.
I am planning to teach all my friends to use the *U8G2 *library for the
Arduino OLED project. Last day I suggested my friend start OLED with your
library.
Thank you

Yours sincerely
*Mohammed Ashad*
*e-lab innovations*

e-labInnovations, 2019-10-05:
Again thanks a lot
I have problem, Malayalam font cant read in Arduino IDE software and not showing in OLED. How to solve it?

[image]

v1993, 2019-10-05:
Not sure about Arduino IDE part, but try [enableUTF8Print](https://github.com/olikraus/u8g2/wiki/u8g2reference#enableutf8print) for display.

e-labInnovations, 2019-10-05:
> Not sure about Arduino IDE part, but try [enableUTF8Print](https://github.com/olikraus/u8g2/wiki/u8g2reference#enableutf8print) for display.

Thanks

olikraus, 2019-10-05:
In order to use `u8g2.print()` you need to enable it (as mentioned above). However this is only the first step. As a second precondition, you need to ensure, that your editor is able to operate in UTF8 mode: The resulting C file must be saved as UTF8 text file. The Arduino IDE by default works in UTF8 code, you should be able to directly type in or copy Malayalam glyphs from a char set tool. Especially there should not be these boxes. Instead you should see the glyphs in the C code string itself. I mean, the gylphs should be visible like here: https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/Chinese/Chinese.ino#L280

> And also Malayalam has Dependent vowel signs. It's 'x' position is not normal. And has Two-part dependent vowel signs, when we use Two-part dependent vowel signs the character position in between Two-part dependent vowel signs.

U8g2 is made for embedded systems with very limited resources. So... it is by far not as powerful as a windows true type font rendering system (which requires magabytes of code).

Moreover, as a open source project we unfortunately do not have the resources to implement the required kerning algorithms for your language. At the end I assume it is up to you to output each glyph individually and shift the next glyph depending of the previous glyph into the right position.

There had been some attempts to implement support procedures for this, but I am not sure whether this will help you. There is an undocumented function called drawExtUTF8, see #529 

> How can I convert this font?
> https://www.dafontfree.net/freefonts-malayalam-f112349.htm

There are several FAQ entries on this:
https://github.com/olikraus/u8g2/blob/master/doc/faq.txt#L219

To obtain .bdf from .ttf you can use otf2bdf (http://sofia.nmsu.edu/~mleisher/Software/otf2bdf/)

e-labInnovations, 2019-10-06:
### Thanks a lot

I found a video tutorial on this topic but is not in English.
**Video**
[https://www.youtube.com/watch?v=V18zwLFAB18](https://www.youtube.com/watch?v=V18zwLFAB18)

I used bdfconv but it not converting full of my font

e-labInnovations, 2019-10-06:
### Malayalam Unicode table
The characters may be denoted by Unicode symbols in the range 0D00-0D7F. Below a table, that will show them if your browser supports UTF-8 and you have a Malayalam font installed.
[image]

### Consonants
The values 0D15-0D39 denote consonants (or, rather, syllables consonant+a), except for 0D29 which doesn't exist.
[image]

### A selection of conjunct consonants
[image]
When combined with vowel diacritics some consonants change shape.
[More details](https://www.win.tue.nl/~aeb/natlang/malayalam/malayalam-alphabet.html)

### Vowels that modify consonant symbols
The values 0D3E-0D43, 0D46-0D48, 0D4A-0D4D, 0D02-0D03 denote various modifying vowels, used to change the an into a different vowel (or no vowel). Here shown following the symbol ക ka.
[image]
[image]

### How to do these with the u8g2 library?
I found a similar problem in the Arabic language
[image]
[image]
[image]
How you solved this in the Arabic Language?
How can I solve it in the Malayalam language?

e-labInnovations, 2019-10-07:
Hello @olikraus 
I think you have solved all the problems in the u8g2 library related to Arabic.
**Some of them are below**
[image]

### The Malayalam language has similar conditions are below
[image]

### And also two characters combine and form a new character
[image]
[**`More details`**](https://en.wikipedia.org/wiki/Malayalam_script)

How can I create a fully working Malayalam font? Can I use any script for that?

olikraus, 2019-10-07:
> I think you have solved all the problems in the u8g2 library related to Arabic.

not true. This is still open i think. 

Arabic and Malayalam require the following extra steps:
 1. Combination of glyphs 
 2. Construction of new glyphs based on two other glyphs

The first point is partly supported by the above mentioned ExtUTF8 function.
The second point should be handled by your text editor. Or, if this is not the case, you need to do a string replace step before printing: Search for two chars replace it by one other char. However, I do not know whether the second point is true for you.

> I want to convert that font. But the problem is that the Fony software doesn't show Malayalam characters.

Fony is only for bdf editing, but I think the fonts you provided are ttf only, so otf2bdf and bdfconv are sufficient.

e-labInnovations, 2019-10-07:
> Fony is only for bdf editing, but I think the fonts you provided are ttf only, so otf2bdf and bdfconv are sufficient.

I am using Windows 7. Can you suggest me a software to convert TTF to bdf?

olikraus, 2019-10-07:
> What are you mean by that line?

It just marks the end of the table.

> Is there any code available for that?

You probably need some kind of search and replace functionality.
There is a lot of information in the net. But I do not know, whether this fits to Arduino World...

https://stackoverflow.com/questions/1494399/how-do-i-search-find-and-replace-in-a-standard-string

> I am using Windows 7. Can you suggest me a software to convert TTF to bdf?

All my own development happens within Ubuntu. As I said, I have no idea how to do this with windows.

e-labInnovations, 2019-10-07:
Thanks, @olikraus

> It just marks the end of the table.

Can I set first char to any char and second is a constant?

> All my own development happens within Ubuntu. As I said, I have no idea how to do this with windows.

 Can I use otf2bdf in my android phone using Termux? (I think android support Linux commands)

olikraus, 2019-10-07:
> Can I set first char to any char and second is a constant?

No, you have to provide all the combinations.

> Can I use otf2bdf in my android phone using Termux? (I think android support Linux commands)

Unknown...

e-labInnovations, 2019-10-10:
Hi @olikraus 

>  All my own development happens within Ubuntu. As I said, I have no idea how to do this with windows.

Is OTF2BDF convert a C programming file?
Can I run it in a C compiler?

olikraus, 2019-10-10:
> Is OTF2BDF convert a C programming file?
Can I run it in a C compiler?

Yes: http://sofia.nmsu.edu/~mleisher/Software/otf2bdf/

e-labInnovations, 2019-10-10:
@olikraus 
Where I get otf2bdf's commands?

olikraus, 2019-10-10:
> Where I get otf2bdf's commands?

You mean? You have to download the zip, extract the same, compile everything, then you will have the otf2bdf command.

closed 2019-11-23 by olikraus