olikraus/u8g2 #1017: Display show "snow" after power on (ssd1322_nhd_128x64)

meldano, 2019-10-06:
Hi olikraus,

Thanks for your nice lib!
I have a Problem with my newheaven NHD-2.7-12864WDW3 with 4-wire SPI.
Controller is a STM32L431.
After (cold) power on it shows for a short time many Pixel.

[image]

If i power down the circuit for a short time, the Display is black / empty.

Here is my init / first Screen Code

`
u8g2_Setup_ssd1322_nhd_128x64_f(&u8g2, U8G2_R0, u8x8_byte_stm32_hal_hw_spi,   u8g2_gpio_and_delay_stm32);
	u8g2_InitDisplay(&u8g2);
	u8g2_SetPowerSave(&u8g2, 1);
	HAL_Delay(500);
	u8g2_SetPowerSave(&u8g2, 0);
	u8g2_SetContrast(&u8g2, 200);
	HAL_Delay(500);

uint16_t xPos;
	
	u8g2_ClearBuffer(u8g2);
	u8g2_SetFontMode(u8g2, 1);
	u8g2_SetFontDirection(u8g2, 0);
	u8g2_SetFont(u8g2, u8g2_font_inb16_mf);
	
	const char text[] = "XPre(i)";
	
	xPos = (u8g2_GetDisplayWidth(u8g2) - u8g2_GetStrWidth(u8g2, text)) / 2;
	
	u8g2_DrawStr(u8g2, xPos, 42, text);
	u8g2_SendBuffer(u8g2);
`

Do you know this issue and have an idea to solve it?

Thanks
Daniel

olikraus, 2019-10-06:
Call clearDisplay after initDisplay but before the powersave comnand

meldano, 2019-10-06:
I test your solution.
Perfect, works fine.
Thanks.

closed 2019-10-06 by meldano