olikraus/u8g2 #1018: Help in Fony & How to use OTF2BDF

e-labInnovations, 2019-10-07:
Hello @olikraus 
I am trying to create a new font for the Malayalam language. But the problem is that the Fony app doesn't allow to select a range of more than 0-255.
[image]
Malayalam font is starting from 0xd00 (3328). Therefore I can't convert to *.bdf format from *.TTF format.
I also tried to work with OTF2BDF software. Unfortunately, I can't start because of my little knowledge of OTF2BDF software.
Can you help me to convert  *.TTF into *.bdf file?
[image]

olikraus, 2019-10-07:
> But the problem is that the Fony app doesn't allow to select a range of more than 0-255.

ok... but I can not help here. I am not the author of Fony

> I also tried to work with OTF2BDF software. Unfortunately, I can't start because of my little knowledge of OTF2BDF software.

otf2bdf is a unix tool. It will work nicely on any Linux distribution (like Ubuntu).
Again, I am not the author of otf2bdf... 
There are other tools for ttf to bdf conversions like fontforge, but fontforge might be even more complicated (and again, my knowledge is not existing on fontforge). However there had been reports here in the issue list from users who successfully converted ttf to bdf by using fontforge.

I can convert fonts for you, if you require only some two or three fonts.

e-labInnovations, 2019-10-07:
Thanks, @olikraus for sudden reply
 How you converted the Malayalam font?
I want to convert another Malayalam font that contains more letters than yours.

olikraus, 2019-10-07:
If it is a .ttf font, then I will use otf2bdf. You can just attach the font here.

hukkax, 2019-10-07:
Fony was primarily meant for making classic Windows bitmap fonts, and as such has all the limitations of that format. All the import/export functionality for other formats was added as an afterthought. So unfortunately no, there is no support for more than 256 glyphs per font, since that's all the FNT/FON bitmap font format supports.

olikraus, 2019-10-07:
@hukkax Thanks for the quick response
@e-labInnovations You can still use Fony to draw the glyphs and change the encoding later in the exported .bdf file.

e-labInnovations, 2019-10-07:
Thanks, @olikraus @hukkax 
@olikraus I want to convert these fonts
[MALAYALAM REGULAR](https://www.dafontfree.net/freefonts-malayalam-f112349.htm)
[Gayathri](https://fonts.google.com/specimen/Gayathri)

olikraus, 2019-10-07:
what should be the target size in pixel of the uppercase A?

e-labInnovations, 2019-10-08:
> what should be the target size in pixel of the uppercase A?

Please set pixel size of previously converted Malayalam font

Can you give me both fonts are in *.bdf and *.c formats with different sizes?

**Previous Size**
> Max width 15, Max height 20
> 'A' height 10, Font size 4378

e-labInnovations, 2019-10-10:
@olikraus 
Are you converted?

olikraus, 2019-10-10:
will try to do the conversion soon...

e-labInnovations, 2019-10-10:
Thanks @olikraus 
I waiting 😍

olikraus, 2019-10-10:
For Gayathri you need provide the .ttf file.

here is the other code
```
/*
  Fontname: -FreeType-malayalam-Medium-R-Normal--16-160-72-72-P-120-ISO10646-1
  Copyright: Created by Shaji N Vyapron <nvshaji@netscape.net> with PfaEdit from Original TeX fonts for Malayalam (C)Jeroen Hellingman <jehe@kabelfoon.nl>  Licensed under GPL
  Glyphs: 244/246
  BBX Build Mode: 0
*/
const uint8_t u8g2_font_malayalam_tn[6120] U8G2_FONT_SECTION("u8g2_font_malayalam_tn") = 
  "\364\0\3\2\5\5\5\5\7#\25\375\370\14\374\14\374\1\370\4n\6'!\10\201IH\304S\0\42"
  "\10\204DlD\344\24#\32\211E\270\134$Fi\224\15\203\26\205\212\30I\303\240Ei\224F\241\224"
  "\1$\35\350A\227d:HJ\242D\225(T\322!\36\302h\252D\25%\221\6\65\7\62\0%\35"
  "\215E\370\314\232)Z\230\25\263(U\244x\312\321h\216\24\61+f\215\212\226.\0&\31\211E\270"
  "\324\252\210Y\30\251\263\252HI\66$\241\24J\322\20\15\221\0'\7a\304<\304\0(\15\3FV"
  "T\22U\242~\213j\1)\17\3FVD\224E\221\324\27%J\242\10*\13\245\304{T\245\62%"
  "a\2+\22\10E\250\134\16\344@\70\14Z\16\344@\16\204\0,\7\201\310F\304\20-\7$Dj"
  "\304\20.\7!HXD\0/\14\204AH\134S\326\224\225\262\14\60\17\207E\230\324\246\330\306\325\343"
  "\220)\266\11\61\12\204E\210\134\66h\375\3\62\21\207E\230\314\20Y\302\65N\265M\264\306\303\20\63"
  "\22\207E\230\314\240h\253\232j\263U\34\242e\210\0\64\26\210E\230l\254\256I\30eRV\12\243"
  "aPs \7\42\0\65\22\207E\230\314\60\250\326d\62\313\261\70(\322\20\1\66\22\207E\230\324&"
  "%\332\234LZ\22\256F\305\66\1\67\21\210E\250\304A\226\313r,\347@,\347@\12\70\23\207E"
  "\230\314\20i\253\272)\203\242\255\306A\221\206\10\71\24\207E\230\314\20Y\302U\335\224!\211\323!S"
  "LC\4:\10!IHD\16\4;\11\201\311FD\16\14\1<\17\10E\250<\220N\333\272\3;"
  "\260\3\2=\13\210D\251\304A'\15\203\0>\16\10E\250D\16\354\310\16\314\332Q\6\77\20\206I"
  "\230\314\220H\243\30\212\241\230\346\264\10@#\316E\7\355\240\3\333\250\3\222\266T\64\345\246eZ\230"
  "ia\246\325\66)R\264e\323\341\35\36F\0A\31\212A\270\344\16\354@\22+j\244f\241\246\15"
  "C\244F\252\22\17\71\20B\21\210I\270\304!\11\207\330:\134\322\331u\270\0C\31\212E\270\334\20"
  "NS\252\310I\16\345P\16\345\300\20G\251t\34\62\0D\27\211I\310\304!\12\227\70\211w@\7"
  "t@\7\344\71\11\227a\3E\27\210I\270\304;\220\3\71\220\3\207$\7r \7r \7\16\2"
  "F\15\207I\250\304An\36\6\271g\0G\31\212E\310\334\240mR<\344H\16\205\303\216\350\310\254"
  "\304\312\244dC\24H\30\211I\310D\16\350\200\16\350\200\16<\350\200\16\350\200\16\350\200\16\4I\7"
  "\201IH\304\7J\15\206E\210l\277\212\342\244\14\11\0K\27\211I\270D\252\204R\230\25\243\64\31"
  "\25\61\323B)\224R%\26L\13\207I\230D\334\177\36\206\0M\35\213E\330\204\16\14\71\60\250C"
  "\222&J\250\330\24)\213\244\310%\323\222L\333\214\1N\25\211E\270\204|H\25Q\12%\315$F"
  "\242\242\16\351\20\13O\31\213E\330\334 nZ\254\350\300\16\351\220\16\351\320\16(\261t\35\64\0P"
  "\24\210I\270\304!\11\207t\266\16\227\34\310\201\34\310\201\34Q\32\213\305\327\334 nZ\254\350\300\16"
  "\351\220\16\351\320\16(\341\20\35\302A\21R\31\211I\270\304!\12\227\70\211\223T\31\206(U\342$"
  "N\342$Nb\1S\30\211E\270\314\240\231R%Ntd\320\201!Gv`U\64m\220\0T\27"
  "\211E\270\304c\216\344H\216\344H\216\344H\216\344H\216\344H\10U\31\211I\310D\16\350\200\16\350"
  "\200\16\350\200\16\350\200\16\310\203\246\15\22\0V\27\211E\270D\16\254C\32\245\221&ia\26*\242"
  "\22'\361,\2W$\217A\370\204**\341\250hI\30iI&i\211\244E\212\244EY\24*Y"
  "\24*\331\220\216\253Y\325\0X\26\211E\270\204\252\204\222fQ\223X\7V%\215\64\213\232\304\2Y"
  "\30\211E\270D<\244J(i\241\242&\361\16\344H\216\344H\216\204\0Z\22\211E\270\314\60\350\200"
  "\254\3\71 \233g\363\341\0[\12\3FF\304\20\365\377\64\134\14\204AHD\26f\215Y\61+]"
  "\12\3BF\304\324\377\323\20^\16\346D\213T\252mI\26%Y\22\6_\7)\300\226\304\3`\7"
  "aD<\304\0a\23(E\230\314 M\212\30KC\222\210I\232H\207Db\21\207E\230D\334\62"
  "-\266\325\272\15\212\222L\0c\16'E\210\324\246\330\346\306A\221\206\10d\22\207E\230t\313\220\14"
  "\311\240\255\326mP\16I\0e\17'E\230\324\246\330\326\341\234N\322\20\1f\15\204Eh\314\222E"
  "C\222\365\23\0g\24\247E\226\324\222(\203\266:\16\312!\211\305!S\6\5h\14\206E\210D\332"
  "\262H\243\37\3i\10\201E\70D\64\34j\13\3>\66T\216D\375\227Ak\24\207E\210D\334\246"
  "DR\242\15\231\222ER\26eJ(l\7\201E\70\304\7m\24+E\330D\62-\322\61\24C\61"
  "\24C\61\24C\261\0n\13&E\210D\262H\243\37\3o\16'E\230\324\305\266:\16\212\64D\0"
  "p\22\247E\226D\62-\266\325\272\15\212\222Lq\63\0q\21\247E\226\324\222(\203\266Z\267A\71"
  "$q\7r\13$EXD\62(Y\337\0s\20&E\210\314\220H\333\220\16\351\70)C\2t\14"
  "\204EhLV\32\222\254o\3u\14&E\210D\350\267!R\226\0v\21(A\210\204\250\204Q&"
  "\331\242\60\21g\33\0w\31+A\310\204\246m\222\22MJ\224D\221\222(Y\222(\331\264\31\65\11"
  "x\17&E\210D\70)C&\332\242D\32\3y\24\247E\206D\70&Z\24I\65%\33B\325\232"
  "\212*\0z\14&E\210\304Q,\212\255\303\0{\15\3FV\214%\352\22eQO\2|\10\1J"
  "V\304\17\1}\16\4BV\204\246\365&IY'E\2~\12g\304\232\214\246H\332\0\0\0\0\4"
  "\377\377\15\2\13\245Dx\314\240\331\222\5\15\3\17\4Eh\214\22I\211\244DR\242\0\15\5 \22"
  "\301'\335\260\15\231$%aT\311B\251\222Uj[)\211\332\222\250[\222EZ\266\0\15\6&t"
  "\301V\335\260\315\222\224di\224d\331\220EI\26Uj[T\211\372%\352'-\232rjN\211t"
  "\222\6\15\7\36nA\347\324\240\14Y\32F[\30\265&QQ\31\242\60G\245\341 \345pN\370\20"
  "\15\10.wAw\325\240\14\331\62\245a\222E\321\26&Y%\252j\225\250\232DY\62Da\216\344"
  "p\16H\303A\326r\226\234ix\320\21\0\15\11\32\211\305\246d\272Hq\62dI\224%YT\311"
  "\332\6-Grh\30\4\15\12)\221\305\66e\16/\322\262\304I\224\15\232\226I\231\226iQ\22e"
[... cut ...]
	-../otf2bdf/otf2bdf -a -r 72 -p 16 -o tmp.bdf malayalam.ttf
	./bdfconv -v -f 1 -m '$$21-$$7e,$$d02-$$d6f,$$ece3-$$ed59' tmp.bdf -o u8g2_font_malayalam_tn.c  -n u8g2_font_malayalam_tn -d ../bdf/helvB18.bdf 
	convert bdf.tga malayalam.png

meera: bdfconv
	-../otf2bdf/otf2bdf -a -r 72 -p 24 -o tmp.bdf Meera.ttf
	./bdfconv -v -f 1 -m '$$21-$$2dc,$$d02-$$d6f,$$200c-$$2215,$$10003-$$1043f>$$e003' tmp.bdf -o u8g2_font_meera_tn.c  -n u8g2_font_meera_tn -d ../bdf/helvB18.bdf 
	convert bdf.tga meera.png

olikraus, 2019-10-10:
[image]

[image]

e-labInnovations, 2019-10-11:
Thanks @olikraus

> For Gayathri you need provide the .ttf file.

[Gayatri link](https://docs.google.com/uc?id=10jeDKGdxSKR3EzXv08hp3IMGKQYJwXnI&export=view) 

 Meera font is not completed. Range: 0x21 to 0x2DC, U+0D02 to U+0D6F, 0x200C to 0x2215 and **0x10003 to 0x1043F**
please convert all glyphs

Please give me all Fonts in bdf file also.

olikraus, 2019-10-11:
> Meera font is not completed. Range: 0x21 to 0x1043F

U8g2 only supports the  Basic Multilingual Plane. I have remapped those chars to 0x0e000 (see the commands above).

The Gayatri font only contains latin chars.

```
/*
  Fontname: -FreeType-Gayatri-Medium-R-Normal--24-240-72-72-P-102-ISO10646-1
  Copyright: Copyright © 2009, TypeGirl!
  Glyphs: 106/108
  BBX Build Mode: 0
*/
const uint8_t u8g2_font_gayathri_tn[1943] U8G2_FONT_SECTION("u8g2_font_gayathri_tn") = 
  "j\0\4\4\5\5\3\5\6\30\25\374\374\17\374\17\0\2\71\4\215\7\6!\11\343\21\226\360\301\230\4"
  "\42\11\247P\247\60\302.\6#\32\255\21\276\62\325\220\61\17\36$\231j\310\250!c\36<H\62\325"
  "\220\61\0$\32\355\21\272\242\311\222\251\206\214\32\62j\315VCF\15\31\65\202\310\212M\0%$\356"
  "\21\276PH\21\11i\210\210\20B\246\10\231\22$\211\222\244\204\20\11#D$\31#D\4\31U$"
  "\0&\34\354\21\272b\352\20\242!\203\206\14\32\62\210I&\23\15\31Db\20\13\65H\0'\7\243"
  "P\227\360\0(\17\345\21\236\62\242\305\220\371\204D\11\42\3)\16\344\25\236 d\4\212\371\302\205\20"
  "\0*\22(u\256#L\214\10)\314\220\61!\33ab\0+\17kQ\266\64pF\17\36\14\32\70"
  "#\0,\10\243\320\225\220\204\0-\10i\320\256\360\200\1.\7c\20\226\220\0/\27\350\21\242\65\210"
  "\320\250A\243\6\21\32\65\210\320\250A\204F\215\2\60\24\356\25\306\242\206\305\203a\7\355\303c\17F"
  "\260Q\2\0\61\7\343\21\226\360\3\62\23\354\21\272\240f\211\312Y<x\60d\344\224\17\36\30\63\24"
  "\354\21\272\202f\5\303\221\223$y\200r&*\36\30\1\64\32\354\21\272G\256\230U\207H\214!\62"
  "\204\314\210\62#\36<\60\70r\12\0\65\22\354\21\272\360A\312)W<H\71\23\25\17\214\0\66\23"
  "\354\21\272\242B\223\221S\256x\300\314\262\7L\220\0\67\27\354\21\272\242\342\1B\202\3\11\16$G"
  "p \301\201\4\7\22\3\70\24\354\21\272\202\344\1\63\313\334\250pf\223\23\42\220 \1\71\23\354\21"
  "\272\202\344\1\63\313\36\60Q\71\23\25\17\214\0:\10\3\21\226\220,\1;\11C\321\225\220,\11\1"
  "<\21\345\21\232\62\5\211!#\206\314\315\220!D&=\14\13\221\266\360\301xp\17\36\14>\24\345"
  "\21\232\60d\10\221I\210\314\202\304\220\21$\206\14\1\77\25\354\21\272sH\5\303\221#\37<\70\62"
  "r<\260\221S\2@*\357\21\302U\60\325\64\2\207\10!#BL\31\62e\210\230!\42\15\21\63"
  "\42D\214\20\42\225\210\61cJ\221 \67B\24\0A\36\356\21\272E\322\240u#\210\15\31Ed\324"
  "\30B\323\220\60\63\2\11\13b#\306\35\34B\25\354\21\272\240\344\1\63\233\230\30\361\0\231MT<"
  "\30\2\0C\20\354\25\276\242B\23\202#\347K\202,\226(D\21\354\21\272\240d\305\253c\366\253#"
  "*N(\1E\24\354\21\272\242\342\1\303\221\223\250x\200r$A\26K\24F\21\354\21\272\242\342\1"
  "\303\221\223\250x\200r^\2G\23\354\25\276\242\342\1\303\221\23\241\311\314\330\251\27K\24H\16\354\21"
  "\272\60\314>Q\361\0\231=\33I\7\343\21\226\360\3J\13\351\21\256\66\377''.\1K\16\354\21"
  "\272\60\314\276\70\361\200\231=\33L\15\352\21\256\60n\376;b*\222 M&\361\21\316Pn\331*"
  "V\215R\214\31ab\314\10\23T\30\31\61\304\310\210!F\216\230)c\246\214\231\62\206&N\32\354"
  "\21\272@\352\24\242\64i\224\230\30b\202\204\221\21F\324d\204\352\24\1O\24\356\21\302\242\206\305\203"
  "a\7\355wGR\210H\243\4\0P\21\354\21\272\240\344\1\63\233\250x\60d\344\274\4Q$\361\21"
  "\306\242\214\221\33bd\6\216\31\70f\340\230\201c\6\216\31\70f\240\240\201\241\206\260\20\261\210\15\0"
  "R\20\354\21\272\240\344\1\63[\234x\300\314\236\15S\23\354\21\272\242\342\301\312)W\60Q\71\23\25"
  "\17\214\0T\15\354\25\272\242\342\1\242\221\363\277\2U\20\356\21\302\60\320\376wGR\210H\243\4\0"
  "V\36\356\21\272\60\320\34\262!\303\206\214\42Bhn\10\21\31\66d\330\10r\6\255$\5\0W\62"
  "\364\21\322\60\212\224\241R\206\12\241\61\63d\214\231!C\316\14\231b\10\25CF\14\32\61d\304\240"
  "##\6\231\61e\250X\241b\205\212\221\42\4\0X \355\21\272@\212\304\250!d\250 \65b\330"
  "\271\222#\313\35\33\61\212\4\31JF\215 E\0Y\26\355\21\272@\212\304\250!d\250 \65b\330"
  "\271\222C\347W\0Z\16\354\21\272\360\201Ar\364\207\17\36\30[\13\345\21\236\360`\310\374\311\3\134"
  "\23\350\21\242\60j\24\251\251HMEj\262Q\243H\15]\12\344\25\236\300b\376\342\1_\10l\20"
  "\272\360\201\1a\23\212\25\266r\344\1\271\21\17\36\10\62\364\200\4\222\1b\23\13\22\266\60pN\210"
  "\214\60\341\312>Ia\42\11\0c\15\213\21\262\202\233\201\363p\205\222\4d\23\13\22\266\70')L"
  "$\61e\257\134\230\30Bd\0e\25\213\21\262r&\205\212A#F\31I\361\300\240\212\64g\0f"
  "\16\10\22\246\362@\325\20\23\257\346\257\0g\25\13\222\265r&\205+{\345\302\304\20\42\3\207\244\310"
  "\4\0h\20\13\22\266\60pN\210\214\60\341\312\376j\0i\12#\22\226\300\346\301\3\3j\16\207\202"
  "\225\64\333A\363\177Rb\11\0k\21\13\22\266\60p\256\354\244\304\210\42\256\354j\0l\10\3\22\226"
  "\360\203\1m\36\217\21\306\60\4\311\210\7n\306\230\31cf\214\231\61f\306\230\31cf\214\231\61f"
  "&n\16\213\21\266\60\204\310\10\23\256\354\257\6o\20\213\21\266r&\205+\373\304\204\10\63G\0p"
  "\24\13\222\265\60\204\310\10\23\256\354\223\24&\222\14\234!\0q\23\13\222\265\222\302D\22S\366\312\205"
  "\211!D\6\316\1r\14\210\21\246\60\304\304\253\371+\0s\23\213\21\262\222\342A\302IR<\30G"
  "\4I\212$\0t\17\10\22\246\60jNL\274\232W\17\202\30u\16\213\21\266\60\312\376\312\205\211!"
  "D\6v\31\214\21\262\60\352\324\10BC\306\314\206\4\241\21\243F\214:V\256\42\0w \217\21\306"
  "\60f\214\231\61f\306\230\31cf\214\231\61f\306\230\31cf\314\203\21*\206 \31x\16\213\21\266"
  "\60\312V-\22\261\262\253\1y\23\13\222\265\60\312\376\312\205\211!D\6\16I\221\11\0z\21\213\21"
  "\266\360\301\260R\245\210\221\252\354\301\203\1{\25\346\25\246\63\204D\11\42c\206\214!Bf\32\42E"
  "\310\14}\25\346\25\242 h\14\21\62\323\224\31Bb\314\20J\306\10\2\240\6\0\20\226\0\243\24\354"
[... cut ...]
  "EP\220\250\200a\202\30\211\344D\64\222\210\21\61F\314\21\61\222\210\221$D\20\61\42\202\4\21#"
  "\42L\300P\202\4\222\2\377\23\13\222\265\60\312\376\312\205\211!D\6\16I\221\11\0\0\0\0\4\377"
  "\377\1\61\11\203\21\226\360\201\1 \23\11k\320\266\360\301\0 \24\10o\320\306\360\3 \30\11\243P"
  "\227@$\1 \31\11\243P\227\220\204\0 \32\11\243\320\225\220\204\0 \34\15\247P\247\60\202L\220"
  "\21V\14 \35\15\247P\247\60\302\212!aH\14 \36\15\247\320\245\60\302\212!aH\14 \42\13"
  "\306\220\242P\342A\212\2 &\15k\20\262\60b\204\211\21&&\0";

```

olikraus, 2019-10-11:
[image]

olikraus, 2019-10-11:
BDF files, extention changed from .bdf to .txt 

[meera.txt](https://github.com/olikraus/u8g2/files/3715785/meera.txt)
[gayathri.txt](https://github.com/olikraus/u8g2/files/3715786/gayathri.txt)
[malayalam.txt](https://github.com/olikraus/u8g2/files/3715787/malayalam.txt)

e-labInnovations, 2019-10-11:
Thanks @olikraus
Thanks for you time and hardworking.

olikraus, 2019-10-11:
Hope it helps

closed 2019-11-23 by olikraus