olikraus/u8g2 #1019: Write a 'xbm' file from SD

TheCook2, 2019-10-09:
I have a UNO and an LT7920 based LCD using hardware SPI that is shared with an SD card reader.  I am trying to draw a graphic on my LCD based on a modified 'xbm' file sitting on my SD card.
I modified the xbm by getting the 0x## bits and stripping the 0x, the comma, spaces and carriage returns.  What is left is a long line of HEX chars.  I get these chars and mess around to turn them into a binary number that represents pixels.  I then try and use drawPixel to turn on the relevant pixels on the LCD.  I am not really getting much.  If I Serial.print a # for 1 and " " for a 0 the serial monitor shows what I expect.  The LCD really doesn't.  At best I get a full horizontal line halfway down the LCD and along the bottom of the LCD and the last 9 lines of graphic.
Here is a video showing the pixels that should be set on the serial monitor and the pixels that are set on the LCD... it takes a while [output.zip](https://github.com/olikraus/u8g2/files/3705551/output.zip)
This is all my code it has been messed around with for a few days so some of the comments are irrelevant. 
 `//#include "U8glib.h" // u8
//U8GLIB_ST7920_128X64_1X u8g(9,8);

#include <Arduino.h>
#include <U8g2lib.h>

#include <SPI.h> //sd
#include <SD.h> //sd
U8G2_ST7920_128X64_1_HW_SPI u8g(U8G2_R0, 9);
const int chipSelect = 4; //sd

File myFile;
int text1 = 1;
int y = 65;
char charStuff[25] = " ";
String stuff;
char pixelChar = " ";
byte pixels = 0;
byte whichBit = 0;
int width = 128;
int height = 64;
int ypos = 0;
int xpos = 0;
boolean firstPart = true;

void setup() {// setup is all sd
// Open serial communications and wait for port to open:
  pinMode (A0, INPUT);
  Serial.begin(9600);
  Serial.println("Starting setup");
  Serial.print("Init SD card- ");
//  u8g.setHardwareBackup(u8g_backup_avr_spi);
  if (!SD.begin(chipSelect)) {
    Serial.println("failed");
    while (1);
  }
  Serial.println("card up.");
  u8g.begin();
    u8g.clear();
  Serial.println("Screen Cleared");
  delay(5000);
  
}

void loop() 
{
  xpos = 0;
  ypos = 0;
  u8g.firstPage();    //clear LCD and prepare for data
  myFile = SD.open("pass.hex");
  if (myFile)
  {
    while (myFile.available()&& ypos <= height) {
      if (firstPart) pixels = B00000000;
      pixelChar = myFile.read();      
      switch (pixelChar)
      {
        case '0':
          //Do nothing 
          break;
        case '1':
          if(firstPart) pixels = B00001000;
          else pixels |= B10000000;          
          break;
        case '2':          
          if(firstPart) pixels = B00000100;
          else pixels |= B01000000;          
          break;
        case '3':         
          if(firstPart) pixels = B00001100;
          else pixels |= B11000000;          
          break;
        case '4':         
          if(firstPart) pixels = B00000010;
          else pixels |= B00100000;          
          break;
        case '5':         
          if(firstPart) pixels = B00001010;
          else pixels |= B10100000;          
          break;
        case '6':         
          if(firstPart) pixels = B00000110;
          else pixels |= B01100000;          
          break;
        case '7':         
          if(firstPart) pixels = B00001110;
          else pixels |= B11100000;          
          break;
        case '8':         
          if(firstPart) pixels = B00000001;
          else pixels |= B00010000;          
          break;
        case '9':         
          if(firstPart) pixels = B00001001;
          else pixels |= B10010000;          
          break;
        case 'a':         
          if(firstPart) pixels = B00000101;
          else pixels |= B01010000;          
          break;
        case 'b':         
          if(firstPart) pixels = B00001101;
          else pixels |= B11010000;          
          break;
        case 'c':         
          if(firstPart) pixels = B00000011;
          else pixels |= B00110000;          
          break;
        case 'd':         
          if(firstPart) pixels = B00001011;
          else pixels |= B10110000;          
          break;
        case 'e':         
          if(firstPart) pixels = B00000111;
          else pixels |= B01110000;          
          break;
        case 'f':         
          if(firstPart) pixels = B00001111;
          else pixels |= B11110000;          
          break;                 
      }
      if (!firstPart)
      {
        for (int count=7;count>=0;count--)
        {
          if (bitRead(pixels,count)==0) 
          {
            xpos++;
          }
          else 
          {
            do 
            {
              drawBit(xpos, ypos);
            } while( u8g.nextPage() );
            Serial.print("xpos is : ");
            Serial.print(xpos);
            Serial.print("  ypos is : ");
            Serial.println(ypos);
            xpos++;
          }
        }
      }
      firstPart = !firstPart;
      if (xpos >= width) 
      {
        //delay(1000);
        xpos = 0; 
        ypos++;
        Serial.println();       
      }
    }
  }
//     close the file:
  myFile.close();
  xpos = 0;
  ypos = 0;
  while(1);
}

void drawBit(int x, int y)
{
//  char theseChars[4];
//  itoa(x,theseChars,4);
  u8g.drawPixel(x,y);
//  u8g.setFont(u8g_font_courB10);
//  u8g.drawStr(10,10,theseChars);
}`

Here is pass.hex
`0e000000000000000000001e000000001f000000000000000000003e000000001b0000000000000000000036000000001b0000000000000000000036000000001b0000000000000000000036000000000300000000000000000000360000000003000000000000000000003600000000030000000000000000000036000000000300000000000000000000360000000003000000000000000000003600000000030000000000000000000036000000000300000000000000000000360000000003000000000000000000003600000000c3f83cdb7e0000f876000036be01e779c3f87cdb7e0000f877000036be01f7fbc3d86cdb180000d877000036860133dbc3d86cdb180000d877000036860133dbc3d86cdb180000d877000036860133dbc3d80cdb180000d877000036860133dbc3fe0cdb180000fc77000036bf0133dbc3fe0cdb180000fc7f000036bf0133dbc3fe6cdb180000fc7d000036bf0133dbc3d86cdb180000d871000036860133dbc3d87ddf180000f87f000036bedff3dbc3983dde180000f83c000036bedfe1d91b0000000000000000000036000000001b0000000000000000000036000000001b0000000000000000000036000000001f000000000000000000003600000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000c0ff7f0000fe0700ff01c07f00000000c0ffff0300fe0700ff00c03f00000000c0ffff0700fc0700ff01c03f00000000c0ffff0f00fc0f00ff01c07f00000000c0ffff0f00fc0f00ff01c07f00000000c0ffff1f00f80f00fe03c07f00000000c0ffff1f00f80f00fe0780ff000000004000fc1f00f81f00fe0780ff010000000000f01f00f01f00fc0f00ff010000000000f01f00f01f00f81f00fe030000000000f01f00e03f00f81f00fe07000000c0ffff1fe0ef3f00f03f00fc0f000000c0ffff1ff0ef3f00e07f00fc0f000000c0ffff0ff0cf7f00c07f00f81f000000c0ffff0ff0c77f0080ff00f03f000000c0ffff07f0c77f0080ff01e03f000000c0ffff03f887ff0000ff01c07f000000c0ffff00f887ff0000ff01c07f000000c07f0000f803ff0100fe03807f000000c07f0000fc03ff0100fe03807f000000c07f0000fc03ff0100fe03807f000000c07f0000fc03fe0100ff03c07f000000c07f0000fc01fec3fffff3ff7f000000c07f0000feffffc3fffff1ff7f000000c07f0000feffffc3fffff1ff7f000000c07f0000feffffc7fffff0ff3f000000c07f0000ffffffc7ff7ff0ff1f000000c07f0000ffffffc7ff7ff0ff1f000000c07f0080ffffffcfff1ff0ff07000000c07f0080ffffffcfff07f0ff000000 `

Does anyone have any hints on what I am doing wrong please?

For what it is worth, here is the xbm file too.
[xbmFile.zip](https://github.com/olikraus/u8g2/files/3705576/xbmFile.zip)

TheCook2, 2019-10-09:
Back again.

I have included  a display of the expected X and Y positions of the pixel and put a firstPage() just before the do...while (u8g2.nextPage() ) loop.  I get a a good readout on the bottom of the LCD and I can see each pixel light up in the expected places.  BUT, of course, I only see one pixel light up at a time.  The sketch for this is 
`//#include "U8glib.h" // u8
//U8GLIB_ST7920_128X64_1X u8g(9,8);

#include <Arduino.h>
#include <U8g2lib.h>

#include <SPI.h> //sd
#include <SD.h> //sd
U8G2_ST7920_128X64_1_HW_SPI u8g2(U8G2_R0, 9);
//U8G2_ST7920_128X64_1_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* CS=*/ 9, /* reset=*/ 8);
typedef u8g2_uint_t u8g_uint_t;

const int chipSelect = 4; //sd

File myFile;
int text1 = 1;
int y = 65;
char charStuff[25] = " ";
String stuff;
char pixelChar = " ";
byte pixels = 0;
byte whichBit = 0;
int width = 128;
int height = 64;
u8g_uint_t ypos = 0;
u8g_uint_t xpos = 0;
boolean firstPart = true;

void setup() {// setup is all sd
// Open serial communications and wait for port to open:
  pinMode (A0, INPUT);
  Serial.begin(9600);
  Serial.println("Starting setup");
  Serial.print("Init SD card- ");
//  u8g.setHardwareBackup(u8g_backup_avr_spi);
  if (!SD.begin(chipSelect)) {
    Serial.println("failed");
    while (1);
  }
  Serial.println("card up.");
  u8g2.begin();
  u8g2.clear();
  Serial.println("Screen Cleared");
  delay(5000);
  
}

void loop() 
{
  xpos = 0;
  ypos = 0;
  u8g2.firstPage();    //clear LCD and prepare for data
  myFile = SD.open("pass.hex");
  if (myFile)
  {
    while (myFile.available()&& ypos <= height) {
      if (firstPart) pixels = B00000000;
      pixelChar = myFile.read();      
      switch (pixelChar)
      {
        case '0':
          //Do nothing 
          break;
        case '1':
          if(firstPart) pixels = B00001000;
          else pixels |= B10000000;          
          break;
        case '2':          
          if(firstPart) pixels = B00000100;
          else pixels |= B01000000;          
          break;
        case '3':         
          if(firstPart) pixels = B00001100;
          else pixels |= B11000000;          
          break;
        case '4':         
          if(firstPart) pixels = B00000010;
          else pixels |= B00100000;          
          break;
        case '5':         
          if(firstPart) pixels = B00001010;
          else pixels |= B10100000;          
          break;
        case '6':         
          if(firstPart) pixels = B00000110;
          else pixels |= B01100000;          
          break;
        case '7':         
          if(firstPart) pixels = B00001110;
          else pixels |= B11100000;          
          break;
        case '8':         
          if(firstPart) pixels = B00000001;
          else pixels |= B00010000;          
          break;
        case '9':         
          if(firstPart) pixels = B00001001;
          else pixels |= B10010000;          
          break;
        case 'a':         
          if(firstPart) pixels = B00000101;
          else pixels |= B01010000;          
          break;
        case 'b':         
          if(firstPart) pixels = B00001101;
          else pixels |= B11010000;          
          break;
        case 'c':         
          if(firstPart) pixels = B00000011;
          else pixels |= B00110000;          
          break;
        case 'd':         
          if(firstPart) pixels = B00001011;
          else pixels |= B10110000;          
          break;
        case 'e':         
          if(firstPart) pixels = B00000111;
          else pixels |= B01110000;          
          break;
        case 'f':         
          if(firstPart) pixels = B00001111;
          else pixels |= B11110000;          
          break;                 
      }
      if (!firstPart)
      {
        for (int count=7;count>=0;count--)
        {
          if (bitRead(pixels,count)==0) 
          {
            xpos++;
          }
          else 
          {
            u8g2.firstPage();
            do 
            {
              drawBit(xpos, ypos);
            } while( u8g2.nextPage() );
            delay(500);
            Serial.print("xpos is : ");
            Serial.print(xpos);
            Serial.print("  ypos is : ");
            Serial.println(ypos);
            xpos++;
          }
        }
      }
      firstPart = !firstPart;
      if (xpos >= width) 
      {
        //delay(1000);
        xpos = 0; 
        ypos++;
        Serial.println();       
      }
    }
  }
//     close the file:
  myFile.close();
  xpos = 0;
  ypos = 0;
  while(1);
}

void drawBit(u8g_uint_t x, u8g_uint_t y)
{
  char theseChars[15];
  itoa(x,theseChars,15);
  u8g2.drawPixel(x,y);
  u8g2.setFont(u8g2_font_HelvetiPixelOutline_tr);
  u8g2.drawStr(0,63,"X = ");
  u8g2.drawStr(30,63,theseChars);
  itoa(y,theseChars,15);
  u8g2.drawStr(60,63,"Y = ");
  u8g2.drawStr(90,63,theseChars);
}`

Still not sure why the pixels don't light up in the original code above.

Got to say I am enjoying some of those extra fonts in u8g2.

Taa

olikraus, 2019-10-09:
I do not have the time to review your code. At least you should use code tags for it.
Also the problem is not clear to me (and other readers). A picture of the result would be good to see what is wrong. 
One general question: Obviously this is not XBM format, so: why don't you store everything as a binary? Then the code might be more simpler and you do not need to do the hex conversion.

TheCook2, 2019-10-09:
Doing it with a binary file would be great but (and there always is a but) I have had some real troubles creating a valid binary file.  I searched for a long time for one that gives a good approximation of my image.  One site merely converted the xbm hex to binary which puts each nibble of each byte the wrong way around.  Finally, late last night after converting my image into png, gif, jpg and a few others I found this site [https://www.dcode.fr/binary-image](url) that took a gif and gave a true binary 0=unset 1=set image binary.  I'll modify my code to remove the 'xbm' conversion (which wasn't easy given that xbm hex is back to front and inside out compared with the image) and I'll tidy up and add some comments.  The only reason I put all that scary code in is that people are always trying to hide their code and are always asked to show the code.  I have no shame and am happy to throw it out there warts and all.  

I remember long ago (30+ years) reading an article that said, amongst other things, I don't use comments... it took me long enough to work out how to do... why deny others the same experience.

Taa for now I'll mess with code and see what happens.

TheCook2, 2019-10-10:
Back again.
I cleaned up the code and used a 0s and 1s file on the SD card.  

I did put a video on the first entry to give an idea of what I was getting and when compared to the serial monitor report on what I should be getting.  For what it is worth here is a photo of the final result now (much the same as before except I don't get a solid line along the bottom of the LCD).
[image]
You can see that the first expected pixel is set at position (1,0) but nothing else until you get to the last 8 rows.  There is a horizontal line through the middle too.  The line appears when I am trying to set pixels at about row 4.  It is very satisfying seeing those last few lines come out but the rest would be nice too.

If I uncomment the u8g2.firstPage(); line inside the case '1'  section (and add a bit of a delay) I can see each pixel light up one by one and turn off when the loop gets back to the u8g2.firstPage(); i.e I can only see one pixel at a time but they are in the right place.   

The trouble I have is that the pixels in the first 56 rows don't turn on (or stay on) using my sketch.

Here is the sketch:
`#include <Arduino.h>
#include <U8g2lib.h>

#include <SPI.h> //sd
#include <SD.h> //sd
U8G2_ST7920_128X64_1_HW_SPI u8g2(U8G2_R0, 9);
//U8G2_ST7920_128X64_1_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* CS=*/ 9, /* reset=*/ 8);
typedef u8g2_uint_t u8g_uint_t;

const int chipSelect = 4; //sd

File myFile;
char pixelChar = " ";
int width = 128;
int height = 64;
u8g_uint_t ypos = 0;
u8g_uint_t xpos = 0;

///////////////////////////////////////////////////
//    T I M E    T O    S T A R T    IT    UP    //
///////////////////////////////////////////////////
void setup() 
{
  // Open serial communications and wait for port to open:
  pinMode (A0, INPUT);
  Serial.begin(9600);
  Serial.println("Starting setup");
  Serial.print("Init SD card- ");
  // Open the SD card
  if (!SD.begin(chipSelect)) {
    Serial.println("failed");
    while (1);
  }
  Serial.println("card up.");
  u8g2.begin();
  u8g2.clear();
  // Clear the LCD screen and pause for 5 seconds
  Serial.println("Screen Cleared");
  //u8g2.setDrawColor(2);
  delay(5000);
  
}

/////////////////////////////////////////////
//      L E T S    D O    A    L O O P     //
// (It really only does it once and stops) //
/////////////////////////////////////////////

void loop() 
{
  xpos = 0;
  ypos = 0;
  u8g2.firstPage();    //clear LCD and prepare for data
  myFile = SD.open("pass.0n1");  //Open up the 0s and 1s file
  if (myFile)                    //If the file is there and opened
                                 //Do stuff until the end of file or 
                                 //the image has been drawn
  {
    while (myFile.available()&&(ypos < height)) 
    {
      pixelChar = myFile.read();      //read a character from the image file
      switch (pixelChar)
      {
        case '0':
          //If it is a 0 we don't want to set the pixel so just
          //increment the position
          xpos++; 
          break;
        case '1':
          //If it is a 1 we want to set the pixel and
          //increment the position
//////////u8g2.firstPage();
          do 
          {
            drawBit(xpos, ypos);
          } while( u8g2.nextPage() );
          //Report what pixels should be turned on via the serial monitor
          Serial.print("xpos is : ");
          Serial.print(xpos);
          Serial.print("  ypos is : ");
          Serial.println(ypos);
          xpos++; 
//////////delay(500);    
          break;                      
      }      
      if (xpos >= width)    //If we are at the end of a line of the image
                            //reset the x position and increment the
                            //y position
      {
        xpos = 0; 
        ypos++;
        Serial.println();   //put a bit of a break in the serial monitor
                            //output
      }
    }
  }
//     close the file:
  myFile.close();
  xpos = 0;
  ypos = 0;
  while(1);                 //Stop everything for now
}

void drawBit(u8g_uint_t x, u8g_uint_t y)
{
  char theseChars[5];
  
  u8g2.drawPixel(x,y);            //Turn on the pixel at location (x,y)
  //Display the x and y values on the LCD (only if I firstPage every time)
//  sprintf(theseChars, "%3d",x);   //Convert x into a char array so I can
//                                  //display it on LCD 
//  u8g2.setFont(u8g2_font_HelvetiPixelOutline_tr);  
//  u8g2.drawStr(0,63,"X = ");
//  u8g2.drawStr(30,63,theseChars);
//  sprintf(theseChars, "%2d",y);   //Convert y into a char array so I can
//                                  //display it on LCD 
//  u8g2.drawStr(70,63,"Y = ");
//  u8g2.drawStr(100,63,theseChars);
}
`
Here is the 0s and 1s file and the output to the Serial Monitor:
[pass.zip](https://github.com/olikraus/u8g2/files/3709933/pass.zip)

Taa.

olikraus, 2019-10-10:
> I remember long ago (30+ years) reading an article that said, amongst other things, I don't use comments... it took me long enough to work out how to do it... why deny others the same experience.

Fine then, I we can close this issue. Without doubt I can easily live without this experience.

Please remember: Whatever I do here is for free. I do not get payed and I spent my spare time here.

It you want me to solve a problem, then you should name your problem. Short and clear.

If you want me to look into your code, then you should make the code as understandable, clear and beautiful as possible.

If you have a question, then ask the question. However... this is the u8g2 issue tracker. The question should be related to u8g2. 

Hint: I am sure you have used the code tag. And you wonder why it doesn't work. And now you complain about the issue tracker here, that the code still looks ugly. But It can be done better...

``` C
for( i = 0; i < 10; i++ )
  printf("%d ", i);
```

I have spent a lot of my personal time into u8g2. Wouldn't it be fair to minimize my time which I have to spend on these issues here? Please have a look at the issuenumber, the started at #1 and they are increased by one.

TheCook2, 2019-10-10:
Issue has not been closed.  It has been moved.

closed 2019-10-10 by TheCook2

olikraus, 2019-11-23:
Just for reference: I have created a demo for reading and writing XBM data from and to SD Card: https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/LoadFromSD/LoadFromSD.ino