olikraus/u8g2 #1020: Improve compression of fonts

MarkJeronimus, 2019-10-09:
The compression of fonts can be improved by replacing run-length compression by Huffman compression. As a small proof of concept, I encoded u8g2_font_unifont_tr (60 glyphs of 32x32)

- Original: 3072 bytes
- Your compression: 1273 bytes (41%)
- Huffman compression: 935 bytes + 72 lines Huffman codes (35%)

Each huffman code is 9 bits, but even if you store these in uint16's the total is still less.

The compression ratio is only expected go down slightly for larger font bitmaps (up to about 37%)

My Huffman decoder is relatively simple. I have not done speed comparisons, this might pose an issue.

```
uint8_t HuffmanDecoder::nextSymbol()
{
    uint8_t row = 0;
    while (true)
    {
        bool codon = nextCodon();
        uint16_t huffmanCode = pgm_read_word(&table[row][codon]);
        if (huffmanCode & 0x100)
            row = huffmanCode;
        else
            return huffmanCode;
    }
}

bool HuffmanDecoder::nextCodon()
{
    bool codon = pgm_read_byte(ptr) & readBit;

    readBit >>= 1;
    if (readBit == 0)
    {
        ptr++;
        readBit = 0x80;
    }

    return codon;
}
```

olikraus, 2019-10-09:
Sounds like a huge effort. Actually I do not have the time to work on this :-(

MarkJeronimus, 2019-10-09:
I'd like to work on this, but I've just found u8g2 the other day, after I made my algorithm, so I'm still trying to find mu way. Is there a folder with the original images of all fonts in this repo?

olikraus, 2019-10-09:
I like the idea of a new compression method, but I think it will become a complete new repo. And yes, there is a full automatic toolchain for the font construction.

There are two sources: pure bdf file and ttf files.
BDF files are here: https://github.com/olikraus/u8g2/tree/master/tools/font/bdf
TTF files are here: https://github.com/olikraus/u8g2/tree/master/tools/font/ttf

TTFs are converted to BDF with otf2bdf, which is here: https://github.com/olikraus/u8g2/tree/master/tools/font/otf2bdf
BDFs are converted to u8g2 internal format with bdfconv, which is here: https://github.com/olikraus/u8g2/tree/master/tools/font/bdfconv

The build tool for the fonts is here: 
https://github.com/olikraus/u8g2/tree/master/tools/font/build

The tool itself is build.c, which is here: https://github.com/olikraus/u8g2/blob/master/tools/font/build/build.c#L228

As you can see, it contains a huge list of all fonts and all the details to convert these fonts by calling otf2bdf and bdfconv with suitable arguments.

There are two build targets fro build.c: build1 and build2
build1 will convert the fonts
build2 will render the pictures for all of the fonts and the word cloud for the group overview (https://github.com/olikraus/u8g2/wiki/fntgrp)

The complete build process probably requires Linux. I assume it will not at all run on Windows.

Some more details are also here:
https://github.com/olikraus/u8g2/wiki/internal#rebuild-fonts

The current RLE compression happens here: https://github.com/olikraus/u8g2/blob/master/tools/font/bdfconv/bdf_rle.c#L362

So I assume, that bdfconv needs to be updated (a lot) to implement the compression algorithm.

bdfconv includes a reference implementation of the decoder: https://github.com/olikraus/u8g2/blob/master/tools/font/bdfconv/fd.c

There reference implementation was taken over to the actual uC code: https://github.com/olikraus/u8g2/blob/master/csrc/u8g2_font.c#L359

Someone once write a font format description, which is here: https://github.com/olikraus/u8g2/wiki/u8g2fontformat

olikraus, 2019-10-10:
> prefer font bitmaps as 'original' instead of BDF

Not exactly sure what you mean. BDF is the original and there are still a lot of tools, which can deal with them. Moreover it is a perfect startingpoint for u8g2.

olikraus, 2019-11-23:
Closing this. I do not have the time for this project to change the compression mode again...

closed 2019-11-23 by olikraus