olikraus/u8g2 #1022: SSD1306 with ATtiny85 (2)

SchneHa, 2019-10-09:
In combination of a SSD1306 OLED and an ATtiny85 MC left upper corner is always showing a white dot. I suppose that it's the cursor. I found no method to switch it off.

olikraus, 2019-10-09:
any picture? What is your exact u8g2 constructor?

SchneHa, 2019-10-09:
My constructor is U8G2_SSD1306_128X64_NONAME_1_SW_I2C u8g2(U8G2_R0,/* clock=*/ 2, /* data=*/ 0, /* reset=*/ U8X8_PIN_NONE);
Picture: [https://www.dropbox.com/s/tzpc805g87hvmqo/IMG_1547.JPG?dl=0](url)
Pixel in the upper left corner is permanent, other pixels are generated by the sketch.
Code:
#include <U8g2lib.h>

U8G2_SSD1306_128X64_NONAME_1_SW_I2C u8g2(U8G2_R0,/* clock=*/ 2, /* data=*/ 0, /* reset=*/ U8X8_PIN_NONE);

#define MAX_LENGTH 30 //30 segments max

#define X 0
#define Y 1

#define JOYSTICK_X 2
#define JOYSTICK_Y 3
#define JOYSTICK_SW 6

#define DIR_THRESH 300 //Values must be either 0-300 or 723-1023 to count
uint8_t segmentPositions[MAX_LENGTH][2];

uint8_t headPosition[2] = {63, 31}; //Place snake in beginning to start
uint8_t foodPosition[2]; //Where the food is located
uint8_t tempPosition0[2]; //Store position of previous segment to pass on to next
uint8_t tempPosition1[2]; //Store position of previous segment to pass on to next

int segmentLength = 3;

void gameUpdate();

enum DIRECTIONS {
  RIGHT,
  DOWN,
  LEFT,
  UP
} currentDirection;

void setup() {
  u8g2.begin();
  u8g2.setPowerSave(0);
  pinMode(JOYSTICK_X, INPUT);
  pinMode(JOYSTICK_Y, INPUT);
  pinMode(JOYSTICK_SW, INPUT_PULLUP);
  randomSeed(analogRead(0));
  beginGame();
}

void loop() {
  u8g2.firstPage();
  do {
    gameUpdate();
  }
  while(u8g2.nextPage());

}

void beginGame(){
  currentDirection = RIGHT;
  spawnFood();
  delay(1000);
}

bool checkCollisions(){
  for(int i=1; i<segmentLength; i++){
    if(segmentPositions[i][X] == headPosition[X]){
      if(segmentPositions[i][Y] == headPosition[Y]){
        return 1;
      }
    }
  }
  if(headPosition[X] <= 0 || headPosition[X] >= 128) return 1;
  else if(headPosition[Y] <= 0 || headPosition[Y] >= 64) return 1;
  return false;
}

void spawnFood(){
  int randomX = random(5, 123);
  int randomY = random(5, 60);
  foodPosition[X] = randomX;
  foodPosition[Y] = randomY;
}

void checkFoodEaten(){
  if(headPosition[X] == foodPosition[X] || headPosition[Y] == foodPosition[Y]){
    segmentLength += 1;
    spawnFood();
  }
}

void updateDirection(){
  int joy_x_val = analogRead(JOYSTICK_X);
  int joy_y_val = analogRead(JOYSTICK_Y);
  if(joy_x_val <= DIR_THRESH) currentDirection = RIGHT;
  else if(joy_x_val >= 1023-DIR_THRESH) currentDirection = LEFT;
  else if(joy_y_val <= DIR_THRESH) currentDirection = UP;
  else if(joy_y_val >= 1023-DIR_THRESH) currentDirection = DOWN;
}

void displaySegments(){
  for(int segment=0; segment < segmentLength; segment++){
    u8g2.drawPixel(segmentPositions[segment][X], segmentPositions[segment][Y]);
  }
  u8g2.drawPixel(foodPosition[X], foodPosition[Y]);
}

void gameUpdate(){
  updateDirection();
  tempPosition0[X] = headPosition[X];
  tempPosition0[Y] = headPosition[Y];
  switch(currentDirection){
    case UP:
      headPosition[Y] -= 1;
      break;
    case DOWN:
      headPosition[Y] += 1;
      break;
    case LEFT:
      headPosition[X] -= 1;
      break;
    case RIGHT:
      headPosition[X] += 1;
      break;
  }
  //Update each segment's position relative to the head
  for(int segment=1; segment < segmentLength; segment+= 2){
    tempPosition1[X] = segmentPositions[segment][X];
    tempPosition1[Y] = segmentPositions[segment][Y];
    segmentPositions[segment][X] = tempPosition0[X];
    segmentPositions[segment][Y] = tempPosition0[Y];
    
    tempPosition0[X] = segmentPositions[segment+1][X];
    tempPosition0[Y] = segmentPositions[segment+1][Y];
    segmentPositions[segment+1][X] = tempPosition1[X];
    segmentPositions[segment+1][Y] = tempPosition1[Y];
  }
  displaySegments();
  checkFoodEaten();
  if(checkCollisions()) endGame();
  if(segmentLength>= MAX_LENGTH) endGame();
  delay(50);
}

void endGame(){
  segmentLength = 3;
  headPosition[0] = 63;
  headPosition[1] = 31;
  beginGame();
}

SchneHa, 2019-10-09:
Sorry, "inserting code" doesn't work properly.

SchneHa, 2019-10-09:
I can't test because the ATtiny has too less flash memory to upload examples. I can only upload "Hello world" example and that's a text example.

olikraus, 2019-10-10:
What about

> u8g2.drawPixel(segmentPositions[segment][X], segmentPositions[segment][Y]);

Maybe you should try to remove the drawPixel commands to see whether the pixel disappears.

SchneHa, 2019-10-10:
Yes, we are on the right way. Commenting out the line let the pixel disappear. I changed the prior line to 
`for(int segment=1; segment < segmentLength; segment++){`
and the pixel only appears when the sketch is started. After one step it is gone.
Do you have an idea because it appears when started?

olikraus, 2019-10-10:
> Do you have an idea because it appears when started?

I am sorry to say this: But I do not have the time to review your code. 

I mean, I will be happy to fix an issue in u8g2, but at the moment, I think it is a bug in your code.

SchneHa, 2019-10-11:
I agree with you, it's a bug in the code. But it's not my code I adapted from a Hackster project. Therefore, it's not so easy for me to debug. But I will try my best.
Thank you very much.

closed 2019-11-23 by olikraus