olikraus/u8g2 #1023: u8g2.drawPixel(x,y); fails on first 50 lines of LCD.

TheCook2, 2019-10-10:
I have been setting pixels using u8g2.drawPixel(x,y) but only the first pixel at (1,0) and the pixels on the last few lines of the display turn on.

TheCook2, 2019-10-10:
If I set the buffer size to F it seems to work for all the screen.  It looks like using a smaller buffer size may be the issue.  Is there anything I should do if I only want a buffer size of 1?

AnHardt, 2019-10-10:
https://github.com/olikraus/u8glib/wiki/tpictureloop

TheCook2, 2019-10-10:
Outstanding!   I have no idea what it means but it is something to investigate and another learning opportunity.

Taa

TheCook2, 2019-10-11:
I've had a slow read of the reference.  It really does help.  I wrote a quick sketch to have a look at what is happening:

`#include <Arduino.h>
#include <U8g2lib.h>
#include <SPI.h>
U8G2_ST7920_128X64_1_HW_SPI u8g2(U8G2_R0, 9);

typedef u8g2_uint_t u8g_uint_t;
u8g_uint_t ypos = 0;
u8g_uint_t xpos = 0;

void setup()
{
  Serial.begin(9600);
  u8g2.begin();
  u8g2.clear();
}

void loop()
{
  u8g2.firstPage();    //clear LCD and prepare for data
  for (int count=0; count<500; count++)
  {
    xpos = random(0,128);
    ypos = random(0,64);        
    do 
    {
      drawBit();
      Serial.println(count);
    } while( u8g2.nextPage() );  
  }
  while(1);
}

void drawBit(void)
{  
  u8g2.setDrawColor(2);
  u8g2.drawPixel(xpos,ypos);
}`

PLEASE NOTE: the sketch is only here to show what I did.  I DON'T WANT ANYONE TO CORRECT THE CODE.  It is just a quick and dirty thing to show how I am trying to set random pixels all over the LCD. 

Back to the issue.  As you can see I did put in a picture loop.  Inside the loop I print a number so that I can count how many times it goes around the loop.

What I found is that the first time I run through the picture loop it takes 8 loops to draw that first pixel.  Every other time the picture loop only goes around once.  I think the one loop only impacts the bottom (last) eighth of the LCD.  I have tried to use an F constructor and the result is very pretty BUT I get a low memory warning.

The reference states 

> In general the value of a pixel must not depend on any other pixel values: Only information about pixel in the current page is available. As a consequence, it is not possible to copy regions within the display memory or to do scrolling.

Does this mean that the random pixels set in the previous passes through the picture loop are lost and only the pixels that have been set on the current page (the last 8 lines) can be drawn?

OR Is the issue that nextPage() was set the first time through the loop and stays that way?

Still learning and still hopeful I can do what I want.  I have used PROGMEM to store xbm files and they display and scroll really well but I want to try more pictures than PROGMEM can hold.  That is why I moved to SD memory.  If I do that with an F constructor the sketch is too large and will not compile.  But that is another issue.  For now I'll be happy if I can set random pixels all over the screen.

I am very grateful for the library and it does almost everything I want.  I just to know if it is me or the library that is not functioning the way I want.

Taa

TheCook2, 2019-10-11:
OK,

me again.  I managed to get it working.  You need to do everything in that first picture loop.  I put a loop inside the picture loop to generate a bunch of 'random' pixels... and they all show up!!!!!

Woooooo Hoooooo.   Now all I need to do is do my SD reading inside the picture loop and see what happens.

olikraus, 2019-10-11:
Thanks: The code is much clearer and does focus on the problem (the formating I fixed). :+1: 

Taking your example
``` C
void loop()
{
  u8g2.firstPage();    //clear LCD and prepare for data
  for (int count=0; count<500; count++)
  {
    xpos = random(0,128);
    ypos = random(0,64);        
    do 
    {
      drawBit();
      Serial.println(count);
    } while( u8g2.nextPage() );  
  }
  while(1);
}
```

the calls to `firstPage` and `nextPage` must fit together. These two commands form a loop which must not be altered. In your case the loop is somehow nested with the for loop, which will not work.
In fact, using random() violates the request to draw the same pattern for all the loops in the same way. For this purpose you need to store the random values in an array, maybe like this:

``` C
void loop()
{
  int xa[500];
  int ya[500];
  for (int count=0; count<500; count++)
  {
    xa[count] = random(0,128);
    ya[count] = random(0,64);        
  }

  u8g2.firstPage();    //clear LCD and prepare for data
  do 
  {
    for (int count=0; count<500; count++)
    {
      xpos = xa[count];
      ypos = ya[count];        
      drawBit();
    }
  } while( u8g2.nextPage() );  
  while(1);
}
```
... not tested. BTW `firstPage`will only clear the top page (first 8 rows).

TheCook2, 2019-10-11:
I had no idea you can use an array of x and y positions.  How do you use drawPixel with the two arrays?

olikraus, 2019-10-11:
Via xpos and ypos.

TheCook2, 2019-10-11:
I did this:

void drawBit(void)
{ 
  u8g2.setDrawColor(2);
  for (int count=0; count<100; count++) 
  {
    u8g2.drawPixel(xpos[count],ypos[count]);   
  }    
}

I tried u8g2.drawPixel(xpos,ypos); by itself and it didn't work.  Anyway.     I'll try and pre-load some arrays from SD and then use draw loop once.

By the way the formatting of my sketch got lost in translation when I used the <> insert code thing.  It seems to strip spaces.

Taa

olikraus, 2019-10-11:
> u8g2.drawPixel(xpos[count],ypos[count]);

In may code I named the arrays as xa and ya, so it should be `u8g2.drawPixel(xa[count],ya[count]);` if you took over my code.

> By the way the formatting of my sketch got lost in translation when I used the <> insert code thing. It seems to strip spaces.

The code has to start with ` ``` C` in a single line and has to be terminated with  ` ``` `

closed 2019-11-23 by olikraus