olikraus/u8g2 #1024: Nagging Drawstring error

Dr-Bundolo, 2019-10-14:
Hi:
I've been working on this problem for days with no luck.  The program works well on the serial screen.  It's just when I try and print the content of packet on the OLED  does the problem surface.  So if you could point me in the right direction of fixing that line, I'd be grateful.
Jeff
`
#include <U8x8lib.h>
#include <LoRa.h>
#include <SPI.h>
int receivedText;
int targetDistance;

// WIFI_LoRa_32 ports
// GPIO5  -- SX1278's SCK
// GPIO19 -- SX1278's MISO
// GPIO27 -- SX1278's MOSI
// GPIO18 -- SX1278's CS
// GPIO14 -- SX1278's RESET
// GPIO26 -- SX1278's IRQ(Interrupt Request)

#define SS      18
#define RST     14
#define DI0     26
#define BAND    433E6

// the OLED used
U8X8_SSD1306_128X64_NONAME_SW_I2C u8x8(/* clock=*/ 15, /* data=*/ 4, /* reset=*/ 16);

void setup() {

  SPI.begin(5, 19, 27, 18);
  LoRa.setPins(SS, RST, DI0);

  Serial.begin(115200);
  while (!Serial); //if just the the basic function, must connect to a computer
  delay(1000);

  u8x8.begin();
  u8x8.setFont(u8x8_font_chroma48medium8_r);

  Serial.println("Distance is  ");
  // u8x8.drawString(0, 1, "LoRa Receiver");

  if (!LoRa.begin(BAND)) {
    Serial.println("Starting LoRa failed!");
    u8x8.drawString(0, 1, "Starting LoRa failed!");
    while (1);
  }
}

void loop() {

  // try to parse packet
  int packetSize = LoRa.parsePacket();
  if (packetSize) {
    // received a packet
    Serial.print("Distance is ");
    u8x8.drawString(0, 1, "Distance is ");

    // read packet
    while (LoRa.available()) {
      receivedText = (char)LoRa.read();
      Serial.print(receivedText);
       char currentid[64];
       //receivedText.toCharArray(receivedText, 64);
       u8x8.drawString(0, 3, LoRa.parsePacket());
    }

    Serial.print(" cm ");
    Serial.println(" ");
    int distance = LoRa.parseInt();

  }
  delay(1000);
  
}
`
Here is the contents of the error pane:
Arduino: 1.8.7 (Windows 10), Board: "Heltec WiFi LoRa 32(V2), Disabled, default_8MB, 240MHz (WiFi/BT), QIO, 80MHz, 8MB (64Mb), 921600, None"

D:\Documents\Arduino\oled-RX4\oled-RX4.ino: In function 'void loop()':

oled-RX4:61:46: error: invalid conversion from 'int' to 'const char*' [-fpermissive]

        u8x8.drawString(0, 3, LoRa.parsePacket());

                                              ^

In file included from D:\Documents\Arduino\oled-RX4\oled-RX4.ino:1:0:

D:\Documents\Arduino\libraries\U8g2\src/U8x8lib.h:263:10: note:   initializing argument 3 of 'void U8X8::drawString(uint8_t, uint8_t, const char*)'

     void drawString(uint8_t x, uint8_t y, const char *s) {

          ^

Multiple libraries were found for "LoRa.h"
 Used: D:\Documents\Arduino\libraries\LoRa
 Not used: D:\Documents\Arduino\libraries\arduino-LoRa-master
exit status 1
invalid conversion from 'int' to 'const char*' [-fpermissive]

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

olikraus, 2019-10-14:
Use "print" instead of "drawString": https://github.com/olikraus/u8g2/wiki/u8g2reference#print

Dr-Bundolo, 2019-10-14:
Thanks.  Already tried that.  Still get error:

no matching function for call to 'U8X8_SSD1306_128X64_NONAME_SW_I2C::print(int, int, int&)'

 

any other suggestions?  As I mentioned earlier sketch works in serial mode.

Jeff

olikraus, 2019-10-14:
You could use sprintf.

Dr-Bundolo, 2019-10-14:
Sorry, I need a bit more info.  Is it a separate library?

Jeff

olikraus, 2019-10-15:
>  Is it a separate library?

sprintf is a standarf c function. I do not know whether it exists on Arduino Env or not

Lets go back to your problem. The only hint I have is this:

> invalid conversion from 'int' to 'const char*' 

So you have a number and you want to print this.
This is (a) a very common request from users, (b) was done successfully by many many uses with u8g2.print(), (c) is not a u8g2 problem and (d) because of c does not really fit into this issue tracker.

Maybe you should look in the internet on how to convert a number into a string. There are MANY resources on this.

Dr-Bundolo, 2019-10-15:
Thanks for the advice.

Jeff

closed 2019-11-23 by olikraus