olikraus/u8g2 #1025: SSD 1283A Constructor Request

limmers2015, 2019-10-14:
Hello everyone:

I am trying to get an SSD1283A driven 130x130 color display ([link](https://www.amazon.com/gp/product/B07GSL66HS/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1)) working in SPI with this library (u8g2). I already have the other display in my project running on it, and I would like to consolidate the two under one library. I cannot seem to find a constructor to make this work. 

Here is the code that runs the two now:
https://create.arduino.cc/editor/limmers2015/69b3f6fe-343a-4ef7-9433-eb32d8aef913/preview

I have had to modify the "LCDWIKI_SPI.h" library to even get it to work at all, if that makes a difference.

Despite extensive searching on the web, I cannot find a constructor for this display controller. If anyone has any ideas, can they point me in the proper direction? If this controller is simply not supported right now, may I ask that it become supported?

Thanks in advance!

olikraus, 2019-10-15:
U8g2 does not (and can not) support color displays.

olikraus, 2019-10-15:
Setting up a color TFT is a nightmare. Even more worse: I do not have any information about the controller.

limmers2015, 2019-10-16:
Oh. I see. I guess I will have to try another library then. Any suggestions as to what library to try? I am using this display with an I2C 128x64 OLED display if that makes a difference.

olikraus, 2019-10-17:
Hmm.. At least I do not know a suitable lib for your display.

limmers2015, 2019-11-22:
Solved the issue! Switched to an "SSD1327" display.

closed 2019-11-22 by limmers2015