olikraus/u8g2 #1028: U8G2_IL3820_V2_296X128_1_4W_SW_SPI: only left 240x128 area is drawable

talp1, 2019-10-20:
Hi olikraus,
I'm using a Waveshare 2.9inch 296x128 with an Arduino UNO, all works great but I can use only 240x128 pixel, also getting size with u8g2.getWidth() and u8g2.getHeight() I get 240 and 128, same results using u8g2.getDisplayWidth() and u8g2.getDisplayHeight().

This is my test code:

```
#include <U8g2lib.h>

const int U8G2_CLOCK = 13;
const int U8G2_DATA = 11;
const int U8G2_CS = 10;
const int U8G2_DC = 9;
const int U8G2_RESET = 8;
U8G2_IL3820_V2_296X128_1_4W_SW_SPI u8g2(U8G2_R0, U8G2_CLOCK, U8G2_DATA, U8G2_CS, U8G2_DC, U8G2_RESET);

int i;

void setup() {
  u8g2.begin();
  u8g2.setPowerSave(1);

  i = 0;
}

void loop() {
  char line1[64];
  char line2[64];
  sprintf(line1, "Dsp: %dx%d (%dx%d)", u8g2.getWidth(), u8g2.getHeight(), u8g2.getBufferTileWidth(), u8g2.getBufferTileHeight());
  sprintf(line2, "Cnt: %d", i);

  u8g2.setPowerSave(0);
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_helvR18_tr);
    u8g2.drawStr(0,24,line1);
    u8g2.drawStr(0,48,line2);

    u8g2.drawLine(20, 63, 20, 127);
    u8g2.drawLine(20, 63, 240, 63);

    int y;
    for (int x = 0; x < i % 200; x ++) {
      y = x / 3;
      u8g2.drawLine(21 + x, 127, 21 + x, 127 - y);
    }
    
  } while ( u8g2.nextPage() );
  u8g2.setPowerSave(1);
  
  delay(2000);
  i ++;
}

```

The display shows, in the first row:
Dsp: 240x128 (37x1)

If I try to draw something over x=240 on the left side (i.e. using u8g2.drawLine(20, 63, 295, 63)) I have unpredictable results, i.e. seeing them on right side, like it was using X mod 240.

If i try to rotate the display, with U8G2_R2, it also "skip" 56x128 area on the left, starting drawing (x=0) far from border.

olikraus, 2019-10-20:
https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#16-bit-mode

https://github.com/olikraus/u8g2/blob/master/doc/faq.txt#L159

talp1, 2019-10-20:
Thank you, I confirm now it works:
Dsp: 296x128 (37x1)

I hadn't thought about this problem, I would have expected it to be 256, not 240 wide.
Anyway thanks for the suggestion.

olikraus, 2019-11-23:
:)

closed 2019-11-23 by olikraus