olikraus/u8g2 #1029: Only half of SSD1306 128x64 OLED display written, other half shows noise #264

unmateria, 2019-10-20:
Hi, just sorry if I revive and old topic (#264) but I has been becoming mad with these and I saw some people had the same problem so maybe it can help others.

Solution to "Only half of SSD1306 128x64 OLED display written, other half shows noise" is just to replace the ESP32 repository with 
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json

olikraus, 2019-10-20:
ok, great if this helps. But what could be the relationship?

unmateria, 2019-10-20:
I suppose is any wire library problem because i was having problems with another i2c sensor (a fingerprint reader). The other repository was 1.0.1 but, i dont remember and cant find the url. I think It was a repository for heltec boards instead of the official one.

closed 2019-11-23 by olikraus