olikraus/u8g2 #1030: Please see whats wrong here

GrahamLambert, 2019-10-22:
[image]

Hi as you can see I only get 3/4 of the width using your example posted previously

olikraus, 2019-10-22:
Did you enable 16 bit mode?

https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#16-bit-mode

https://github.com/olikraus/u8g2/blob/master/doc/faq.txt#L159

GrahamLambert, 2019-10-22:
Many thanks!
that did it.  In your section "Controller "il3820", Display 
"296x128"Controller "il3820", Display "296x128"
how do I choose the right one?  I only want to display text on an UNO
I'm currently using this one and it seems ok

U8G2_IL3820_V2_296X128_1_4W_SW_SPI u8g2(U8G2_R0,5, 7, 8, 2, 3);

Regards from Graham Lambert
graham@versatronics.co.za
021 713 1334 / 082 419 1334
www.versatronics.co.za

<author@noreply.github.com>
Sent: 10/22/2019 8:26:33 AM

GrahamLambert, 2019-10-22:
PS
Is there a page somewhere with font examples?
  u8g2.setFont (u8g2_font_ncenB14_tr);

Regards from Graham Lambert
graham@versatronics.co.za
021 713 1334 / 082 419 1334
www.versatronics.co.za

olikraus, 2019-10-22:
> how do I choose the right one? 

Basically its a speed vs RAM trade off... HW SPI would be also better, but I think the overall delay for e-paper is much bigger.

> Is there a page somewhere with font examples?

https://github.com/olikraus/u8g2/wiki/fntlistall

or:

https://github.com/olikraus/u8g2/wiki/fntgrp

closed 2019-11-23 by olikraus