olikraus/u8g2 #1032: Mirror Display 128x32 "0.96 ?

MagierKummer537, 2019-10-22:
Hi olikraus,
I really enjoy your u8g Libary. Good work and sorry for my bad english. But i wanna mirror/invert text of my 128x32 SSD1306 "0.96 Display, wich I defined in my Code with "U8GLIB_SSD1306_128X32 u8g(U8G_I2C_OPT_NONE);". I know, it has to do with this 0x0c8, but i dont really know, what to do...
Hope you can help me. :D
Thank you!

MagierKummer537, 2019-10-22:
U8G2_SSD1306_128X32_UNIVISION_F_SW_I2C u8g2(U8G2_MIRROR, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);

olikraus, 2019-10-22:
Es wäre dann diese Zeile: https://github.com/olikraus/u8g2/blob/f0efab642af931ba228e5094e27f8efc0c736ad5/csrc/u8x8_d_ssd1306_128x32.c#L56

Kannst du das bei dir lokal ändern? c0 vs. c8
IDE restart nicht vergessen

MagierKummer537, 2019-10-22:
Ich habe diese Zeile bereits geändert, leider nichts gebracht, an den restart habe ich gedacht. Was meinst du mit lokal? und mit c0 vs c8?
MFG Alex

olikraus, 2019-10-23:
> Was meinst du mit lokal? 

Na, bei Dir auf deiner Festplatte im Library Unterverzeichnis wo U8g2 installiert ist

> und mit c0 vs c8?

Der Austausch des Wertes in der U8g2 Datei, also anstelle "0x0c8" dann "0x0c0"
Einen visuellen Effect muss das schon haben.

olikraus, 2019-10-23:
Es sollte schon etwas passieren. Du könntest ja einen Syntax Error einbauen um zu sehen ob es die richtige Datei ist.

MagierKummer537, 2019-10-23:
Ok, wie?

olikraus, 2019-10-23:
Ähm... Naja einfach irgendeinen Unsinn in die Datei schreiben. Wie wäre es mit einem + an den Anfang der Datei.

MagierKummer537, 2019-10-23:
Okay, danke für den Tipp. War anscheinend nicht die richtige Datei, kann es in der IDE ohne Probleme kompilieren. Werde morgen, weiter nach einer Lösung suchen, da ich jetzt leider keine Zeit mehr habe.
Danke soweit, ich antworte morgen. Die Datei war diese: "u8x8_d_ssd1306_128x32.c". :D 
MFG Alex

olikraus, 2019-10-24:
Die Arduino IDE ist da etwas seltsam. Manchmal gibt es mehrere Varianten einer lib an verschiedenen Stellen.

MagierKummer537, 2019-10-24:
Jo, richtige Datei gefunden. Geändert und siehe da, es funktioniert. :D Danke dir, olikraus! Coole Libary, hoffe man kann vlt. in der nächsten u8g3, das einfacher machen. ^^ 
MFG Alex

MagierKummer537, 2019-10-24:
Ja, habe ich mir gedacht. :,D
Aber wäre ein cooles Feature.
Naja, danke auf jeden Fall. :)

MagierKummer537, 2019-10-25:
Machst du das eigentlich aus Spaß oder wirst du dafür bezahlt?  Nur mal so aus Interesse. :)

olikraus, 2019-10-25:
Schön wäre es, dafür bezaht zu werden :-)
Tatsächlich mache ich das meistens nur aus Spaß. 
Klar, wenn mir mal jemand ein Display überlässt, habe ich schon auch mal den Support dafür programmiert. 
U8g2 selbst ist weitgehend abgeschlossen, d.h. ich werde vermutlich keine großen neuen Features einbauen (vermutlich auch keinen Mirror R3 mode) weil mit das mittlerweile etwas zu langweilig geworden ist und meine eigenen Projekte (zumindest derzeit) fertig sind.

Das gute an einem Spassprojekt ist, dass man sich nicht mit den Anwendern rumärgern muss. Oder wenn es eine besser Lib (für ein bestimmtes Display) gibt, dann ist mir das auch egal ;-)

Support für neue Displays mache ich aber trotzdem gerne, wenn es mein Budget hergibt. Die Lib ist so aufgebaut, dass ein neues Display in ein paar Minuten implementiert ist. Schwieriger ist es dagegen das Display dann auch zum Laufen zu bekommen. Das kann zwischen wenigen Minuten bis mehrere Tage dauern.
Mein Rekord war mal ein Jahr mit dem ST75256. Aber das macht dann schon Laune, wenn man dann endlich was auf dem ST75256 display (z.B. JLX256160) sieht.

So ein bischen kommt auch der Sammeltrieb hinzu: Es ist eine tolle sache möglichst alle monochromen displays, die es für Privatanwender zu kaufen gibt auch zu unterstützen.
Anders gesagt: Alle Displays die u8g2 unterstützt, gab oder gibt es für Privatanwender zu kaufen.

MagierKummer537, 2019-10-25:
Jo, danke. Interessant, cool auf jeden Fall, kannst Leuten wie mir (bin 15 Jahre alt) dabei helfen, mehr zu lernen. :D 
Ich schließe dann mal das Ganze. Danke dir auf jeden Fall für deine Hilfe.

olikraus, 2019-10-25:
> Jo, danke. Interessant, cool auf jeden Fall, kannst Leuten wie mir (bin 15 Jahre alt) dabei helfen, mehr zu lernen.

Na denn... viel Erfolg. Lass es mich wissen, wenn ich irgendwie helfen kann.

closed 2019-10-25 by MagierKummer537

StigDenmark, 2021-03-21:
Had a problem with a u8x8_st7567_enh_dg128064 and this solution also worked for me when I found the right place and made the change.