olikraus/u8g2 #1033: Dynamic memory and u8x8 does not name a type

tomsharpe123, 2019-10-23:
Hi olikraus, 
I am trying to use the libraries to code a speed coach to tell me how fast I am travelling in a boat.
I tried coding it with the U8g2 library but it takes 110% of the
dynamic memory of my Arduino nano so will not upload. When I tried to start changing it
to U8x8 it started to chuck up
U8X8_SSD1309_128X64_NONAME0_F_4W_SW_SPI' does not name a type errors.
Is there any easy way to reduce the dynamic memory of the U8g2 code or to change it to the u8x8? 

Thanks Tom

olikraus, 2019-10-23:
All your ideas are possible and we can discuss in both directions, but I need little bit more information.

> but it takes 110% of the dynamic memory

Which constructor did you use and what is your Arduino board?

> U8X8_SSD1309_128X64_NONAME0_F_4W_SW_SPI

How does the full line like like?

tomsharpe123, 2019-10-23:
When I try and upload the U8g2 code it does not upload because it says: 
Sketch uses 18432 bytes (60%) of program storage space. Maximum is 30720 bytes.
Global variables use 2259 bytes (110%) of dynamic memory, leaving -211 bytes for local variables. Maximum is 2048 bytes.
Not enough memory; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing your footprint.
Error compiling for board Arduino Nano.

This is the code: 
`#include <SPI.h>
#include <SandTimer.h>
#include <millisDelay.h>
#include <Adafruit_GPS.h>   //Libraries
#define GPSSerial Serial
Adafruit_GPS GPS(&GPSSerial);
SandTimer timer1;
#define GPSECHO false
#include <Arduino.h>
#include <U8g2lib.h>
#ifdef U8X8_HAVE_HW_SPI
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif
U8G2_SSD1309_128X64_NONAME0_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
uint32_t timer = millis();
const int chipSelect = 4;
float ms = 0; //Meters/Second variable
float avems = 0;  //Average M/S variable
float mscounter = 0;  //Speed total (for average)
int integcounter = 0; //Counts half-seconds
float meterstraveled = 0; //Total meters
int splittotal = 0;   //Split total (in seconds)  (Essential to be an integer)
int splitseconds = 0; //Seconds on split display  (Essential to be an integer)
int splitminutes = 0; //Minutes on split display  (Essential to be an integer)
int seconds = 0;
int minutes = 0;
int hours = 0;    //Self-explanatory (for timers)
int splitaveragetotal = 0; //All split seconds added together (for average split)
int splitaverageseconds = 0; //Split average seconds
int splitaverageminutes = 0; //Split average minutes
int splitaverage = 0; //Split average
int splitaveragesecondsdisp = 0; //Display variables, fixes a bug
int splitaverageminutesdisp = 0;
float splitmdisp = 0;
float splitsdisp = 0;  //Display Variables
int tc = 0; //Counter variable
int dp = 0; //Data-point variable
int dpave = 0; //Average of the data-points
int dprec = 0; //Counter variable
float knot = 0;//Speed in knots recorded from GPS
float aveknots = 0;  //Speed in knots averaged over 1/2 second (equal to 1 data point and 5 knot readings)
int knotcounter = 0; //Counter variable
int sensorValue = analogRead(A3); //For the data write switch
float msfm = 0; //M/S for the meters calculation

millisDelay secdelay;   //Timing delay (for timer)

void setup() {
  float voltage = sensorValue * (5.0 / 1023.0);   //Converts A3 reading into a 5v equivalent
  Wire.begin();
  Wire.setClock(400000L);
  Serial.begin(115200);
  GPS.begin(9600);
  GPS.sendCommand(PMTK_SET_NMEA_OUTPUT_RMCGGA);
  GPS.sendCommand(PGCMD_ANTENNA);
  delay(1000);    //GPS initialization
  u8g2.begin();
  GPSSerial.println(PMTK_Q_RELEASE);
  int realhours = GPS.hour - 8;
  if (realhours < 0) {
    realhours = realhours + 24; //For SD timestamp (24 hour format)
[... 105 lines cut ...]
            u8g2.setFont(u8g2_font_courR08_tr);
            u8g2.setCursor(1, 25);
            u8g2.print(splitmdisp);
            u8g2.drawStr(1, 50, ":");

            u8g2.setFont(u8g2_font_courR08_tr);
            u8g2.setCursor(1, 75);
            u8g2.print(splitsdisp); //adjust pixels to get a good display
            u8g2.sendBuffer();          // transfer internal memory to the display

          }
        }
      }
    }
  }
}
`
The full line of the error for the U8x8 is:
 exit status 1
'U8X8_SSD1309_128X64_NONAME0_F_4W_SW_SPI' does not name a type
and 

u8x8_code:185:17: error: expected unqualified-id before '.' token

             U8X8.setCursor(1, 75);

tomsharpe123, 2019-10-23:
For some reason it won't let me add all the code in one section

olikraus, 2019-10-23:
> U8G2_SSD1309_128X64_NONAME0_F_4W_SW_SPI

The "F" constructor will create a 128x64 pixel buffer (1024 bytes) in your Nano. This is 50% of the overall RAM size. A solution could be to learn how to use the page mode constructors (they have "1" or "2" in their name instead of "F")

> The full line of the error for the U8x8 is:

I mean the source code line instead of the error message. Basically this indicates an error in your constructor call. From where did you copy the constructor?

AnHardt, 2019-10-23:
To save a lot of memory try to avoid the "\*_F_\*" constructors.
U8G2_SSD1309_128X64_NONAME0_1_4W_HW_SPI will need 128 byte instead of 1024.
But the you have to switch from `u8g2.clearBuffer()/u8g2.sendBuffer()` to `firstPage()/nextPage()`

tomsharpe123, 2019-10-23:
That's the code with too much dynamic memory the code with U8x8 does not name a type error is: 

`#include <SPI.h>
#include <SandTimer.h>
#include <millisDelay.h>
#include <Adafruit_GPS.h>   //Libraries
#define GPSSerial Serial
Adafruit_GPS GPS(&GPSSerial);
SandTimer timer1;
#define GPSECHO false
#include <Arduino.h>
#include <U8g2lib.h>
#ifdef U8X8_HAVE_HW_SPI
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif
U8X8_SSD1306_128X64_4W_SW_SPI u8x8(/* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
uint32_t timer = millis();
const int chipSelect = 4;
float ms = 0; //Meters/Second variable
float avems = 0;  //Average M/S variable
float mscounter = 0;  //Speed total (for average)
int integcounter = 0; //Counts half-seconds
float meterstraveled = 0; //Total meters
int splittotal = 0;   //Split total (in seconds)  (Essential to be an integer)
int splitseconds = 0; //Seconds on split display  (Essential to be an integer)
int splitminutes = 0; //Minutes on split display  (Essential to be an integer)
int seconds = 0;
int minutes = 0;
int hours = 0;    //Self-explanatory (for timers)
int splitaveragetotal = 0; //All split seconds added together (for average split)
int splitaverageseconds = 0; //Split average seconds
int splitaverageminutes = 0; //Split average minutes
int splitaverage = 0; //Split average
int splitaveragesecondsdisp = 0; //Display variables, fixes a bug
int splitaverageminutesdisp = 0;
float splitmdisp = 0;
float splitsdisp = 0;  //Display Variables
int tc = 0; //Counter variable
int dp = 0; //Data-point variable
int dpave = 0; //Average of the data-points
int dprec = 0; //Counter variable
float knot = 0;//Speed in knots recorded from GPS
float aveknots = 0;  //Speed in knots averaged over 1/2 second (equal to 1 data point and 5 knot readings)
int knotcounter = 0; //Counter variable
int sensorValue = analogRead(A3); //For the data write switch
float msfm = 0; //M/S for the meters calculation

millisDelay secdelay;   //Timing delay (for timer)

void setup() {
  float voltage = sensorValue * (5.0 / 1023.0);   //Converts A3 reading into a 5v equivalent
  Wire.begin();
  Wire.setClock(400000L);
  Serial.begin(115200);
  GPS.begin(9600);
  GPS.sendCommand(PMTK_SET_NMEA_OUTPUT_RMCGGA);
  GPS.sendCommand(PGCMD_ANTENNA);
  delay(1000);    //GPS initialization
 U8X8.begin();
  GPSSerial.println(PMTK_Q_RELEASE);
  int realhours = GPS.hour - 8;
  if (realhours < 0) {
    realhours = realhours + 24; //For SD timestamp (24 hour format)
  }
}

const char s[] PROGMEM = "Progmem";
const char s1[] PROGMEM  = "ABC" ;
[... 93 lines cut ...]
            U8X8.setFont(U8X8_font_courR08_tr);

            U8X8.setCursor(0, 25);
            U8X8.setFont(U8X8_font_courR08_tr);
            U8X8.print(meterstraveled);
            U8X8.drawStr(1, 10, "S:");

            U8X8.setFont(U8X8_font_courR08_tr);
            U8X8.setCursor(1, 25);
            U8X8.print(splitmdisp);
            U8X8.drawStr(1, 50, ":");

            U8X8.setFont(U8X8_font_courR08_tr);
            U8X8.setCursor(1, 75);
            U8X8.print(splitsdisp); //adjust pixels to get a good display
            U8X8.sendBuffer();          // transfer internal memory to the display

          }
        }
      }
    }
  }
}`

I will have a look into page constructors and see if I can get them to work

olikraus, 2019-10-23:
type error: You must use "u8x8" instead of "U8X8".

tomsharpe123, 2019-10-23:
Thanks, I've changed that and still throwing up the same errors. 
The U8g2 version of the code is now down to 72% of the memory by changing the constructor to a page buffer one. Is there anything else that I need to change in the code for it to work?

olikraus, 2019-10-23:
"1" constructor: you can not use sendBuffer any more. See the u8g2 examples...

tomsharpe123, 2019-10-24:
I have tried to rewrite it with a page buffer but am still getting loads of errors
 `#include <SPI.h>
#include <SandTimer.h>
#include <millisDelay.h>
#include <Adafruit_GPS.h>   //Libraries
#define GPSSerial Serial
Adafruit_GPS GPS(&GPSSerial);
SandTimer timer1;
#define GPSECHO false
#include <Arduino.h>
#include <U8g2lib.h>
#ifdef U8X8_HAVE_HW_SPI
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif
U8G2_SSD1309_128X64_NONAME2_2_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
uint32_t timer = millis();
const int chipSelect = 4;
float ms = 0; //Meters/Second variable
float avems = 0;  //Average M/S variable
float mscounter = 0;  //Speed total (for average)
int integcounter = 0; //Counts half-seconds
float meterstraveled = 0; //Total meters
int splittotal = 0;   //Split total (in seconds)  (Essential to be an integer)
int splitseconds = 0; //Seconds on split display  (Essential to be an integer)
int splitminutes = 0; //Minutes on split display  (Essential to be an integer)
int seconds = 0;
int minutes = 0;
int hours = 0;    //Self-explanatory (for timers)
int splitaveragetotal = 0; //All split seconds added together (for average split)
int splitaverageseconds = 0; //Split average seconds
int splitaverageminutes = 0; //Split average minutes
int splitaverage = 0; //Split average
int splitaveragesecondsdisp = 0; //Display variables, fixes a bug
int splitaverageminutesdisp = 0;
float splitmdisp = 0;
float splitsdisp = 0;  //Display Variables
int tc = 0; //Counter variable
int dp = 0; //Data-point variable
int dpave = 0; //Average of the data-points
int dprec = 0; //Counter variable
float knot = 0;//Speed in knots recorded from GPS
float aveknots = 0;  //Speed in knots averaged over 1/2 second (equal to 1 data point and 5 knot readings)
int knotcounter = 0; //Counter variable
int sensorValue = analogRead(A3); //For the data write switch
float msfm = 0; //M/S for the meters calculation

millisDelay secdelay;   //Timing delay (for timer)

void setup(void) {
  float voltage = sensorValue * (5.0 / 1023.0);   //Converts A3 reading into a 5v equivalent
  Wire.begin();
  Wire.setClock(400000L);
  Serial.begin(115200);
  GPS.begin(9600);
  GPS.sendCommand(PMTK_SET_NMEA_OUTPUT_RMCGGA);
  GPS.sendCommand(PGCMD_ANTENNA);
  delay(1000);    //GPS initialization
  u8g2.begin();
  GPSSerial.println(PMTK_Q_RELEASE);
  int realhours = GPS.hour - 8;
  if (realhours < 0) {
    realhours = realhours + 24; //For SD timestamp (24 hour format)
  }
}

const char s[] PROGMEM = "Progmem";
const char s1[] PROGMEM  = "ABC" ;
const char s2[] PROGMEM  = "DEF" ;
[... 118 lines cut ...]

Page_buffer:185:3: error: expected 'while' before '}' token

   }

   ^

Page_buffer:185:3: error: expected '(' before '}' token

Page_buffer:185:3: error: expected primary-expression before '}' token

Page_buffer:185:3: error: expected ')' before '}' token

Page_buffer:185:3: error: expected ';' before '}' token

C:\Users\User\OneDrive\Documents\Arduino\Page_buffer\Page_buffer.ino: At global scope:

Page_buffer:186:1: error: expected declaration before '}' token

 }

 ^

exit status 1
expected 'while' before '}' token

olikraus, 2019-10-25:
your "if" does not have a closing "}"

tomsharpe123, 2019-10-25:
Which if statement?

olikraus, 2019-10-25:
line 153

tomsharpe123, 2019-10-25:
I've corrected that but it still has the same 6 errors

olikraus, 2019-10-25:
> I've corrected that but it still has the same 6 errors

Then it is not corrected. What can I say, the "while" does not fit to the corresponding "do"...

tomsharpe123, 2019-10-26:
Is there an easy way of fixing that?

olikraus, 2019-10-26:
Sure

olikraus, 2019-10-26:
Excause the short answer (from my mobile phone). Long version: I really like to answer questions regarding u8g2, but I can not teach or fix C/C++ basics...

tomsharpe123, 2019-11-02:
I have fixed the basic issues with the code and it is now uploading but I am only seeing the top half of the number

tomsharpe123, 2019-11-02:
Is there any way to fix this?

olikraus, 2019-11-02:
Probably the picture loop (first/next page) loop is not executed correctly. 
My suggestion is to put all u8g2 graphics commands into a procedure (e.g. "draw()")
Then put the while loop around this call.

You can see this here: The complexity of the draw commands are hidden inside the function calls. The first/next page loop is clearly visible:

https://github.com/olikraus/u8g2/blob/f0efab642af931ba228e5094e27f8efc0c736ad5/sys/arduino/u8g2_page_buffer/Weather/Weather.ino#L366-L370

closed 2019-11-23 by olikraus