olikraus/u8g2 #1034: Fonts resize

EliasA97, 2019-10-25:
Hi, this library does a fine job but I have a small problem. 
I must use greek fonts (u8g2_font_10x20_t_greek) but unfortunatelly I didn't found much fonts with different sizes. I mean there are about 5 different fonts with different sizes that include greek characters. I like this specific font but is there any way to make it in other size, like 8x9 or 15x16?

olikraus, 2019-10-25:
First: There is no resize possible. U8g2 only supports fixed size bitmap fonts. This means we have to rely on font authors to create greek letters.

The good news is: Many fonts have greek letters included, but I did not always render the greek fonts for u8g2. 

Probably all fonts here should include greek letters in the font source, but I did not include them for u8g2: https://github.com/olikraus/u8g2/wiki/fntgrpadobex11

I am very sure that all fonts here also include greek letters: https://github.com/olikraus/u8g2/wiki/fntgrpx11

So for example, if you want greek letters for this font:
[image]

just let me know. Or: You can also render the fonts by your self with the bdfconv tool (https://github.com/olikraus/u8g2/tree/master/tools/font/bdfconv) and the corresponding font source (e.g https://github.com/olikraus/u8g2/blob/master/tools/font/bdf/8x13.bdf)

Another approach is to look around in the u8g2 fonts: For example NBP group contains a lot of fonts with greek gylphs: https://github.com/olikraus/u8g2/wiki/fntgrpnbp

EliasA97, 2019-10-26:
Hi Oliver, I think it's interesting to make my own fonts but I tried with the bdfconv without success. The program does not even start. I managed to convert a font to bdf with fony but I'm stuck.  Is there any tutorial or video to explain the conversion procedure? Also how can I insert my custom font in the u8g2 library? I'm asking that because I don't want to bother you with my tests!!

olikraus, 2019-10-26:
> Update: I managed to get a custom font converted by fony and bdfconv but it only shows english glyphs and not the greek characters that I want. Any ideas about that??

You need specify the unicode range for bdfconv with the -m command.
```
bdfconv -f 1 -v -m '32-127,895-1023'
```
where  895-1023 should be the unicode range for the greek letters.

EliasA97, 2019-10-26:
I don't think thats the problem. When I'm importing a font from windows to fony, the programm only imports characters up to 255, so there is no way to load the greek glyphs which are on 895-1023. Also it exports the bdf only up to 255.

olikraus, 2019-10-26:
Windows might require " instead of single quotes. See also the u8g2 FAQ

EliasA97, 2019-10-26:
I did that with ". Also I did exactly what daq says. Do you want to send you my custom font to try it yourself?

olikraus, 2019-10-26:
If you created the font with fony then indeed the glyphs go until 255 only. But they will appear under the code you have entered them in fony

EliasA97, 2019-10-26:
So you're saying that I have to import a font in fony, export it as bdf, use the bdfconv, copy the code in my arduino project and then display the greek glyphs?

EliasA97, 2019-10-26:
Also I have that problem in the picture. The glyphs with ? are greek letters. Do you know why this happens?
[image]

olikraus, 2019-10-26:
> So you're saying that I have to import a font in fony,

Ah, no, you just mentioned fony before, so I thought you might have used it.
Actually it is better not to use fony if you have an existing font with greek letters at the right position, because fony can not work with glyphes above 255.

You could also use -m "32-65535" to convert everything.

EliasA97, 2019-10-26:
I didn't know that. I have many fonts with greek glyphs but they are .otf and .ttf. How can I convert them to bdf without fony?

olikraus, 2019-10-26:
I personally use "otf2bdf".

EliasA97, 2019-10-26:
I just found out that otf2bdf is for linux and I can't access it. Would it be easy for you to convert my font? If you have time for that, please check the link. 

https://we.tl/t-50ZGk1mlMW

olikraus, 2019-10-26:
Which pt site?

EliasA97, 2019-10-26:
20

olikraus, 2019-10-26:
Conversion:

```
	./otf2bdf/otf2bdf -a  -p 20 -o analecta.bdf Analecta.otf
	./bdfconv -v -f 1 -m '32-65535' analecta.bdf -o u8g2_font_analecta.c  -n u8g2_font_analecta -d ../bdf/helvB18.bdf 
```

[image]

the zip contains the bdf and the u8g2 font.

[analecta.zip](https://github.com/olikraus/u8g2/files/3775169/analecta.zip)

EliasA97, 2019-10-26:
wow. Works like a charm. You are the best!!!

I have a last question about doing this on my own. Did you made the font only by these 2 commands? Also how did you made image?

olikraus, 2019-10-26:
> Did you made the font only by these 2 commands? 

yes

> Also how did you made the image?

The second command will generate a fixed file called bdf.tga (because of the -d option), which was converted with the tool "convert" to png (https://imagemagick.org/script/convert.php).

EliasA97, 2019-10-26:
Great. Thank's again Oliver. I must tell you that I'll contribute to your library with some greek fonts that I hope to make and send you to add them. For now I'm closing the case.

closed 2019-10-26 by EliasA97

EliasA97, 2019-10-26:
Hi again Oliver, I think I'm being very annoying by now but I followed your steps in order to recreate the bdf file but with no result. Is it possible to make a video or something to show me exactly your steps?
[image]

EliasA97, 2019-10-26:
Actually I made it with this command (otf2bdf -r 72 -p 16 Analecta.otf -o Analecta.bdf) that I fount here:
https://github.com/olikraus/u8g2/wiki/internal#ttf-to-bdf

What are the differences between `otf2bdf` -r 72 -p 16 Analecta.otf -o Analecta.bdf` and `otf2bdf -a  -p 20 -o analecta.bdf Analecta.otf` ??

olikraus, 2019-10-27:
Looks like you have installed linux :-)

>  Is it possible to make a video or something to show me exactly your steps?

I hope we can manage this without video. 

"-p" sets the desired point size (you had requested 20)
"-r" would set the resolution. Actually I also do not exactly know its effect.
"-a" will force auto hinting. It is a new feature of the freetype lib which I added to otf2bdf by myself (https://github.com/olikraus/u8g2/blob/master/tools/font/otf2bdf/otf2bdf.c#L1300)

To get the "-a" option, you need to download my version of otf2bdf: https://github.com/olikraus/u8g2/tree/master/tools/font/otf2bdf
More specific: You can clone u8g2 repoo (under linux):
`git clone https://github.com/olikraus/u8g2.git`
The goto tools/font/otf2bdf and type make

Regarding otf2bdf: it is more kind of experimenting to figure out which options look best. 

A nice tool is "ftview" (again linux) which allows you to see the effect of each option and size immediately.

EliasA97, 2019-10-29:
Hi Oliver, I just wanted to tell you that everything works. I made many fonts with your help. Thank's again!!!