olikraus/u8g2 #1035: Is there any Cyrillic font for U8x8?

ilgiznurgaliev, 2019-10-29:
Hi,

So the question is in subject? Seems no any Cyrillic font is available for U8x8...

Regards, Casper.

olikraus, 2019-10-29:
Correct, u8x8 does not support Unicode. However Cyrillic glyphs could be mapped into this region and a new font could be created. Do you have any font in mind for this purpose?

ilgiznurgaliev, 2019-10-29:
Thanks for your responce.
Lets take u8g2_font_5x8_t_cyrillic as example. Did I understand you correctly, that I can import some glyphs from any cyrillic font? If so I would really appreciate for a small example of how to do this.

olikraus, 2019-10-29:
The command probably will be something like this:
```
bdfconv -v -f 2 -m "1024-1279>0" -o u8x8_5x8_cyrillic.c -n u8x8_5x8_cyrillic 5x8.bdf
```
The "1024-1279>0" expression will map the cyrillic unicode block into the range from 0 to 255. I am not sure whether this really makes sense, it depends also which you really require.

```
bdfconv -v -f 2 -m "32-127;1024-1151>128;" -o u8x8_5x8_cyrillic.c -n u8x8_5x8_cyrillic 5x8.bdf
```

This should keep the latin chars from ASCII range 32 to 127 and map the lower part of the cyrillic chars to the range from 128 to 255.

Not tested, but it should work like this...

ilgiznurgaliev, 2019-10-29:
Many thanks. I'll try to install bdfconv and test this.

closed 2019-11-23 by olikraus