olikraus/u8g2 #1037: I2C addressing question.

bobbryce, 2019-11-03:
Hi,
First off, thanks for a great library. I'm using a Cortex M0 processor (Feather M0 Express) and a none 128x64 I2C display.

The code examples work well (Hello World and Graphics test tried so far) but I don't understand how the code knows the I2C address to use. I ran an I2C scan to get the display address and then had no need to use it in the example code. This will become important  later on as I want to use three different devices on the same I2C bus.

v1993, 2019-11-03:
AFAIK, there is default address for each display model hardcoded into u8g2 library, so you don't have to change it in most cases. But keep in mind that this means that two displays with default address can't be used on same bus.

olikraus, 2019-11-03:
There is a command to change the controller specific I2C address: https://github.com/olikraus/u8g2/wiki/u8g2reference#seti2caddress

To my knowledge all the controller only support up to two different I2C addresses. Adding more than three devices *with the same controller* will be difficult.

bobbryce, 2019-11-03:
Thank you for the responses, especially the seti2caddress.

Understood on the likely max two displays per controller but I'm adding an
i2c PLL Clock chip plus an i2c encoder control and hopefully they can all
share the same bus.

Time will tell.

closed 2019-11-23 by olikraus